openwrt/packages #1092: micropython-lib compiling error

taeasy, 2015-03-29:
cp: `./example_yield_coro.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/example_yield_coro.py' are the same file
cp: `./uu.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/uu.py' are the same file
cp: `./json' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/json' are the same file
cp: `./functools.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/functools.py' are the same file
cp: `./asyncio_slow.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/asyncio_slow.py' are the same file
cp: `./getopt.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/getopt.py' are the same file
cp: `./hashlib' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/hashlib' are the same file
cp: `./shutil.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/shutil.py' are the same file
cp: `./os' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/os' are the same file
cp: `./re.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/re.py' are the same file
cp: `./fnmatch.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/fnmatch.py' are the same file
cp: `./datetime.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/datetime.py' are the same file
cp: `./getpass.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/getpass.py' are the same file
cp: `./html' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/html' are the same file
cp: `./upip.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/upip.py' are the same file
cp: `./timeit.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/timeit.py' are the same file
cp: `./tempfile.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/tempfile.py' are the same file
cp: `./cmd.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/cmd.py' are the same file
cp: `./example_sigint_ign.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/example_sigint_ign.py' are the same file
cp: `./test' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/test' are the same file
cp: `./reprlib.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/reprlib.py' are the same file
cp: `./pprint.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/pprint.py' are the same file
cp: `./shelve.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/shelve.py' are the same file
cp: `./queue.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/queue.py' are the same file
cp: `./logging.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/logging.py' are the same file
cp: `./tests' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/tests' are the same file
cp: `./collections' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/collections' are the same file
cp: `./shlex.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/shlex.py' are the same file
cp: `./errno.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/errno.py' are the same file
cp: `./keyword.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/keyword.py' are the same file
cp: `./csv.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/csv.py' are the same file
cp: `./hmac.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/hmac.py' are the same file
cp: `./abc.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/abc.py' are the same file
cp: `./random.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/random.py' are the same file
cp: `./example-extract.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/example-extract.py' are the same file
cp: `./dbm.py' and `/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/_install_tmp/dbm.py' are the same file
[... cut ...]
make[3]: *** [/home/main/x86_64/build_dir/target-x86_64_uClibc-0.9.33.2/micropython-lib-0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-654c7d288603f7dae09eb09b57fb67b38c7ac6c3/.built] Error 2
make[3]: Leaving directory `/home/main/x86_64/feeds/packages/lang/micropython-lib'
make[2]: *** [package/feeds/packages/micropython-lib/compile] Error 2
make[2]: Leaving directory `/home/main/x86_64'
make[1]: *** [/home/main/x86_64/staging_dir/target-x86_64_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/x86_64'
make: *** [world] Error 2
main@bogon:~/x86_64$

roger-, 2015-03-30:
Hmm... maybe you need to do a `make clean`?

This seems to build fine on the [buildbot](https://downloads.openwrt.org/snapshots/trunk/x86_64/generic/packages/packages/micropython-lib_0.1-20150302-654c7d288603f7dae09eb09b57fb67b38c7ac6c3-1_x86_64.ipk).

roger-, 2015-09-17:
@taeasy

Any updates on this issue?

hnyman, 2016-08-26:
No action for a while, micropython-lib compiles ok in the trunk buildbots.

closed 2016-08-26 by hnyman