openwrt/packages #1095: mjpg-streamer build error (ERROR 404: Not Found) on 14.07 branch

mariarti, 2015-03-30:
make[3]: Entering directory `/home/tt/mt2/openwrt/feeds/packages/multimedia/mjpg-streamer'
mkdir -p /home/tt/mt2/openwrt/dl
/home/tt/mt2/openwrt/scripts/download.pl "/home/tt/mt2/openwrt/dl" "cambozola-latest.tar.gz" "6c48fd994685d4d72668850eeb613e24" "http://www.andywilcock.com/code
/cambozola"
--2015-03-30 15:17:23--  http://www.andywilcock.com/code/cambozola/cambozola-latest.tar.gz
Resolving www.andywilcock.com... 50.53.126.251
Connecting to www.andywilcock.com|50.53.126.251|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 237089 (232K) [application/x-gzip]
Saving to: `STDOUT'

     0K .......... .......... .......... .......... .......... 21%  121K 1s
    50K .......... .......... .......... .......... .......... 43%  239K 1s
   100K .......... .......... .......... .......... .......... 64%  261K 0s
   150K .......... .......... .......... .......... .......... 86%  836K 0s
   200K .......... .......... .......... .                    100%  862K=0.9s

2015-03-30 15:17:25 (255 KB/s) - written to stdout [237089/237089]

MD5 sum of the downloaded file does not match (file: c9b0da91f8e6e72efccd307e04e2b75b, requested: 6c48fd994685d4d72668850eeb613e24) - deleting download.
--2015-03-30 15:17:25--  http://mirror2.openwrt.org/sources/cambozola-latest.tar.gz
Resolving mirror2.openwrt.org... 46.4.11.11
Connecting to mirror2.openwrt.org|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-03-30 15:17:25 ERROR 404: Not Found.

Download failed.
--2015-03-30 15:17:25--  http://downloads.openwrt.org/sources/cambozola-latest.tar.gz
Resolving downloads.openwrt.org... 78.24.191.177
Connecting to downloads.openwrt.org|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-03-30 15:17:25 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[3]: *** [/home/tt/mt2/openwrt/dl/cambozola-latest.tar.gz] Error 2
make[3]: Leaving directory `/home/tt/mt2/openwrt/feeds/packages/multimedia/mjpg-streamer'
make[2]: *** [package/feeds/packages/mjpg-streamer/compile] Error 2
make[2]: Leaving directory `/home/tt/mt2/openwrt'
make[1]: *** [/home/tt/mt2/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/tt/mt2/openwrt'
make: *** [world] Error 2

thess, 2015-03-30:
Update your package repo (./scripts/feeds update) and try again. Fixed in #1079, ref: #1076

mariarti, 2015-03-30:
Issue seems gone away. Thanks!
Just for interesting: cambozola-latest.tar.gz - this package will never appear in http://downloads.openwrt.org/sources/ ?
Only here http://www.andywilcock.com/code/cambozola/cambozola-latest.tar.gz?

closed 2015-03-30 by mariarti

deniska86, 2015-05-16:
Hello! my name Denis. I managed to solve the problem. You need to comment out the line in the file
<you directory>feeds/packages/multimedia/mjpg-streamer/Makefile

#CAMBOZOLA:=cambozola-0.936.tar.gz

#define Download/cambozola
 # URL:=http://www.andywilcock.com/code/cambozola
 # FILE:=$(CAMBOZOLA)
 # MD5SUM:=35c45188aa9635aef2b745c35c311396
#endef

# Fetch latest cambozola that works with latest Java(s)
# Yes, I know this is ugly
#define Build/Prepare
 #   $(call Build/Prepare/Default)
  #  ifeq ($(CONFIG_MJPG_STREAMER_WWW),y)
#       $(eval $(call Download,cambozola))
#       $(TAR) -xvf $(DL_DIR)/$(CAMBOZOLA) --strip=2 --wildcards \
#               -C $(PKG_BUILD_DIR)/www */dist/cambozola.jar
 #   endif
#endef

Then build script will not appeal to the lack of Internet resources.
Sorry for the broken English, I'm bad at school =)

thess, 2015-05-16:
@deniska86 - author's site is back online -- I'm looking for a different solution to fetching cambozola in the meantime

deniska86, 2015-05-17:
Yesterday brought the firmware, it works! 
And what do you cambozola?

thess, 2015-05-17:
Cambozola is an embedded java based mjpeg viewer/control applet
http://www.andywilcock.com/code/cambozola

deniska86, 2015-05-17:
I know what cambozola =) I have successfully implemented their projects and budget without it. If I need something, I just ask google and I correct the code for themselves. Here is one of premera http://77.106.103.137:8081

deniska86, 2015-05-17:
we in Russia normal cameras are expensive, so my work demanded =) ..

deniska86, 2015-05-17:
kambozola good thing, tried it, but in practice nothing more than me does not help ...
Now there is everything in html or see any reason that either complicate Java ..

thess, 2015-05-17:
Sorry, I mis-understood your question - nevermind.