openwrt/packages #1096: ddns-scripts: dnsomatic.com is unusable

miltador, 2015-03-30:
dnsomatic.com is all-in-one ddns services updater.

To use it and update all my accounts I need to specify *all.dnsomatic.com* as Hostname/Domain.
It is not real domain and it has nothing common with ddns domains that redirect to my ip.

When I specify *all.dnsomatic.com* it tries to nslookup on it but, of course, fails and doesnt do anything else.

So it is wrong to check public IP by specified Hostname/Domain in case of dnsomatic.com usage.

Well, I can specify one of my ddns domains which direct to my IP and added to dns-o-matic.
Then nslookup will be correct but request to dnsomatic.com will be wrong cause it puts my specified domain instead of *all.dnsomatic.com*:
```
updates.dnsomatic.com/nic/update?hostname=******.duckdns.org
```
instead of:
```
updates.dnsomatic.com/nic/update?hostname=all.dnsomatic.com
```
I looked into  `packages/net/ddns-scripts/files/services` file and there is such code:
```
# dns-o-matic is a free service by opendns.com for updating multiple hosts and
# dynamic dns services in one api call. To update all your configured services
# at once, use "all.dnsomatic.com as the hostname.
"dnsomatic.com" "http://[USERNAME]:[PASSWORD]@updates.dnsomatic.com/nic/update?hostname=[DOMAIN]&myip=[IP]"
```
But, as said above, currently it is not possible to `use all.dnsomatic.com as the hostname` cause nslookup on it won't work.

If you explicitly specify `all.dnsomatic.com` in request template as hostname parameter it will not be clear what then to use as domain\hostname and not possible to do any nslookup cause dnsomatic.com doesn't offer dynamic domains itself.
I think there needs to be another kind of check for public IP, not nslookup in case of dns-o-matic.

miltador, 2015-03-30:
@chris5560, please take a look.

miltador, 2015-03-30:
As a temporary solution (not perfect at all):

I edited mine services file on router as follows:
```
"dnsomatic.com" "http://[USERNAME]:[PASSWORD]@updates.dnsomatic.com/nic/update?hostname=all.dnsomatic.com&myip=[IP]"
```
Added DuckDNS account to dns-o-matic and specified Hostname/Domain in LuCI `*****.duckdns.org`, given by DuckDNS which points to my IP and is refreshed from dns-o-matic.

chris5560, 2015-03-30:
Hi Vasiliy,

ddns-scripts is designed to update one host/provider per section/configuration.
dnsomatic service is designed to update multiple ddns provider with one configuration.

The entry inside services file updates one specific host configured at dnsomatic.

So option one is to setup multiple sections to directly update each ddns provider 
or host by host at dnsomatic ;-) 
option two is to remove "option service_name" (LuCI setting "custom") and put your url as you posted above into "option update_url" inside /etc/config/ddns.

I think ddns-scripts could not give an answer to all possibilities.

Hope I find time over the Easter weekend to update/extend
http://wiki.openwrt.org/doc/howto/ddns.client#provider_specifics

Christian

miltador, 2015-03-30:
Thanks for reply. My current configuration suits to me, maybe in fresh trunk will try to do it all in LuCI without file edits.
I understood that one ddns - one section. But I need to update opendns IP too, which also doesnt have domain\hostname thing, it easily updates through dns-o-matic.

closed 2015-03-30 by miltador