openwrt/packages #1097: tor fails to build on ar71xx

blogic, 2015-03-31:
src/common/sandbox.c:343:176: note: each undeclared identifier is reported only once for each function it appears in
src/common/sandbox.c: In function 'sigsys_debugging':
src/common/sandbox.c:1643:35: error: 'mcontext_t' has no member named 'M_SYSCALL'
   syscall = (int) ctx->uc_mcontext.M_SYSCALL;
                                   ^
make[5]: *** [src/common/sandbox.o] Error 1
make[5]: Leaving directory `/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/tor-0.2.5.10'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/tor-0.2.5.10'
make[3]: *** [/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/tor-0.2.5.10/.built] Error 2

moonman, 2015-04-07:
Not just ar71xx but ramips and probably others. Full log available here: https://olegtown.pw/openwrt/r45277/broken_packages/build_tor.txt It has been failing to build for a while now.

moonman, 2015-04-07:
Also related: #935

closed 2015-07-12 by hauke