openwrt/packages #1098: gdbm fails to build

blogic, 2015-03-31:
make[4]: Entering directory `/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/gdbm-1.11'
make  all-recursive
make[5]: Entering directory `/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/gdbm-1.11'
Making all in po
make[6]: Entering directory `/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/gdbm-1.11/po'
*** error: gettext infrastructure mismatch: using a Makefile.in.in from gettext version 0.18 but the autoconf macros are from gettext version 0.19
make[6]: *** [check-macro-version] Error 1
make[6]: Leaving directory `/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/gdbm-1.11/po'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/gdbm-1.11'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/gdbm-1.11'
make[3]: *** [/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/gdbm-1.11/.built] Error 2

thess, 2015-04-07:
Wasn't this fixed by your merge 66644f229eee8da311f215e58752a81334f694ae ?

Naoir, 2015-04-07:
Yeah, it's fixed. Since nobody has complained yet, I'll close this.

Thank you.

josvos, 2015-10-09:
This problem still occurs for me using CC's final SDK:

Naoir, 2015-10-10:
Looks like the gettext-version fixup is not working properly in CC SDK. gettext's autopoint seems to include a hard-coded path at compile-time to staging_dir/host, which points to /home/buildbot in the binary download archive.

Do you get a message like this while building?
```
$ make V=s package/gdbm/{clean,configure}
[...]
[BUILD_DIR]/staging_dir/host/bin/autopoint: 454: [BUILD_DIR]/staging_dir/host/bin/autopoint: cannot open /home/buildbot/slave-local/ar71xx_generic/build/staging_dir/host/share/gettext/archive.dir.tar.xz: No such file
tar: This does not look like a tar archive
tar: gettext-0.19.4: Not found in archive
tar: Exiting with failure status due to previous errors
autopoint: *** infrastructure files for version 0.19.4 not found; this is autopoint from GNU gettext-tools 0.19.4
autopoint: *** Stop.
[... build continues ...]
```

If so, a quick hacky workaround would be to edit [BUILD_DIR]/staging_dir/host/bin/autopoint's prefix variable to point to your real staging_dir/host directory.

Do you have any insights on this @jow? I've posted a full build log here: https://gist.github.com/Naoir/a49d89f5486b941b751d . CC final SDK ar71xx.

josvos, 2015-10-10:
Yes, that's similar to the errors I see and your workaround works.  Thanks so far!

josvos, 2015-10-10:
In my test trees I made it:

bk138, 2016-01-13:
Confirmed, having
``` 
prefix=`dirname "$STAGING_DIR"`/host
```
in autopoint fixes building with the CC SDK.

karlp, 2016-05-24:
fixed in https://github.com/openwrt/openwrt/commit/ac2d3bc7d9c5ccbb11b6a6d9490f5484bb85a04b

closed 2016-05-24 by karlp