openwrt/packages #1101: noip fails to compile

blogic, 2015-03-31:
the package looks extremly broken.

make -C /home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/noip-2.1.9-1 PREFIX="" DESTDIR="/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/noip-2.1.9-1/ipkg-install" BINDIR="/home/buildbot/slave-local/ar71xx_generic/build/build_dir/target-mips_34kc_uClibc-0.9.33.2/noip-2.1.9-1/ipkg-install/usr/bin" CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16" CC="ccache_cc -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16" -O -I/home/buildbot/slave-local/ar71xx_generic/build/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/home/buildbot/slave-local/ar71xx_generic/build/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/home/buildbot/slave-local/ar71xx_generic/build/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/buildbot/slave-local/ar71xx_generic/build/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include LD="ccache_cc -L/home/buildbot/slave-local/ar71xx_generic/build/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/home/buildbot/slave-local/ar71xx_generic/build/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/home/buildbot/slave-local/ar71xx_generic/build/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/buildbot/slave-local/ar71xx_generic/build/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib" install
make: invalid option -- 'O'

jow-, 2015-03-31:
Another problem is that the init script uses `pgrep` which is not a standard utility on base OpenWrt systems.

sbyx, 2015-03-31:
Sorry I accidentally seem to have merged the wrong PR, will remove the package again.

closed 2015-03-31 by sbyx