openwrt/packages #1113: Freeradius2 wont build, old version in feeds for Barrier Breaker

delmadord, 2015-04-02:
Greetings, sorry for bothering but I do not know what to do. I reported the issue in https://dev.openwrt.org/ticket/19387 but it got closed and referred here. Files in BB feeds for freeradius2 are outdated (pointing to version 2.2.5 which is no longer present ftp://ftp.freeradius.org/pub/freeradius/ ). Instead 2.2.6 is present on the FTP. Trunk works good, but I would like to  have it working on stable branch too. Thanks!

thess, 2015-04-07:
The source is still available in:
ftp://ftp.freeradius.org/pub/freeradius/old/freeradius-server-2.2.5.tar.bz2
-- or --
https://github.com/FreeRADIUS/freeradius-server/archive/release_2_2_5.tar.gz

I think the best strategy would be to change the source URL and avoid the upgrade for BB.

thess, 2015-05-31:
Closed by 6f36722c7312fe7316d4fea57aa5cf972a8958a8

closed 2015-05-31 by thess