openwrt/packages #1116: pulseaudio 6 playback broken

plntyk, 2015-04-03:
Discovered when trying to add bluetooth playback:

    paplay <filename.wav>

did not play audio by default.
Adding pulse-access group to root  and modifying a module parameter did help

    load-module module-native-protocol-unix auth-group=pulse-access

see http://wiki.openwrt.org/wiki/bluetooth.audio

http://wiki.openwrt.org/doc/howto/pulseaudio is outdated but mentions
a similar auth problem for network streams.
Maybe some edited system.pa should be installed instead of the default ?

Maybe there is a better solution ?

tripolar, 2015-04-03:
i don't know this as i don't have any bluetooth capable device here.  adding a group to root works because there are no access restrictions but this is bad practice and will never make it into a package i maintain. try to use strace to find out what the exact problem is.

plntyk, 2015-04-03:
I experienced the issue with unmodified pulseaudio (before I started doing anything with bt)

According to the provided .init file pulseaudio runs in system mode.
Official documentation "requires" pulse-access group if pulseaudio is run in system mode.

```
To run PulseAudio in system-wide mode, start it as root and pass the --system argument to it. It will then drop priviliges and change to the pulse UNIX user and group. The directory /var/run/pulse/ is used as home directory. In this mode the module module-native-protocol-unix will automatically allow access to all members of the group pulse-access.
```
see http://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/SystemWide/

I opened this ticket because I had a hunch of this "bad practise" problem This is similar to running pulse in system mode - but for embedded system a "user" account for pulse might not make sense.

some observations if I have the defaults:

```
pacmd
No PulseAudio daemon running, or not running as session daemon.
```
pacmd doesnt work by default nor does it with a pulse-access modification (the message seems to suggest pacmd only works in session mode)

```
pactl stat
Connection failure: Access denied
```
failure in default case - success with a pulse-access modification

```
paplay /network/.../casino.brass.wav
Connection failure: Access denied
```
So testing if pulse works is impossible from the default root account.
And with all that there is another issue: no output by default in logread ( /etc/pulse/daemon.conf default setting not working together with the procd init script log-target = syslog gets an output)

I dont know how an strace helps ( http://paste.debian.net/164817/ ), if we dont follow a "recommended" path at the moment (I read about auth cookies or other ways (like pulse-access) but these are not installed/modified by default which would be the sane to get a working system by default without much tinkering)
I dont know how/if you have a working setup with pulse with the current package state.I fully understand if you only can do compile testing ! - this bluez/pulse tinkering took ~3 days to figure out and it will take many more to get this all documented & working by default/without much manual tinkering by default.

tripolar, 2015-04-05:
i currently use pulseaudio in system mode on openwrt and stream music from remote machines. You have to configure pulseaudio by hand as all auto-detection dependecies are not included like udev support. this decreases the footprint and normally setups are static at least in my case.

i just set this in /etc/pulse/system.pa

    load-module module-alsa-sink device=hw:0

and

    load-module module-native-protocol-tcp listen=192.168.1.1 auth-ip-acl=127.0.0.1;192.168.0.0/16

with the requiered kernel drivers loaded.

if you run your pulseaudio server on a remote machine you can configure it with pavucontrol

    PULSE_SERVER=192.168.1.1 pavucontrol

The Makefile defines the pulse user, see:

    https://github.com/openwrt/packages/blob/master/sound/pulseaudio/Makefile#L40

and defines the user at configure time:

    https://github.com/openwrt/packages/blob/master/sound/pulseaudio/Makefile#L81

plntyk, 2015-04-08:
thanks for the feedback - I made a patch that creates 3 different flavours
have a look at : https://gist.github.com/plntyk/a68fc3268619cc7b0050

  - 3 because initial proof-of-concept (minimal,"full" aka previous featureset,ext)
  - tried to use ffmpeg as template
  - make testing easier (untested atm)
  - speexdsp is resampling algorithm ; dependency in mini package removed (libspeexdsp 35kB)*
  - openssl dependency is only(?) needed for RAOP/Airtunes/Airplay streaming - this is moved to ext variant
  - for each module-X-protocol" there is a libprotocol-X that has to be packaged
  - OSS and esound are removed/deactivated because they are deprecated/old/"GUI" related
- some modules have similar functionality (tunnel and native-tcp)

- ipkg size for mini/full(prev. feature) is 555kB/600kB ; ext is 811kB

tripolar, 2015-04-10:
nice work. but i see no point in providing 3 distinct packages.

I would like to have a normal "pulseaudio" package providing  the functionality pulseaudio provides now and a "pulseaudio-full" package that provides the exteneded functionality. It's easier to maintain and I think clearer when a user wants to install pulseaudio.

MichaelSp, 2015-12-26:
I'd like to have openwrt as a bluetooth a2dp sink. 
Is this the right issue to subscribe? Is there a way to have bluetooth -> alsa (w/o pulseaudio)?
Any progress here?

thess, 2017-02-17:
Closed for lack of interest (over 1yr)

closed 2017-02-17 by thess

MichaelSp, 2017-02-17:
not really lack of interest but lack of progress 😞

PraneetKaur, 2018-01-09:
Hi, 
I am trying to run pulseaudio as a system wide daemon in openwrt but I am facing challenge in adding rights in /etc/group. Can someone throw light on what is the right way to proceed?

tripolar, 2018-01-21:
@MichaelSp i'm sorry but this is not a paid support channel. if you want to add functionality i'm fine with (and i will commit it, if it works) it but i won't do the work for you.

@PraneetKaur you don't need to add someone to the pulseaudio group. if you run it in system mode. you just have to connect to the pulse audio instance

mutexstream, 2022-06-24:
I'm surprised that this issue is not yet fixed in 2022 even though it was opened in 2015!

diabl0w, 2023-07-31:
> @MichaelSp i'm sorry but this is not a paid support channel. if you want to add functionality i'm fine with (and i will commit it, if it works) it but i won't do the work for you.
> 
> > Is there a way to have bluetooth -> alsa (w/o pulseaudio)?
> 
> no idea. i don't use bluetooth but i guess so as pulseaudio also uses alsa it's just wrapper in between
> 
> @PraneetKaur you don't need to add someone to the pulseaudio group, if you run it in system mode. you just have to connect to the pulse audio instance

can you elaborate on "connect to the pulse audio instance" ... I see no way of connecting to it and being able to use commands such as `pactl` `pacmd` or `pacat`. Openwrt goes through the trouble of offering a `pulseaudio-tools` package, but there doesnt seem to be any way to actually use them

davidandreoletti, 2023-12-20:
>  I see no way of connecting to it and being able to use commands such as pactl pacmd or pacat

opkg update
opkg install sudo
sudo PULSE_RUNTIME_PATH=/var/run/pulse -u pulse pacmd stat