openwrt/packages #1118: [lighttpd] Selections not applied when compiling via SDK

yanosz, 2015-04-03:
Hello,

using 14.07's SDK on the master branch, I noticed that any selection up on models to include is not applied - for example: The systems tries to link against mysql, altough no mysql-vhost model is needed.
This causes complex builds (I need to compile mysql :-/) and is error-prone.

@MikePetullo - as far as I see, the Makefiles is more ore less yours. Can you have a look, please?

Thanks, Jan
.config for SDK (lighttpd only), generated using <code>make menuconfig</code>
<pre>
#
# Web Servers/Proxies
#
CONFIG_PACKAGE_lighttpd=m
# CONFIG_LIGHTTPD_SSL is not set
# CONFIG_PACKAGE_lighttpd-mod-access is not set
# CONFIG_PACKAGE_lighttpd-mod-accesslog is not set
# CONFIG_PACKAGE_lighttpd-mod-alias is not set
# CONFIG_PACKAGE_lighttpd-mod-auth is not set
CONFIG_PACKAGE_lighttpd-mod-cgi=m
# CONFIG_PACKAGE_lighttpd-mod-cml is not set
# CONFIG_PACKAGE_lighttpd-mod-compress is not set
# CONFIG_PACKAGE_lighttpd-mod-evasive is not set
# CONFIG_PACKAGE_lighttpd-mod-evhost is not set
# CONFIG_PACKAGE_lighttpd-mod-expire is not set
# CONFIG_PACKAGE_lighttpd-mod-extforward is not set
# CONFIG_PACKAGE_lighttpd-mod-fastcgi is not set
# CONFIG_PACKAGE_lighttpd-mod-flv_streaming is not set
# CONFIG_PACKAGE_lighttpd-mod-magnet is not set
# CONFIG_PACKAGE_lighttpd-mod-mysql_vhost is not set
CONFIG_PACKAGE_lighttpd-mod-proxy=m
# CONFIG_PACKAGE_lighttpd-mod-redirect is not set
# CONFIG_PACKAGE_lighttpd-mod-rewrite is not set
# CONFIG_PACKAGE_lighttpd-mod-rrdtool is not set
# CONFIG_PACKAGE_lighttpd-mod-scgi is not set
# CONFIG_PACKAGE_lighttpd-mod-secdownload is not set
# CONFIG_PACKAGE_lighttpd-mod-setenv is not set
# CONFIG_PACKAGE_lighttpd-mod-simple_vhost is not set
# CONFIG_PACKAGE_lighttpd-mod-ssi is not set
# CONFIG_PACKAGE_lighttpd-mod-status is not set
# CONFIG_PACKAGE_lighttpd-mod-trigger_b4_dl is not set
# CONFIG_PACKAGE_lighttpd-mod-userdir is not set
# CONFIG_PACKAGE_lighttpd-mod-usertrack is not set
# CONFIG_PACKAGE_lighttpd-mod-webdav is not set
</pre>

Errors in shell:
Call: <code>$ make package/lighttpd/compile V=99

* libxml2 is needed but not declared:
<pre>
checking for XML... no
configure: error: Package requirements (libxml-2.0) were not met:
No package 'libxml-2.0' found
Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.
Alternatively, you may set the environment variables XML_CFLAGS
and XML_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
</pre>
* After installed libxml2 - sqlite3 is need
<pre>
checking for SQLITE... no
configure: error: Package requirements (sqlite3) were not met:
No package 'sqlite3' found
Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.
Alternatively, you may set the environment variables SQLITE_CFLAGS
and SQLITE_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
[... 70 lines cut ...]
make[8]: Nothing to be done for `all'.
make[8]: Leaving directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/tests/docroot/www/expire'
make[8]: Entering directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/tests/docroot/www'
make[8]: Nothing to be done for `all-am'.
make[8]: Leaving directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/tests/docroot/www'
make[7]: Leaving directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/tests/docroot/www'
make[7]: Entering directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/tests/docroot'
make[7]: Nothing to be done for `all-am'.
make[7]: Leaving directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/tests/docroot'
make[6]: Leaving directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/tests/docroot'
make[6]: Entering directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/tests'
make[6]: Nothing to be done for `all-am'.
make[6]: Leaving directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/tests'
make[5]: Leaving directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/tests'
make[5]: Entering directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35'
make[5]: Nothing to be done for `all-am'.
make[5]: Leaving directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35'
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35'
make[2]: *** [/home/jan/freifunk/openwrt-ff-build/ar71xx-build/OpenWrt-SDK-ar71xx-for-linux-x86_64-gcc-4.8-linaro_uClibc-0.9.33.2/build_dir/target-mips_34kc_uClibc-0.9.33.2/lighttpd-1.4.35/.built] Error 2
make[2]: Leaving directory `/
#(...) cropped
</pre>

yanosz, 2015-04-03:
P.S. 
When changing
<pre>
ifneq ($(SDK)$(CONFIG_PACKAGE_lighttpd-mod-mysql-vhost),)
  CONFIGURE_ARGS+= \
        --with-mysql
else
  CONFIGURE_ARGS+= \
        --without-mysql
endif
</pre>
To
<pre>
ifneq ($(SDK)$(CONFIG_PACKAGE_lighttpd-mod-mysql-vhost),)
  CONFIGURE_ARGS+= \
        --without-mysql
else
  CONFIGURE_ARGS+= \
        --without-mysql
endif
</pre> 
MySQL is not demaned anymore. 

I know, that this is not an option (since MySQL) is not used no matter what - But I guess there's something wrong with theses conditions.

gstrauss, 2017-01-05:
@yanosz is this still an issue for you?

diizzyy, 2017-06-07:
@hnyman @thess 
Please close, submitter timeout.

closed 2017-06-07 by hnyman