openwrt/packages #1122: Freeswitch doesn't build and has been like this for a while now

moonman, 2015-04-07:
```checking for libmemcached >= 0.31... no
checking openssl/tls1.h usability... no
checking openssl/tls1.h presence... yes
configure: WARNING: openssl/tls1.h: present but cannot be compiled
configure: WARNING: openssl/tls1.h:     check for missing prerequisite headers?
configure: WARNING: openssl/tls1.h: see the Autoconf documentation
configure: WARNING: openssl/tls1.h:     section "Present But Cannot Be Compiled"
configure: WARNING: openssl/tls1.h: proceeding with the compiler's result
configure: WARNING:     ## ---------------------------------- ##
configure: WARNING:     ## Report this to bugs@freeswitch.org ##
configure: WARNING:     ## ---------------------------------- ##
checking for openssl/tls1.h... no
configure: WARNING: OpenSSL include files were not found
configure: error: OpenSSL >= 1.0.1e and associated developement headers required
Makefile:947: recipe for target '/home/oleg/openwrt/trunk/build_dir/target-mipsel_74kc+dsp2_uClibc-0.9.33.2/freeswitch-1.5.15b```
Full log available here: https://olegtown.pw/openwrt/r45277/broken_packages/build_freeswitch.txt

sbyx, 2015-04-07:
Freeswitch is not part of this feed. See https://github.com/openwrt/telephony instead

closed 2015-04-07 by sbyx

moonman, 2015-04-07:
Apologies and thanks for pointing in the right direction.