openwrt/packages #1125: building python 2.7.3  failed

RIMEH1, 2015-04-07:
2015-04-07 12:21:34 (50.9 KB/s) - written to stdout [11793433/11793433]

MD5 sum of the downloaded file does not match (file: c57477edd6d18bd9eeca2f21add73919, requested: fee5408634a54e721a93531aba37f8c1) - deleting download.
--2015-04-07 12:21:34--  http://mirror2.openwrt.org/sources/Python-2.7.3.tar.bz2
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-04-07 12:21:35 ERROR 404: Not Found.

Download failed.
--2015-04-07 12:21:35--  http://downloads.openwrt.org/sources/Python-2.7.3.tar.bz2
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-04-07 12:21:35 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[2]: *** [/home/ri/openwrt/dl/Python-2.7.3.tar.bz2] Error 2
make[2]: Leaving directory `/home/ri/openwrt/feeds/packages/lang/python'
make[1]: *** [package/feeds/packages/python/host/compile] Error 2
make[1]: Leaving directory `/home/ri/openwrt'
make: *** [package/python/compile] Error 2

commodo, 2015-04-07:
Python 2.7.3 is not supported anymore.
Please try Python 2.7.9.
(This one https://github.com/openwrt/packages/tree/master/lang/python )

Plus, it looks like your error was a bit intermittent ?
I mean, you should have been able to get the Python binary from here:
https://www.python.org/ftp/python/2.7.3/

But maybe your internet connection was down at the time ?

RIMEH1, 2015-04-07:
for python 2.7.3 or 2.7.9 is what I download the .zip this link and I decompress and puts it in the package folder and I compile or how change the version  because I  just change the version of Python 2.6 in makefile 2.7.3.
thanks

RIMEH1, 2015-04-07:
I have this result when I download 2.7.9

/home/ri/openwrt/scripts/download.pl "/home/rimeh/openwrt/dl" "Python-2.7.9.tar.bz2" "fee5408634a54e721a93531aba37f8c1" "http://www.python.org/ftp/python/2.7.9/"
--2015-04-07 12:41:35--  http://www.python.org/ftp/python/2.7.9/Python-2.7.9.tar.bz2
Resolving www.python.org (www.python.org)... 185.31.17.223
Connecting to www.python.org (www.python.org)|185.31.17.223|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://www.python.org/ftp/python/2.7.9/Python-2.7.9.tar.bz2 [following]
--2015-04-07 12:41:35--  https://www.python.org/ftp/python/2.7.9/Python-2.7.9.tar.bz2
Connecting to www.python.org (www.python.org)|185.31.17.223|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-04-07 12:41:36 ERROR 404: Not Found.

Download failed.
--2015-04-07 12:41:36--  http://mirror2.openwrt.org/sources/Python-2.7.9.tar.bz2
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-04-07 12:41:36 ERROR 404: Not Found.

Download failed.
--2015-04-07 12:41:36--  http://downloads.openwrt.org/sources/Python-2.7.9.tar.bz2
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-04-07 12:41:36 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
make[2]: *** [/home/ri/openwrt/dl/Python-2.7.9.tar.bz2] Error 2
make[2]: Leaving directory `/home/ri/openwrt/feeds/packages/lang/python'
make[1]: *** [package/feeds/packages/python/host/compile] Error 2
make[1]: Leaving directory `/home/ri/openwrt'
make: *** [package/python/compile] Error 2

commodo, 2015-04-07:
there is no  `tar.bz2` file
try  `tar.xz` : https://www.python.org/ftp/python/2.7.9/Python-2.7.9.tar.xz

RIMEH1, 2015-04-07:
I replace the python file I in the feeds/packages/lang/python  directory by it.
or exactly where  I put it after download.

commodo, 2015-04-07:
it would be great, if you could use the files under:
https://github.com/openwrt/packages/blob/master/lang/python/

however, please check if you have any conflicts with other local `python` packages;
the behaviour of packages is not very deterministic if you have 2 (or more) `python` packages in your build;
you should also try a clean checkout, or try a clean build

RIMEH1, 2015-04-07:
I do not know if the system version that I use has an effect on the compilation because the python that offers me several files that do not exist in the file I are former python.il not a makefile and compilation failed as usual. :(

closed 2015-04-27 by sbyx