openwrt/packages #1126: aria2 error

taeasy, 2015-04-07:
make[6]: Nothing to be done for `all-am'.
make[6]: Leaving directory `/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7/deps'
make[5]: Leaving directory `/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7/deps'
Making all in src
make[5]: Entering directory `/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7/src'
Making all in includes
make[6]: Entering directory `/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7/src/includes'
make[6]: Nothing to be done for `all'.
make[6]: Leaving directory `/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7/src/includes'
make[6]: Entering directory `/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7/src'
depbase=`echo LibuvEventPoll.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
	/bin/bash ../libtool  --tag=CXX   --mode=compile x86_64-openwrt-linux-uclibc-g++ -DHAVE_CONFIG_H -I. -I..  -Wall -I../lib -I../intl -I./includes -I./includes -DLOCALEDIR=\"/usr/share/locale\" -DHAVE_CONFIG_H  -I../deps/wslay/lib/includes -I../deps/wslay/lib/includes -I/home/main/64/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include  -I/home/main/64/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include  -I/home/main/64/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include -I/home/main/64/staging_dir/target-x86_64_uClibc-0.9.33.2/include -I/home/main/64/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/64/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include   -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable  -pipe -std=c++11  -MT LibuvEventPoll.lo -MD -MP -MF $depbase.Tpo -c -o LibuvEventPoll.lo LibuvEventPoll.cc &&\
	mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  x86_64-openwrt-linux-uclibc-g++ -DHAVE_CONFIG_H -I. -I.. -Wall -I../lib -I../intl -I./includes -I./includes -DLOCALEDIR=\"/usr/share/locale\" -DHAVE_CONFIG_H -I../deps/wslay/lib/includes -I../deps/wslay/lib/includes -I/home/main/64/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include -I/home/main/64/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include -I/home/main/64/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include -I/home/main/64/staging_dir/target-x86_64_uClibc-0.9.33.2/include -I/home/main/64/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/64/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -pipe -std=c++11 -MT LibuvEventPoll.lo -MD -MP -MF .deps/LibuvEventPoll.Tpo -c LibuvEventPoll.cc  -fPIC -DPIC -o .libs/LibuvEventPoll.o
LibuvEventPoll.cc: In member function 'virtual void aria2::LibuvEventPoll::poll(const timeval&)':
LibuvEventPoll.cc:144:59: error: invalid conversion from 'void (*)(uv_timer_t*, int) {aka void (*)(uv_timer_s*, int)}' to 'uv_timer_cb {aka void (*)(uv_timer_s*)}' [-fpermissive]
     uv_timer_start(timer, timer_callback, timeout, timeout);
                                                           ^
In file included from LibuvEventPoll.h:43:0,
                 from LibuvEventPoll.cc:44:
/home/main/64/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include/uv.h:760:15: error:   initializing argument 2 of 'int uv_timer_start(uv_timer_t*, uv_timer_cb, uint64_t, uint64_t)' [-fpermissive]
 UV_EXTERN int uv_timer_start(uv_timer_t* handle,
               ^
make[6]: *** [LibuvEventPoll.lo] Error 1
make[6]: Leaving directory `/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7/src'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7/src'
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory `/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7'
make[2]: *** [/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7/.built] Error 2
make[2]: Leaving directory `/home/main/64/feeds/packages/net/aria2'
make[1]: *** [package/feeds/packages/aria2/compile] Error 2
make[1]: Leaving directory `/home/main/64'
make: *** [package/aria2/compile] 错误 2
main@bogon:~/64$ make package/aria2/compile V=99 > /home/main/output.txt
configure: WARNING: using cross tools not prefixed with host triplet
configure: WARNING: result yes guessed because of cross compilation
configure: WARNING: using cross tools not prefixed with host triplet
libtool: link: warning: `-version-info/-version-number' is ignored for convenience libraries
LibuvEventPoll.cc: In member function 'virtual void aria2::LibuvEventPoll::poll(const timeval&)':
LibuvEventPoll.cc:144:59: error: invalid conversion from 'void (*)(uv_timer_t*, int) {aka void (*)(uv_timer_s*, int)}' to 'uv_timer_cb {aka void (*)(uv_timer_s*)}' [-fpermissive]
     uv_timer_start(timer, timer_callback, timeout, timeout);
                                                           ^
In file included from LibuvEventPoll.h:43:0,
                 from LibuvEventPoll.cc:44:
/home/main/64/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include/uv.h:760:15: error:   initializing argument 2 of 'int uv_timer_start(uv_timer_t*, uv_timer_cb, uint64_t, uint64_t)' [-fpermissive]
 UV_EXTERN int uv_timer_start(uv_timer_t* handle,
               ^
make[6]: *** [LibuvEventPoll.lo] Error 1
make[5]: *** [all-recursive] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all] Error 2
make[2]: *** [/home/main/64/build_dir/target-x86_64_uClibc-0.9.33.2/aria2-1.18.7/.built] Error 2
make[1]: *** [package/feeds/packages/aria2/compile] Error 2
make: *** [package/aria2/compile] Error 2

plntyk, 2015-04-07:
there is a patch available for those who cannot wait : https://github.com/openwrt/packages/pull/1011

plntyk, 2015-04-08:
duplicate #1019

closed 2015-04-09 by taeasy