openwrt/packages #1127: vnstat fails to compile with musl

avbohemen, 2015-04-08:
vnstat fails to compile with musl. I tried both the current version (1.12) as well as 1.13. Both give same error. Log:
```
mips-openwrt-linux-musl-gcc -Os -pipe -mno-branch-likely -march=74kc -mtune=74kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16   -c -o dbaccess.o dbaccess.c
dbaccess.c: In function 'backupdb':
dbaccess.c:268:25: error: 'O_RDONLY' undeclared (first use in this function)
  if ((c = open(current, O_RDONLY)) == -1) {
                         ^
dbaccess.c:268:25: note: each undeclared identifier is reported only once for each function it appears in
make[5]: *** [dbaccess.o] Error 1
```

toofishes, 2015-06-17:
Easy fix, drop this in `patches/004-fix-musl-fcntl-h.patch` or similar. I can turn this into a pull request if that makes sense and is the proper thing to do to get this resolved.

~~~
--- a/src/common.h
+++ b/src/common.h
@@ -10,6 +10,7 @@
 #include <string.h>
 #include <sys/types.h>
 #include <sys/stat.h>
+#include <fcntl.h>
 #include <dirent.h>
 #include <ctype.h>
 #include <signal.h>
~~~

papaj0e, 2015-06-17:
I already did the pull request just a while ago :+1:

closed 2015-07-27 by champtar