openwrt/packages #1129: nginx musl problem

lukas2511, 2015-04-08:
The nginx package doesn't build against musl, missing sys/sysctl.h and something about salts, no idea.

I found these patches and tried them (excluding the zlib patch since it wasn't required): https://github.com/mwcampbell/docker-nginx/tree/master/patches

Nginx does compile with those patches and seems to work completely fine.

heil, 2015-04-08:
do you have more information? e.g what plattform, what revision etc.?

lukas2511, 2015-04-08:
Building this with gcc 4.8.x linaro and musl for soc.allwinner.sunxi / Bananapi, but I don't think platform matters since this seems to be just a platform independent problem with nginx and musl.

I'm building the newest trunk version and nginx feed seems to come from this repository, checked out today.

nekromant, 2015-06-21:
I can confirm this on Sunxi A20 Lamobo R1 board.

heil, 2015-06-21:
Please send a dedicated PR that fixes building with musl.

heil, 2015-06-21:
fixed via 9433d5ad285056c9476a899595df6d4785ec127e

heil, 2015-06-22:
@lukas2511  are you fine with that?

lukas2511, 2015-06-22:
didn't have time to test it yet, but looks about right

nekromant, 2015-06-22:
@heil I've tested it yesterday night - nginx now compiles and even works! Thanks a lot! 

P.S. There's still what looks like a musl-related problem with etherwake with some struct redefined, I'll be investigating it today and maybe throw up a patch.

Noltari, 2015-06-22:
@nekromant that's already fixed and waiting for maintainer: https://github.com/openwrt/packages/issues/1381

nekromant, 2015-06-22:
@Noltari Oh, great! Thanks!

karlp, 2015-07-19:
as far as I can tell, all issues reported here are now fixed, ==> closing this ticket.

closed 2015-07-19 by karlp