openwrt/packages #1130: octeon missing libmysqlclient-r *  and libmysqlclient_r * packages

aigarslv, 2015-04-08:
Octenon has no libmysqlclient-r *  and libmysqlclient_r packages, so strongswan-full cannot be installed.

root@OpenWrt:~# opkg install strongswan-full
Installing strongswan-full (5.2.2-1) to root...
Downloading http://downloads.openwrt.org/barrier_breaker/14.07/octeon/generic/packages/packages/strongswan-full_5.2.2-1_octeon.ipk.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for strongswan-full:
 *      libmysqlclient-r *
 * opkg_install_cmd: Cannot install package strongswan-full.

plntyk, 2015-04-08:
you can try trunk which has libmysqlclient-r for octeon, see
https://downloads.openwrt.org/snapshots/trunk/octeon/generic/packages/packages/

PS: afaik octeon arch suffers from 64bit ABI issues in the toolchain or with some packages

This leads to many broken packages on that arch see: http://buildbot.openwrt.org:8010/broken_packages/octeon/

apparently mysql was not successfully built in 14.07 release for octeon

hnyman, 2016-08-23:
14.07 is unsupported, octeon has the package in trunk and 15.05

closed 2016-08-23 by hnyman