openwrt/packages #1132: Error building MySQL for brcm2708 (CC trunk) on Mac OS X 10.10.3

jammmet, 2015-04-09:
I'm trying to build mysql-5.1.73 (with hard FP support for Raspberry Pi) and get this error:

```g++ -I /Volumes/OpenWrt/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_uClibc-0.9.33.2_eabi/mysql-5.1.73/include -o /Volumes/OpenWrt/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_uClibc-0.9.33.2_eabi/mysql-5.1.73/sql/gen_lex_hash /Volumes/OpenWrt/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_uClibc-0.9.33.2_eabi/mysql-5.1.73/sql/gen_lex_hash.cc
In file included from /Volumes/OpenWrt/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_uClibc-0.9.33.2_eabi/mysql-5.1.73/sql/gen_lex_hash.cc:82:
/Volumes/OpenWrt/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_uClibc-0.9.33.2_eabi/mysql-5.1.73/include/my_global.h:470:10: fatal error: 
      'crypt.h' file not found
#include <crypt.h>
         ^
1 error generated.
make[3]: *** [/Volumes/OpenWrt/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_uClibc-0.9.33.2_eabi/mysql-5.1.73/.built] Error 1
make[3]: Leaving directory `/Volumes/OpenWrt/openwrt/feeds/packages/utils/mysql'```

Any ideas?

plntyk, 2015-04-13:
Maybe some toolchain/OSX issue. Building mysql with hardfloat on Linux seem to work (at least on debian)

- check if there is a crypt.h in staging_dir/ (it should been created there)
- paste the contents config.log from build_dir/target..../mysql-5.1.73/ into gist/pastebin and post the link

jammmet, 2015-04-13:
crypt.h exists in

```
staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/include
```

but not in to target (or host)

config.log here:

https://gist.githubusercontent.com/jammmet/090c47d9da5f0c17df3c/raw/gistfile1.txt

jwhitehorn, 2016-06-12:
I'm getting this same error on OS X 10.11.4, compiling for the Raspberry Pi as well.

@jammmet I know it's over a year later, but did you by chance overcome this?

giveup, 2016-11-17:
Same issue on macOS 10.12

greekstreet, 2017-01-05:
Same issue here.

lucize, 2017-04-22:
did you tried #4221 ?

diizzyy, 2017-06-02:
@jwhitehorn @giveup @greekstreet 
Are you seeing this also on LEDE?
https://lede-project.org/

ibnbd, 2017-09-01:
September 2017 and same issue, tried MariaDB but mirrors are dead.

lucize, 2017-09-01:
https://downloads.mariadb.org/mariadb/5.5.56/

ibnbd, 2017-09-01:
@lucize  how to compile this with LEDE?

lucize, 2017-09-01:
```
git clone https://git.lede-project.org/source.git
cd source
./scripts/feeds update -a
./scripts/feeds install -a
cd feeds/packages
git pull origin pull/4221/head
ln -s ../../../feeds/packages/utils/mariadb ../../package/feeds/packages/
cd ../..
make menuconfig

```

ibnbd, 2017-09-01:
@lucize  just wish this works .... I am so excited about giving it try. Thank You

lucize, 2017-09-01:
I don't own an Mac so I can't promise it will work, I didn't test it on OS X

ibnbd, 2017-09-01:
@lucize `fatal: refusing to merge unrelated histories` it's not merging :(

lucize, 2017-09-01:
ok let me try from scratch. but is saying that `This branch has no conflicts with the base branch`

ibnbd, 2017-09-01:
@lucize well, thank you mate. I can compile MySQL with ubuntu, but doesn't work with OSX. Guess I will stuck with ubuntu for compiling image.

lucize, 2017-09-01:
```
 * branch              refs/pull/4221/head -> FETCH_HEAD
fatal: refusing to merge unrelated histories

```

I'll try a PR update, as 5.5.57 is out

lucize, 2017-09-01:
updated, resynced but in won't allow me to git pull, don't know why (not so git master user)
unpack de zip, there is a` tar.xz`, unpack it into `feeds/packages/utils`
and after that from `feeds/packages` folder run

```
ln -s ../../../feeds/packages/utils/mariadb ../../package/feeds/packages/
cd ../..
make menuconfig
```

[mariadb.zip](https://github.com/openwrt/packages/files/1271432/mariadb.zip)

lucize, 2017-09-01:
you can also download it as a patch and apply it
wget https://github.com/openwrt/packages/pull/4221.patch

ibnbd, 2017-09-02:
I encountered this error 

https://gist.github.com/ibnbd/d532eafc503670698fa96237509d1305

lucize, 2017-09-02:
maybe the tools are too old, unfortunately no record of what line is causing it
can you try to add `--with-gnu-as` argument to make ?
```
         +$(MAKE) $(PKG_JOBS) -C "$(PKG_BUILD_DIR)" \
 		SUBDIRS="include" \
 		DESTDIR="$(PKG_INSTALL_DIR)"
```

ibnbd, 2017-09-02:
@lucize sure thing Captain

lucize, 2017-09-02:
or you could ask for a more professional help on 
https://launchpad.net/~maria-developers
https://launchpad.net/~maria-discuss

ibnbd, 2017-09-02:
Actually I tried with my remote Ubuntu as well. And it also failed. at 58%, while OSX failed at 7%

lucize, 2017-09-02:
on what target ?
I only tested it on x86 under ubuntu 16 and 17

ibnbd, 2017-09-02:
`ramips`
`mt7621`
It's a Netgear router, I think this is the problem. All I wanted to build MySQL from my OSX, which I can do from a remote Ubuntu server. I will discontinue that ubuntu soon, so I wanted to use my OSX for updates.

lucize, 2017-09-02:
yeah, I'll change to glibc to see if is a musl problem or higher version of gcc

```
[ 64%] Linking CXX executable explain_filename-t
../../mysys/libmysys.a(my_context.c.o): In function `my_context_spawn_internal':
my_context.c:(.text+0x3c): undefined reference to `setcontext'
my_context.c:(.text+0x40): undefined reference to `setcontext'
../../mysys/libmysys.a(my_context.c.o): In function `my_context_continue':
my_context.c:(.text+0xa4): undefined reference to `swapcontext'
my_context.c:(.text+0xc4): undefined reference to `swapcontext'
../../mysys/libmysys.a(my_context.c.o): In function `my_context_spawn':
my_context.c:(.text+0x140): undefined reference to `getcontext'
my_context.c:(.text+0x15c): undefined reference to `getcontext'
my_context.c:(.text+0x180): undefined reference to `makecontext'
my_context.c:(.text+0x1c0): undefined reference to `makecontext'
../../mysys/libmysys.a(my_context.c.o): In function `my_context_yield':
my_context.c:(.text+0x204): undefined reference to `swapcontext'
my_context.c:(.text+0x218): undefined reference to `swapcontext'
collect2: error: ld returned 1 exit status
unittest/sql/CMakeFiles/explain_filename-t.dir/build.make:111: recipe for target 'unittest/sql/explain_filename-t' failed
make[6]: *** [unittest/sql/explain_filename-t] Error 1

```

lucize, 2017-09-02:
so using glibc instead of musl solves the problem, under ubuntu

ibnbd, 2017-09-02:
@lucize where I have to change to make it work on ubuntu. Sorry I am not that expert with codes :)

lucize, 2017-09-02:
make menuconfig
advanced configuration options
toolchain options
c library implementation
(alsop you could test with newer gcc versions)

ibnbd, 2017-09-02:
@lucize 

`Collected errors:
 * check_conflicts_for: The following packages conflict with libmariadb:
 * check_conflicts_for: 	libmysqlclient * 
 * opkg_install_cmd: Cannot install package libmariadb.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mariadb-server:
 * 	mariadb * 
 * opkg_install_cmd: Cannot install package mariadb-server.
package/Makefile:64: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/parallels/source'
package/Makefile:106: recipe for target '/home/parallels/source/staging_dir/target-mipsel_24kc_glibc/stamp/.package_install' failed
make[1]: *** [/home/parallels/source/staging_dir/target-mipsel_24kc_glibc/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/parallels/source'
/home/parallels/source/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2`

lucize, 2017-09-03:
@ibnbd I received by mail the conflict between libmariadb and libmysql, did you unchecked the mysql lib? all is well ?

ibnbd, 2017-09-03:
@lucize not actually, I had to give up. I am using PHP PDO (MySQL) which automatic selects libmysql. And having libmysql selected libmariadb not installing.

lucize, 2017-09-03:
it can be set for libmariadb to `PROVIDES:=libmysql` so it thinks that is libmysql,I only set it up as libmysqlclient

ibnbd, 2017-09-03:
@lucize you are an amazing author 👍

lucize, 2017-09-03:
I see in the mail that
```
satisfy_dependencies_for: Cannot satisfy the following dependencies for mariadb-server:
mariadb *
```
there was a typo `DEPENDS:=+mariadb`, but I repaired it quickly, without the +, but I think that you downloaded the + version
I'll make a test with php pdo

lucize, 2017-09-03:
So I corrected the dependencies to mariadb and some dev lib files from staging tree to be compatible with mysql dev location but php is selecting the mysql version despite that I provided the libmysqlclient from mariadb.
It can still be used if you change the deps for php to libmariadb in `feeds/packages/lang/php7` there is a Makefile, replace the `libmysqlclient` with `libmariadb` (two times for mysqli and mysql)
I don't know how PROVIDES can have priority over a package of the same name at build time or in menuconfig.
Even if I first select libmariadb that provides libmysqlclient after selecting php-pdo-mysql, libmysqlclient from mysql will be selected
Maybe @jow- can elucidate on it 
please overwrite the Makefile of mariadb with the new one from the PR

jammmet, 2017-09-04:
Is that 20MB for `mariadb-server_5.5.57-1_mipsel_24kc.ipk`? I would love to use it, but compared to `mysql-server_5.1.73-2_mipsel_24kc.ipk` at 2.7MB, it's simply too large for my 16MB device.

lucize, 2017-09-04:
mysqld has ~9MB unpacked, didn't made a file compare against mysql-5.1, I'll see if I can make a lite version, without all the stuff

lucize, 2017-09-04:
I stripped down to ~6MB unpacked for mysqld and I think more is possible

ibnbd, 2017-09-04:
@lucize I modified the php7 make file still this error coming

`Collected errors:
 * check_conflicts_for: The following packages conflict with libmariadb:
 * check_conflicts_for: 	libmysqlclient * 
 * opkg_install_cmd: Cannot install package libmariadb.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mariadb-server:
 * 	mariadb * 
 * opkg_install_cmd: Cannot install package mariadb-server.
 * check_conflicts_for: The following packages conflict with libmariadb:
 * check_conflicts_for: 	libmysqlclient * 
 * opkg_install_cmd: Cannot install package php7-mod-mysqli.
 * check_conflicts_for: The following packages conflict with libmariadb:
 * check_conflicts_for: 	libmysqlclient * 
 * opkg_install_cmd: Cannot install package php7-mod-pdo-mysql.
package/Makefile:64: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/ahstanin/source'
package/Makefile:106: recipe for target '/home/ahstanin/source/staging_dir/target-mipsel_24kc_glibc/stamp/.package_install' failed
make[1]: *** [/home/ahstanin/source/staging_dir/target-mipsel_24kc_glibc/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/ahstanin/source'
/home/ahstanin/source/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2`

lucize, 2017-09-04:
are you sure you've overwritten the Makefile ? because you still have mariadb * dependency and in libs, make sure you don't have libmysqlclient selected, I made a clean compile with php7-pdo-mysql modified and it worked.
That's why a made some install modification for devel files because something php searched for the libs in another place
try removing all files from tmp dir (`rm -fr .*` and the floder from there) maybe the Makefile is cached

ibnbd, 2017-09-04:
@lucize can you please send the latest mariadb zip file? I trying to get the pull 4221, but git having problem with patching that. Thanks

lucize, 2017-09-04:
remove the mariadb floder from utils, go back to the parent folder

```
wget https://github.com/openwrt/packages/pull/4221.patch
patch  -p1  < 4221.patch
rm -f < 4221.patch
```

lucize, 2017-09-04:
anyway I'm preparing an option for lite package if you don't need all database engines integrated and plugins

ibnbd, 2017-09-04:
@lucize that will be great as because I will run it on my router.

ibnbd, 2017-09-04:
@lucize any idea where this error can cause?

`checking for specified location of the MySQL UNIX socket... no
checking for mysql_set_server_option in -lmysqlclient... no
configure: error: wrong mysql library version or lib not found. Check config.log for more information.
Makefile:556: recipe for target '/home/ahstanin/source/build_dir/target-mipsel_24kc_glibc/php-7.1.8/.configured_17deca3331cf96d6c88f34172b3577f7' failed
make[3]: *** [/home/ahstanin/source/build_dir/target-mipsel_24kc_glibc/php-7.1.8/.configured_17deca3331cf96d6c88f34172b3577f7] Error 1
make[3]: Leaving directory '/home/ahstanin/source/feeds/packages/lang/php7'
package/Makefile:109: recipe for target 'package/feeds/packages/php7/compile' failed
make[2]: *** [package/feeds/packages/php7/compile] Error 2
make[2]: Leaving directory '/home/ahstanin/source'
package/Makefile:105: recipe for target '/home/ahstanin/source/staging_dir/target-mipsel_24kc_glibc/stamp/.package_compile' failed
make[1]: *** [/home/ahstanin/source/staging_dir/target-mipsel_24kc_glibc/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ahstanin/source'
/home/ahstanin/source/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2`

lucize, 2017-09-04:
you need to remove the mysql version from staging tree, and maybe clean the php too
```
make package/mysql/clean
make package/php7/clean
```
and be sure that to erase mysql dir from
`staging_dir/target-ramips_24kc_glibcl/usr/lib`

lucize, 2017-09-04:
@jammmet I added a lite option somehow based on mysql original Makefile
didn't had time to verify the functionality 

```
-rw-r--r-- 1 build build  186101 Sep  4 20:26 libmariadb_5.5.57-1_x86_64.ipk
-rw-r--r-- 1 build build  448553 Sep  4 20:26 mariadb-client_5.5.57-1_x86_64.ipk
-rw-r--r-- 1 build build  984969 Sep  4 20:26 mariadb-client-extra_5.5.57-1_x86_64.ipk
-rw-r--r-- 1 build build 4349105 Sep  4 20:26 mariadb-server_5.5.57-1_x86_64.ipk

```

ibnbd, 2017-09-08:
@lucize I had a new ubuntu installation and everything was new. I faced this error during compiling.

After newly installed ubuntu I installed building dependencies. I cloned source and patched your mariadb. Then I modified my php7 makefile, changed `libmysqlclient` with `libmariadb`. On `make menuconfig` changed `C library implementation` selected mariadb and others. `libmysqlclient` wasn't selected.

`checking for MySQL UNIX socket location... no
checking for mysql_config... /home/ahstanin/source/staging_dir/target-mipsel_24kc_glibc/usr/bin/mysql_config
checking for mysql_commit in -lmysqlclient... no
checking for mysql_commit in -lmysqlclient... (cached) no
configure: error: PDO_MYSQL configure failed, MySQL 4.1 needed. Please check config.log for more information.
Makefile:556: recipe for target '/home/ahstanin/source/build_dir/target-mipsel_24kc_glibc/php-7.1.9/.configured_8a46f42dcdb60136acb961c20740b0ae' failed
make[3]: *** [/home/ahstanin/source/build_dir/target-mipsel_24kc_glibc/php-7.1.9/.configured_8a46f42dcdb60136acb961c20740b0ae] Error 1
make[3]: Leaving directory '/home/ahstanin/source/feeds/packages/lang/php7'
package/Makefile:109: recipe for target 'package/feeds/packages/php7/compile' failed
make[2]: *** [package/feeds/packages/php7/compile] Error 2
make[2]: Leaving directory '/home/ahstanin/source'
package/Makefile:105: recipe for target '/home/ahstanin/source/staging_dir/target-mipsel_24kc_glibc/stamp/.package_compile' failed
make[1]: *** [/home/ahstanin/source/staging_dir/target-mipsel_24kc_glibc/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ahstanin/source'
/home/ahstanin/source/include/toplevel.mk:207: recipe for target 'world' failed  `

lucize, 2017-09-08:
do you have files in staging_dir/target-mipsel_24kc_glibc/usr/lib/mysql, especially `libmysqlclient.a, libmysqlservices.a` ?

ibnbd, 2017-09-08:
I followed your old comment and ran this 

`make package/mysql/clean`
`make package/php7/clean`

And cleared mysql from staging_dir/target-mipsel_24kc_glibc/usr/lib

And ran the make again and same error but this time there is no mysql folder on that location.

lucize, 2017-09-08:
But did you re-download the patch? In the last commit I addressed this issue

ibnbd, 2017-09-08:
@lucize yes, everything was fresh. Even the ubuntu installation was new.

lucize, 2017-09-08:
Can you confirm that you have this 

$(CP) $(PKG_INSTALL_DIR)/usr/lib/lib*.a $(1)/usr/lib/mysql 

entry in te mariadb Makefile?

ibnbd, 2017-09-08:
@lucize sorry for late response, I found this lines there.

`$(CP) $(PKG_BUILD_DIR)/include/mysqld_error.h $(1)/usr/include/mysql/
        $(CP) $(PKG_INSTALL_DIR)/usr/lib/lib*.a $(1)/usr/lib/
        $(CP) $(PKG_INSTALL_DIR)/usr/lib/lib*.a $(1)/usr/lib/mysql
        $(CP) $(PKG_INSTALL_DIR)/usr/share/aclocal/mysql.m4 $(1)/usr/share/aclocal/`

The error happening whiling compiling php, is this because we changed the php makefile?

lucize, 2017-09-09:
your patch is not up to date!
the latest PR has this:
```
+define Build/InstallDev
 +	$(INSTALL_DIR) $(2)/bin $(1)/usr/bin $(1)/usr/include $(1)/usr/lib $(1)/usr/lib/mysql $(1)/usr/share/aclocal
 +	$(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/mysql_config $(1)/usr/bin/
 +	ln -sf $(STAGING_DIR)/usr/bin/mysql_config $(2)/bin/
 +	$(CP) $(PKG_INSTALL_DIR)/usr/include/mysql $(1)/usr/include/
 +	# NOTE: needed for MySQL-Python
 +	$(CP) $(PKG_BUILD_DIR)/include/mysqld_error.h $(1)/usr/include/mysql/
 +	$(CP) $(PKG_INSTALL_DIR)/usr/lib/lib*.a $(1)/usr/lib/
 +	$(CP) $(PKG_INSTALL_DIR)/usr/lib/lib*.a $(1)/usr/lib/mysql
 +	$(CP) $(PKG_INSTALL_DIR)/usr/share/aclocal/mysql.m4 $(1)/usr/share/aclocal/
 +endef
```

ibnbd, 2017-09-09:
@lucize I run bellow code, I thought it will get the latest patch.

``wget https://github.com/openwrt/packages/pull/4221.patch
patch  -p1  < 4221.patch
rm -f < 4221.patch``

ibnbd, 2017-09-09:
```
define Build/InstallDev
        $(INSTALL_DIR) $(2)/bin $(1)/usr/bin $(1)/usr/include $(1)/usr/lib $(1)/usr/lib/mysql $(1)/usr/share/aclocal
        $(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/bin/mysql_config $(1)/usr/bin/
        ln -sf $(STAGING_DIR)/usr/bin/mysql_config $(2)/bin/
        $(CP) $(PKG_INSTALL_DIR)/usr/include/mysql $(1)/usr/include/
        # NOTE: needed for MySQL-Python
        $(CP) $(PKG_BUILD_DIR)/include/mysqld_error.h $(1)/usr/include/mysql/
        $(CP) $(PKG_INSTALL_DIR)/usr/lib/lib*.a $(1)/usr/lib/
        $(CP) $(PKG_INSTALL_DIR)/usr/lib/lib*.a $(1)/usr/lib/mysql
        $(CP) $(PKG_INSTALL_DIR)/usr/share/aclocal/mysql.m4 $(1)/usr/share/aclocal/
endef
```

@lucize Looks like same to me

lucize, 2017-09-09:
But you need to delete de mariadb folder from utils before applying the patch

[28 more replies]