openwrt/packages #1137: ocserv error

taeasy, 2015-04-11:
pcl.c:287:2: warning: implicit declaration of function 'memcpy' [-Wimplicit-function-declaration]
  memcpy(&ctx_creating_sigs, &osigs, sizeof(sigset_t));
  ^
pcl.c:287:2: warning: incompatible implicit declaration of built-in function 'memcpy' [enabled by default]
pcl.c: In function 'co_create':
pcl.c:383:17: warning: unused variable 'r' [-Wunused-variable]
  int alloc = 0, r = CO_STK_COROSIZE;
                 ^
pcl.c: At top level:
pcl.c:60:12: warning: 'co_ctx_stackdir' defined but not used [-Wunused-function]
 static int co_ctx_stackdir(void)
            ^
  CC       pcl_version.o
  CC       pcl_private.o
  AR       libpcl.a
make[7]: Leaving directory `/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/ocserv-0.10.2/src/pcl'
make[7]: Entering directory `/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/ocserv-0.10.2/src'
/bin/true ocserv-args.def
  CC       ocserv-args.o
  AR       libcmd-ocserv.a
/bin/true ocpasswd-args.def
  CC       ocpasswd-args.o
  AR       libcmd-ocpasswd.a
  CC       ocpasswd.o
  CCLD     ocpasswd
  CC       occtl.o
  CC       occtl-pager.o
  CC       occtl-time.o
  CC       occtl-cache.o
  CC       occtl-ip-cache.o
  CC       occtl-nl.o
  CC       ctl.pb-c.o
  CC       ccan/hash/hash.o
  CC       ccan/htable/htable.o
  CC       ccan/list/list.o
  CC       ccan/talloc/talloc.o
  CC       common.o
  CC       system.o
  CC       setproctitle.o
  CC       occtl-unix.o
  CCLD     occtl
  CC       main.o
  CC       main-auth.o
  CC       worker-vpn.o
  CC       worker-auth.o
  CC       tlslib.o
  CC       cookies.o
  CC       main-misc.o
  CC       ip-lease.o
  CC       log.o
  CC       tun.o
  CC       config.o
  CC       worker-resume.o
  CC       main-resume.o
  CC       worker-extras.o
  CC       html.o
  CC       worker-http.o
  CC       main-user.o
  CC       worker-misc.o
  CC       route-add.o
  CC       worker-privs.o
  CC       sec-mod.o
  CC       sec-mod-db.o
  CC       sec-mod-auth.o
sec-mod-auth.c:56:28: fatal error: gssapi/gssapi.h: No such file or directory
 # include <gssapi/gssapi.h>
                            ^
compilation terminated.
make[7]: *** [sec-mod-auth.o] Error 1
make[7]: Leaving directory `/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/ocserv-0.10.2/src'
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory `/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/ocserv-0.10.2/src'
make[5]: *** [all] Error 2
make[5]: Leaving directory `/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/ocserv-0.10.2/src'
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory `/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/ocserv-0.10.2'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/ocserv-0.10.2'
make[2]: *** [/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/ocserv-0.10.2/.built] Error 2
make[2]: Leaving directory `/home/main/x86/feeds/packages/net/ocserv'
make[1]: *** [package/feeds/packages/ocserv/compile] Error 2
make[1]: Leaving directory `/home/main/x86'
make: *** [package/ocserv/compile]  Error 2
main@bogon:~/x86$

LazyZhu, 2015-04-14:
Missing `krb5`: https://github.com/openwrt/packages/tree/master/net/krb5

plntyk, 2015-04-14:
that krb5 (build?) dependency is not in the ocserv makefile

or one can use
```
--without-gssapi
```
in CONFIGURE_ARGS

closed 2015-04-16 by taeasy