openwrt/packages #1138: failed compilation

RIMEH1, 2015-04-11:
I want to build my own package openwrt I realized my Makefile but when compiling I found mistakes.
Note to my script I import some telque packages netaddr nmap python airmon-ng ... there are packages that I have already compiled and are existed in other telques netaddr SDK but no they do not exist.
~/openwrt$ make package/monotoring/compile V=99
make[1]: Entering directory `/home/ri/openwrt'
make[1]: *** No rule to make target `package/monotoring/compile'.  Stop.
make[1]: Leaving directory `/home/ri/openwrt'
make: *** [package/monotoring/compile] Error 2

plntyk, 2015-04-13:
This is the wrong place to ask such questions.
Your problem seems to be with your package.
This issue tracker is for OpenWrt packages.

please use https://forum.openwrt.org/

Use the OpenWrt forum

thess, 2015-04-23:
Same answer as #1180 - please use forums for support if your problem is not an issue in this repo

closed 2015-04-23 by thess