openwrt/packages #1140: PostgreSQL: Not starting (su issue)

BlueMax, 2015-04-12:
PostgreSQL (pgsql-server_9.0.17-1_x86.ipk) is not starting.
Its intended to be invoked and stopped with `su` to change user context but fails to do so.

> OpenWrt ~ $ /usr/bin/su ${pguser} -c "${pgctl} stop -D '${pgdata}' -s -m fast -w"
su: Cannot drop the controlling terminal

2) The config default points to an invalid (non-existant) folder.
> option PGDATA	/var/postgresql/data
	option PGLOG	/var/postgresql/data/postgresql.log

thess, 2015-05-31:
@BlueMax - If, after creating the /var/postgresql/data path or changing PGDATA/PGLOG options, postgresql starts OK, you should close this issue. AFAICT, the current init script issues a warning.(check system log).

Also, you may want to propose a change, via pull-request, for the init script to automatically create this path.

Is there a 2nd bug here? I only see 2) and no 1)

BlueMax, 2015-05-31:
I cannot check it atm so i close it for now.

closed 2015-05-31 by BlueMax