openwrt/packages #1143: net/krb5 - verto.h missing

db260179, 2015-04-16:
Hi,

When compiling, you get 'include verto.h is missing' error.

Please add the the following to Makefile to fix this issue

$(CP) $(PKG_INSTALL_DIR)/usr/include \
                $(1)/usr/include

So it looks like
define Build/InstallDev
        $(INSTALL_DIR) $(1)/usr/include
        $(CP) $(PKG_INSTALL_DIR)/usr/include \
                $(1)/usr/include/krb5
        $(CP) $(PKG_INSTALL_DIR)/usr/include \
                $(1)/usr/include
        $(INSTALL_DIR) $(1)/usr
        $(CP) $(PKG_INSTALL_DIR)/usr/lib \
                $(1)/usr
        rm -f $(1)/usr/lib/libcom_err*
endef

Thanks

plntyk, 2015-04-17:
Do you mean that error:
```
In file included from net-server.c:60:0:
../../include/net-server.h:32:19: fatal error: verto.h: No such file or directory
 #include <verto.h>
```

I only hit this error if I do a parallel build (make -j 8) - is that your case too ?

db260179, 2015-04-17:
compiling with just - make V=s

Produces this problem

psycho-nico, 2015-05-01:
Fixed by f746041323d9e35cd311c9280c06994baae017c4

Thanks !

commit f7460413 (2015-05-01): krb5: fix build failure on rebuild (closes: #1143)
closed 2015-05-01 by psycho-nico