openwrt/packages #1155: gdb seems to be broken

ClaymorePT, 2015-04-20:
I'm unable to build the selected by default gdb.
If I disable it in the toolchain options, everything goes well.

```
[claymore@Claymore trunk]$ make defconfig
Collecting package info: done
#
# configuration written to .config
#
[claymore@Claymore trunk]$ make menuconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

[claymore@Claymore trunk]$ make -j 9
 make[1] world
 make[2] tools/install
 make[2] package/cleanup
 make[3] -C tools/ccache compile
 make[3] -C tools/ccache install
 make[3] -C tools/sed compile
 make[3] -C tools/sstrip compile
 make[3] -C tools/patch-image compile
 make[3] -C tools/flock compile
 make[3] -C tools/patch compile
 make[3] -C tools/firmware-utils compile
 make[3] -C tools/patch-image install
 make[3] -C tools/flock install
 make[3] -C tools/sstrip install
 make[3] -C tools/firmware-utils install
 make[3] -C tools/sed install
 make[3] -C tools/patch install
 make[3] -C tools/yaffs2 compile
 make[3] -C tools/genext2fs compile
 make[3] -C tools/cmake compile
 make[3] -C tools/mkimage compile
 make[3] -C tools/lzma compile
 make[3] -C tools/scons compile
 make[3] -C tools/m4 compile
 make[3] -C tools/pkg-config compile
 make[3] -C tools/xz compile
 make[3] -C tools/yaffs2 install
 make[3] -C tools/scons install
 make[3] -C tools/genext2fs install
 make[3] -C tools/lzma install
 make[3] -C tools/xz install
 make[3] -C tools/mkimage install
 make[3] -C tools/squashfs4 compile
 make[3] -C tools/squashfs4 install
 make[3] -C tools/m4 install
 make[3] -C tools/autoconf compile
 make[3] -C tools/autoconf install
 make[3] -C tools/missing-macros compile
 make[3] -C tools/missing-macros install
 make[3] -C tools/pkg-config install
 make[3] -C tools/automake compile
 make[3] -C tools/automake install
 make[3] -C tools/libtool compile
 make[3] -C tools/cmake install
 make[3] -C tools/libtool install
 make[3] -C tools/libelf compile
 make[3] -C tools/e2fsprogs compile
 make[3] -C tools/flex compile
 make[3] -C tools/gmp compile
 make[3] -C tools/gengetopt compile
 make[3] -C tools/mklibs compile
 make[3] -C tools/patchelf compile
 make[3] -C tools/mm-macros compile
 make[3] -C tools/mm-macros install
 make[3] -C tools/libelf install
[... 159 lines cut ...]
Makefile:8591: recipe for target 'all-gdb' failed
make[5]: *** [all-gdb] Error 2
make[5]: Leaving directory '/home/claymore/OpenWRT/trunk/build_dir/toolchain-i386_i486_gcc-4.9-linaro_uClibc-0.9.33.2/gdb-linaro-7.6-2013.05'
Makefile:828: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/claymore/OpenWRT/trunk/build_dir/toolchain-i386_i486_gcc-4.9-linaro_uClibc-0.9.33.2/gdb-linaro-7.6-2013.05'
Makefile:53: recipe for target '/home/claymore/OpenWRT/trunk/build_dir/toolchain-i386_i486_gcc-4.9-linaro_uClibc-0.9.33.2/gdb-linaro-7.6-2013.05/.built' failed
make[3]: *** [/home/claymore/OpenWRT/trunk/build_dir/toolchain-i386_i486_gcc-4.9-linaro_uClibc-0.9.33.2/gdb-linaro-7.6-2013.05/.built] Error 2
make[3]: Leaving directory '/home/claymore/OpenWRT/trunk/toolchain/gdb'
toolchain/Makefile:81: recipe for target 'toolchain/gdb/compile' failed
make[2]: *** [toolchain/gdb/compile] Error 2
make[2]: Leaving directory '/home/claymore/OpenWRT/trunk'
toolchain/Makefile:80: recipe for target '/home/claymore/OpenWRT/trunk/staging_dir/toolchain-i386_i486_gcc-4.9-linaro_uClibc-0.9.33.2/stamp/.toolchain_install' failed
make[1]: *** [/home/claymore/OpenWRT/trunk/staging_dir/toolchain-i386_i486_gcc-4.9-linaro_uClibc-0.9.33.2/stamp/.toolchain_install] Error 2
make[1]: Leaving directory '/home/claymore/OpenWRT/trunk'
/home/claymore/OpenWRT/trunk/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
[claymore@Claymore trunk]$ make menuconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

[claymore@Claymore trunk]$ 

```

plntyk, 2015-04-29:
Sorry, but you filed the bug in the wrong place
This is bug in the first compile stages (toolchain creation) - accroding to your log "build_dir/toolchain- ..."

You should file a ticket at https://dev.openwrt.org for such issues

BUT:
- Please include your host system (which Linux / BSD variant are you using)
and the output of ./scripts/diffconfig.sh
- Please check if you have this error with a clean build without multiple jobs (no -j 9) since there might be some bugs with parallel builds

ClaymorePT, 2015-04-29:
Ok! Thank you for the correction.

closed 2015-05-01 by psycho-nico