openwrt/packages #1159: libgnutls error

taeasy, 2015-04-21:
cp -fpR /home/main/x86/build_dir/target-mips_34kc_uClibc-0.9.33.2/gnutls-3.4.0/ipkg-install/usr/lib/libgnutls.so.* /home/main/x86/staging_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx/tmp-libgnutls/usr/lib/
SHELL= /home/main/x86/staging_dir/host/bin/flock /home/main/x86/tmp/.root-copy.flock -c 'cp -fpR /home/main/x86/staging_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx/tmp-libgnutls/. /home/main/x86/staging_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx/'
rm -rf /home/main/x86/staging_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx/tmp-libgnutls
touch /home/main/x86/staging_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx/stamp/.libgnutls_installed
mkdir -p /home/main/x86/bin/ar71xx/packages /home/main/x86/build_dir/target-mips_34kc_uClibc-0.9.33.2/gnutls-3.4.0/ipkg-ar71xx/libgnutls-openssl/CONTROL /home/main/x86/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo
install -d -m0755 /home/main/x86/build_dir/target-mips_34kc_uClibc-0.9.33.2/gnutls-3.4.0/ipkg-ar71xx/libgnutls-openssl/usr/lib
cp -fpR /home/main/x86/build_dir/target-mips_34kc_uClibc-0.9.33.2/gnutls-3.4.0/ipkg-install/usr/lib/libgnutls-openssl.so.* /home/main/x86/build_dir/target-mips_34kc_uClibc-0.9.33.2/gnutls-3.4.0/ipkg-ar71xx/libgnutls-openssl/usr/lib/
cp: cannot stat `/home/main/x86/build_dir/target-mips_34kc_uClibc-0.9.33.2/gnutls-3.4.0/ipkg-install/usr/lib/libgnutls-openssl.so.*': No such file or directory
make[2]: *** [/home/main/x86/bin/ar71xx/packages/packages/libgnutls-openssl_3.4.0-1_ar71xx.ipk] Error 1
make[2]: Leaving directory `/home/main/x86/feeds/packages/libs/gnutls'
make[1]: *** [package/feeds/packages/gnutls/compile] Error 2
make[1]: Leaving directory `/home/main/x86'
make: *** [package/gnutls/compile] Error 2
main@bogon:~/x86$

nmav, 2015-04-21:
A work around is to disable the libgnutls-openssl package.

nmav, 2015-04-21:
That should be addressed with e76eacc6d9f488916955c71c9a2da802f236859f

closed 2015-04-21 by nmav