openwrt/packages #1160: Boost 1.58 does not build for MIPS32

paalsteek, 2015-04-21:
Boost 1.58 does not build for MIPS32 on a x86_64 machine. It worked fine for boost 1.57.

The error message is "mips-openwrt-linux-gcc: error: unrecognized command line option '-m32'"

Setting the 32-bit option in <pkg_build_dir>/bin.v2/project-cache.jam to false solves the compiler error. Didn't test the resulting package yet.

ClaymorePT, 2015-04-23:
I will work on it ASAP. 
This is probably just the BOOST_ABI variable with the wrong value selected.

paalsteek, 2015-04-23:
I have a patch that fixes the build but in my opinion not the source of the problem.

```patch
--- a/tools/build/src/tools/gcc.jam	2015-04-23 12:01:37.723438995 +0200
+++ b/tools/build/src/tools/gcc.jam	2015-04-23 12:00:21.427441384 +0200
@@ -451,7 +451,7 @@
         else
         {
             local arch = [ feature.get-values architecture : $(properties) ] ;
-            if $(arch) != arm
+            if $(arch) != arm && $(arch) != mips1
             {
                 if $(model) = 32
                 {

```

The target platform is Atheros AR231x/AR5312 btw.

ClaymorePT, 2015-04-23:
I'm trying an other thing. From the previous version (1.57) I removed a patch which was also patching gcc.jam. Since that was the only thing I changed, related to the CPU arch, it was probably the thing that broke everything without me noticing. 
I'm currently compiling a version and checking it if it works.

Thank you for your suggestion.

ClaymorePT, 2015-04-23:
It seems that the cause for the problem was this change: https://github.com/boostorg/boost/commit/beb53b6b9574f95309384bfc5afffb40d467bc04

I confirm that your patch work. I will create a pull-request ASAP which I also hope it adds support for Python 3.4

Cheers

ClaymorePT, 2015-04-24:
Pull request added (https://github.com/openwrt/packages/pull/1186)

closed 2015-04-27 by sbyx