openwrt/packages #1165: README.md or Wiki should contain information on how to install packages onto an openwrt system

NucleaPeon, 2015-04-21:
For those who come to this github project hoping to use the packages on openwrt but without extensive knowledge of its workings (like myself), it would be nice to have either links to or a README.md description to let users know how to get up and running quickly.

For example, I checked out the repo yet I have no scripts folder or feeds script which are both mentioned in the README.md.
There is no mention of the application dependencies required or other software and how they interact with these packages. (Does it use opkg? Another program not included with openwrt?)

What I would love to see is a short guide on taking any one package in your repository and installing it via opkg or the community software you use on an openwrt device. 

I know documentation probably isn't the highest priority in your eyes, but I have spent a lot of time trying to get an openwrt device setup and python3.x installed. I'd be willing to help with this if only I knew what your expectations are and how best to satisfy them.

Thank you.

NucleaPeon, 2015-04-21:
I'm looking at this page now, but I have no feeds.conf file nor does it mention what is required to get one: 

http://wiki.openwrt.org/doc/devel/feeds

Upon further investigation, it looks like I need something like buildroot to build the packages ( http://wiki.openwrt.org/doc/howto/buildroot.exigence#downloading_sources ) and that comes from a git repo that isn't this one.

plntyk, 2015-04-22:
added link to 
http://wiki.openwrt.org/doc/packages

to pages:
http://wiki.openwrt.org/doc/howto/generic.overview
http://wiki.openwrt.org/doc/howto/start

plntyk, 2015-04-22:
> For example, I checked out the repo yet I have no scripts folder or feeds script which are both mentioned in the README.md

scripts folder & feeds script are included in buildroot

-> Add "Inside `(link to: OpenWrt buildroot wiki page) run:" to readme.md ?

> There is no mention of the application dependencies required or other software and how they interact with these packages. (Does it use opkg? Another program not included with openwrt?)

since its an openwrt feed it needs the openwrt buildroot infrastructure.
Only packages with "host-build.mk" inclusion might need some host tools that should be covered by the wiki entry for buildroot for most of packages.

thess, 2015-04-23:
Added small addtion to [README.md](https://github.com/openwrt/packages/blob/master/README.md) to clarify purpose of this repostory.

Feel free to re-open this if these updates by myself and @plntyk are not enough.

closed 2015-04-23 by thess