openwrt/packages #1180: compilation of my own package fails

RIMEH1, 2015-04-23:
I have a python script and I wrote my Makefile this my path for the Makefile  package/mypackage/Makefile
 and for my script I have other path kam/home/Documents/src but I have this error en the compilation of my package 
cp: cannot stat `./src/*': No such file or directory
make[2]: *** [/home/kam/Bureau/opW/build_dir/target-mipsel_r2_uClibc-0.9.32/mypackage-1.0/.prepared_64a1177b42ea2ff7a6fd52c19a3a0fad] Error 1
make[1]: *** [package/mypackage/compile] Error 2
make: *** [package/mypackage/compile] Error 2

or in may Makefile I did:
include $(TOPDIR)/rules.mk

PKG_NAME:=mypackage
PKG_VERSION:=1.0
PKG_RELEASE:=1 
PKG_BUILD_DEPENDS:=python

include $(INCLUDE_DIR)/package.mk

define Package/mypackage
  SECTION:=net
  DEPENDS:= +python
  CATEGORY:=Network
  TITLE:=mypackage
  
endef
 
define Package/mypackage/description
package of my project 
endef

define Build/Prepare
	mkdir -p $(PKG_BUILD_DIR)
	$(CP) ./src/* $(PKG_BUILD_DIR)/
    
endef

define Package/mypackage/install
	$(INSTALL_DIR) $(1)/bin
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/mypackage $(1)/bin/
endef

$(eval $(call BuildPackage,mypackage,+python))

I need help please, 
thanks

thess, 2015-04-23:
'/home/ri/Bureau/opW/src' obviously does not exist.

RIMEH1, 2015-04-23:
@thess  you means that I should put my script in /opW 
opW it my folder openwrt

thess, 2015-04-23:
You would be better served if you posted this request for help in the OpenWrt forums.

closed 2015-04-23 by thess