openwrt/packages #1182: [utils/xz] liblzma breaks build for other packages

Wedmer, 2015-04-23:
After some observation I've found that some packages are linking against liblzma
For now I've found only two affected packages: python3 and snort. I will try to build all packages to find if any other are affected.

If liblzma is already compiled, then we will recieve
>Package python3-light is missing dependencies for the following libraries:
liblzma.so.5

Way to reproduce: select xz-utils (selects liblzma) and python3 packages then run
>make package/python3/clean;make package/xz/compile;make package/python3/compile;

Workaround: 
>make package/xz/clean;make package/python3/clean;make package/python3/compile;

psycho-nico, 2015-04-24:
ffmpeg is affected too (fix on the way)

psycho-nico, 2015-04-24:
ffmpeg fixed by c3a822ef3d79bda29dfd70b99e2d7600355b8df2

commodo, 2015-04-24:
For the `python3` case this fixes it: https://github.com/openwrt/packages/pull/1188
Of course, adding `+liblzma` needs to be done after this patch is applied.

closed 2015-04-28 by sbyx