openwrt/packages #1193: freeradius2 shouldn't be passed bind IPs or ports on the command line

jackmcbarn, 2015-04-26:
In radiusd.conf, the IPs and ports that the server binds to can be specified, but these are ignored, since /etc/init.d/radiusd passes IPs and ports on the command line. I don't see a need for this behavior. If there isn't one, then this logic should be removed from /etc/init.d/radiusd, since it seems to add complexity for no gain. If there is some need, the user should be given some way to configure this other than editing the init.d script. (Although /etc/default/radiusd does allow overriding the IP address, even it doesn't allow overriding the port.)

czakey, 2015-09-12:
Init script for freeradius2 should be refreshed. There are multiple problems with it. See #1769.

dangowrt, 2015-10-21:
@jackmcbarn @czakey Please test.
Any volunteers for maintaining the freeradius2 package?

diizzyy, 2017-06-07:
@dangowrt @thess @hnyman @hauke 
freeradius2 is EoL, perhaps the fastest/best way to handle it is to remove the package and deal with the potential breakage?

yousong, 2018-01-19:
Close as freeradius2 is now abandoned state by 5455a88d45397bf9bdb4c87d087e93ab57709619

closed 2018-01-19 by yousong