openwrt/packages #1194: python3-light is missing deps

blogic, 2015-04-27:
compile aborts with 
Package python3-light is missing dependencies for the following libraries:
liblzma.so.5
make[3]: *** [/home/buildbot/slave-local/malta_le/build/bin/malta/packages/packages/python3-light_3.4.3-2_malta_mipsel.ipk] Error 1

commodo, 2015-04-27:
Hmm, will retry.

But shortly before this was report Steven Barth @sbyx merged in some changes that specifically deal with `python3-lzma`.

Is this page accurate for monitoring broken packages ?
http://buildbot.openwrt.org:8010/broken_packages/malta/

blogic, 2015-04-27:
this happened on the release builder while running the CC-rc1 build. its a different machine to the link that you posted. the fail happened at 5am CET. i will do a refresh build when the machine is idles to see if the merge earlier today fixed the issue.

commodo, 2015-05-27:
@blogic 

is this still an issue ?
if not can it be closed please ? :)

thanks

hnyman, 2016-08-26:
python3-light was compiled ok for 15.05.1 and is compiling ok in trunk buildbot

closed 2016-08-26 by hnyman