openwrt/packages #1195: ruby module is missing deps

blogic, 2015-04-27:
build aborts with
Package ruby-psych is missing dependencies for the following libraries:
libyaml-0.so.2
make[3]: *** [/home/buildbot/slave-local/malta_le/build/bin/malta/packages/packages/ruby-psych_2.2.2-1_malta_mipsel.ipk] Error 1

luizluca, 2015-04-27:
I'll take a look

blogic, 2015-04-27:
i stumbled across this when building the package inside the SDK.

luizluca, 2015-04-27:
I already identified the problem. libyaml is an optional dep for ruby. If not present, it uses a bundled one.
I'll use  +PACKAGE_libyaml:libyaml

blogic, 2015-04-27:
please make it always use the external one in favour of the builtin one. the reason being that the resulting ipkg will always be the same and not depend on the content of the staging folder.

PR #1201 (2015-04-27): ruby: fix build when libyaml is present
libyaml is an optional dep for ruby psych. When missing, it uses
a bundled version of yaml. However, when libyaml is present in
openwrt build, ruby-psych packaging failed with deps not declared.
Now libyaml is configured as a hard dep for ruby-psych.

Also, the tk module was disabled in order to avoid a possible similar
problem if tk+x11 is provided in openwrt build. It was currently not
build because of missing deps.

Other minor changes:
- win32* modules where disabled (avoid err msg, no compile changes)
- Some files where removed in 2.2.x (like gserve [... cut]
closed 2015-04-27 by luizluca