openwrt/packages #1198: ocserv fails to build due to missing headers

blogic, 2015-04-27:
build aborts with

sec-mod-auth.c:56:28: fatal error: gssapi/gssapi.h: No such file or directory
 # include <gssapi/gssapi.h>
                            ^
compilation terminated.

sbyx, 2015-04-27:
Cannot reproduce this with current version.
https://github.com/openwrt/packages/commit/899724bd35ea190c81710e06d22dc30b7981416b
should fix this.

blogic, 2015-04-27:
happens when building the feed inside the SDK using 1b3fcf7af32.

ocserv won't be in RC1 if the bug is not reproduced/fixed.

sbyx, 2015-04-27:
hmm probably weird sdk-related failure since libgssapi is not in any active package feed. nevertheless i pushed: fa4365dd9b50a5bc032498d6f281daeefe02e49e so we can build RC1

@nmav hope that's okay with you

closed 2015-04-27 by sbyx

nmav, 2015-04-27:
Indeed, I wouldn't expect that to happen. The solution looks fine to me (the library which provides gssapi should be called libkrb5 or so). Thanks.

nmav, 2015-04-27:
Ok, it seems there is krb5 package, but I really doubt one would use Kerberos authentication for the VPN in an openwrt router. Let's leave it disabled by default unless someone asks for it.