openwrt/packages #1202: sqm-scripts: hotplug script should addtionally check if sqm init script is enabled before executing

braveheartleo, 2015-04-27:
Just like qos-scripts, there should be an additional check if sqm init script is enabled, before executing the hotplug script ``/etc/hotplug.d/iface/11-sqm``, so that we prevent unintentional activation of sqm if the service is disabled:

instead of
``[ "$ACTION" = ifup ] && /usr/lib/sqm/run.sh start ${DEVICE}``

should be
``[ "$ACTION" = ifup ] && /etc/init.d/sqm enabled && /usr/lib/sqm/run.sh start ${DEVICE}``

hnyman, 2015-04-27:
@tohojo @moeller0
Sounds reasonable to me.

tohojo, 2015-04-28:
Just pushed this change to the cero repo. Someone confirm that it works as expected? Then I'll push it here as well :)

moeller0, 2015-04-28:
This is bunk,

sqm-scripts is NOT functioning like qos-scripts, so this is not going to work. Please revert.

moeller0, 2015-04-28:
Hi Toke,

> Just pushed this change to the cero repo. Someone confirm that it works as expected? Then I'll push it here as well :)

	Unlike qos-scripts we do not have a single toggle enabled/disabled since we allow multiple instances of sqm-scripts on (hopefully) different interfaces, so we do not have a simple enabled switch. In our case we check in /usr/lib/run/sh whether the interface in question should be enabled or not. But most of all we do not handle �enabled� in /etc/init.d/sqm, so the change at best does nothing, or it might silently disable hotplug (I have not tested this).
	I appreciate the sentiment of the patch and want to thank braveheartleo, but think we need to pass on this�

Best Regards
	Sebastian

tohojo, 2015-04-28:
Why not? Having `/etc/init.d/sqm-scripts disable` turn off the whole thing doesn't seem too unreasonable?

I realise that there's additional checking for per-interface enabling in run.sh -- however, that is an addition, I'd think?

Maybe having the GUI do `/etc/init.d/sqm-scripts enable` whenever changes are applied would be good, though?

moeller0, 2015-04-28:
Hi Toke,

> Why not? Having /etc/init.d/sqm-scripts disable turn off the whole thing doesn't seem too unreasonable?

	It depends, I am open to this, but this needs further pluming in the GUI and the /etc/init.d script before it can work. Personally I am not to fluent in UCI, so I try to keep my interactions with it to a minimum ;) (I also think that /etc/init.d/sqm could use a procd-ification more urgently but even that is low on my list...)

> 
> I realise that there's additional checking for per-interface enabling in run.sh -- however, that is an addition, I'd think?

	For qos-scripts life is simpler as there ever is going to be maximally one instance enabled so that allows a simple toggle.

> 
> Maybe having the GUI do /etc/init.d/sqm-scripts enable whenever changes are applied would be good, though?

	Could you please elaborate on this idea, I am confused and do not see where you want to go with this.

Best Regards
	Sebastian

tohojo, 2015-04-28:
To expand on this, I think it's reasonable to stick to the standard way
of "turning everything off". We don't want people complaining that we
are being "bad citizens" by not turning off when asked to.

Of course, the existing checks should remain for per-interface
'enabled'. But these are complementary, aren't they?

The only adverse effect I could see is, as you say, that hotplug may
break if the user forgets to do the enable step. However, this is more
of a UI issue, and could probably be fixed in the Luci package, either
by an indicator or simply by running enable on the user's behalf (not
sure if the latter is frowned upon, though).

tohojo, 2015-04-28:
moeller0 <notifications@github.com> writes:

> Could you please elaborate on this idea, I am confused and do not see
> where you want to go with this.

Well, either we make the GUI run `/etc/init.d/sqm-scripts enabled` and
if that returns false, we put up a nice red banner saying "SQM scripts
are currently turned off globally, so this is not going to work". Or we
simply run `/etc/init.d/sqm-scripts enable` when applying the changes
and at least one interface is set to enabled.

>From a separation of responsibilities, messing with people's init
settings when not asked to feels icky. But on the other hand, a
pragmatic approach might be that if a user is applying a configuration,
it is reasonable to assume that he would want it re-applied on reboot ;)

-Toke

moeller0, 2015-04-28:
Hi Toke,

> To expand on this, I think it's reasonable to stick to the standard way
> of "turning everything off". We don't want people complaining that we
> are being "bad citizens" by not turning off when asked to.

	I do not think qos-scripts is setting a standard consciously, it is just that with maximally one interface the check is relatively simple to do. And let me iterate I am not against introducing a big on-off switch, but that switch needs to be wired up and the patch in question left out the wiring completely (thereby indicating that our indirection level in run.sh is too opaque to understand)
	@OpenWrt-Masters what do you consider default here?

> 
> Of course, the existing checks should remain for per-interface
> 'enabled'. But these are complementary, aren't they?

	We could argue so, and I am not opposed to this, just want to make sure we have an implementation that does what it promises ;)

> 
> The only adverse effect I could see is, as you say, that hotplug may
> break if the user forgets to do the enable step.

	Well, I am sure we can implement this in a way that hotplug still works; all I wanted to say is the actually commit was probably not too well tested. qos-scripts /etc/init.d.qos is a completely different beast than our /etc/init.d/sqm in that it incorporates some of what we do in run.sh. We could try to emulate that, but currently our /etc/init.d is more portable to other distributions. (OK, this is not going to be a winning argument on the openwrt list ;) )

> However, this is more
> of a UI issue, and could probably be fixed in the Luci package, either
> by an indicator or simply by running enable on the user's behalf (not
> sure if the latter is frowned upon, though).

Best Regards
	Sebastian

tohojo, 2015-04-28:
moeller0 <notifications@github.com> writes:

> I do not think qos-scripts is setting a standard consciously, it is
> just that with maximally one interface the check is relatively simple
> to do.

I didn't so much mean 'the qos-scripts standard' but rather 'the init
system standard'. I.e. from an OS perspective, if I turn off something
in my init system, it should stay off. I shouldn't have to go hunting
for application-specific switches.

It's similar to how if I do `/etc/init.d/httpd disable` I wouldn't want
Apache to turn itself back on because there are still sites configured
in /etc/apache/httpd.conf...

-Toke

moeller0, 2015-04-28:
Hi Toke,

> I didn't so much mean 'the qos-scripts standard' but rather 'the init
> system standard'. I.e. from an OS perspective, if I turn off something
> in my init system, it should stay off. I shouldn't have [...]

	Now I see, thanks for persisting. Then we need to modify /etc/config/sqm from /etc/init.d/sqm. How does qos-scripts behave in that regard? (My interpret ion so far was that qos script�s 10-qos if �asked� enabled? will look into /etc/config/qos and use the bool in there to decide whether to start or not, basically ignoring the user as much as we do; unless procd takes care of that. I really ned to look into procd documentation�)

Best Regards
	Sebastian

tohojo, 2015-04-28:
moeller0 <notifications@github.com> writes:

> Now I see, thanks for persisting. Then we need to modify
> /etc/config/sqm from /etc/init.d/sqm. How does qos-scripts behave in
> that regard? (My interpret ion so far was that qos script�s 10-qos if [...]

No, `/etc/init.d/FOO enabled` will check whether or not the FOO init
script is set to run at boot. I.e. whether it is linked into /etc/rc.d/
as SNNFOO where NN is the priority. And `/etc/init.d/FOO enable` will
create this symlink. This is what you manage under System -> Startup in
LUCI.

If that symlink does *not* exist, the script is not going to be started
(at all) during boot. Which is how it's supposed to be; so a hotplug
script shouldn't override this.

Whether or not the init script uses the procd features directly is
orthogonal to this. The procd daemon *is* the init system, so it's going
to be launching the scripts anyway; procd support in the init script is
more a matter of life-cycle support and automatic restarting of
long-running daemons. Which I don't really think we need in this case.

Now, when a package is installed as part of the base image, it is
usually `/etc/init.d/FOO enabled`'ed by default. But when installing a
new package it is not. (And as I'm writing that I realise that maybe
some of the issues we had that we solved by adding the hotplug script
were actually cause by the init script not being enabled? But I
digress).

So actually, IMO, from an init system perspective, the right thing to do
would be to just apply the patch and do nothing else. Having to manually
enable an init script after installing a package is the default, and
users should expect that. But in reality, they probably don't, so having
some sort of indicator in the GUI if it's not enabled would probably be
good. A red bar saying 'sqm-scripts is not enabled', with a link to the
System -> Startup page would probably do the trick.

-Toke

moeller0, 2015-04-28:
Hi Toke,

I am obviously out of my league here, but...

> No, `/etc/init.d/FOO enabled` will check whether or not the FOO init
> script is set to run at boot. I.e. whether it is linked into /etc/rc.d/
> as SNNFOO where NN is the priority. And `/etc/init.d/FOO [...]

	So bear with me, I have a hotplug script that calls the executable script /etc/init.d/qos but that script is not executed with the given argument “enabled", but caught by some other machinery? I guess my unix-foo is pretty weak. I will need to do some testing once I have shell access to my router.

> If that symlink does *not* exist, the script is not going to be started
> (at all) during boot. Which is how it's supposed to be; so a hotplug
> script shouldn't override this.
>
> Whether or not the init [...]

	Could well be, but for pppoe-links hotplug definitely was/is a good idea.

> 
> So actually, IMO, from an init system perspective, the right thing to do
> would be to just apply the patch and do nothing else.

	Well, we are back at step one then, someone should test this in cerowrt (which I can do this weekend ;) )

Best Reards
	Sebastian

moeller0, 2015-04-28:
Hi Toke,

I think I need to read /etc/rc.common next, so ignore my last mail…

best regards
	Sebastian

moeller0, 2015-04-28:
Hi Toke, and OP,

it seems I was wrong and the initial proposal good. So I am happy to change my opinion to just keep the change in cerowrt, confirm that it fully works (not that anybody should entrust me to do that after my display of ignorance). And then commit upstream to openwrt.

> Now, when a package is installed as part of the base image, it is
> usually `/etc/init.d/FOO enabled`'ed by default. But when installing a
> new package it is not. (And as I'm writing that I realise [...]

	I will have a look at the weekend about a noticeable warning in the GUI,I can give no ETA though…

Best Regards
	Sebastian

hnyman, 2015-04-28:
Lively discussion ;-)

The patch looks ok to me, but also bump the package version/revision in Makefile...

I can test it the patch tonight, but it looks so standard that I see no objection in committing it right away.

moeller0, 2015-04-28:
Hi Hnyman,

> Lively discussion ;-)

	More a verbose way of y showing my incompetence ;) . But hey. learned something new about classic unix init startup handling, so that all is well ;) *

> 
> The patch looks ok to me, but also bump the package version/revision in Makefile...
> 
> I can test it the patch tonight, but it looks so standard that I see no objection in committing it right away.

	On my todo list is testing noe only enable/enabled but also a full disable/enable loop�

Best Regards
	Sebastian

> 
> �

*) given the systems revolution I wonder how much use I will get out of this knowledge besides having stories to tell to me eventual grandkids ;)

hnyman, 2015-04-28:
The full enable/disable loop on package level would be duplicating the functionality in the main Luci GUI. Not maybe really worthwhile. 

But indicator about the disable status and possibly a link to the "startup" control page in case of disabled status might be ok.

In addition, the enabled/disabled status in a new firmware is so automatic that if you intentionally disable the service (via Luci or console "/etc/init.d/sqm disable") and then flash a new firmware containing the service package, it is then again automatically enabled in that firmware. The make system automatically enables all services (creates the symlinks in /etc/rc.d) that are installed into the firmware, as it has no knowledge of your router's current settings. Toke mentioned that vaguely, but not quite explicitly. There is no option to define a package to be installed but to be left disabled.  (uci-defaults can be used to circumvent that, but that is unorthodox and rare.)  That is one reason, why many packages have included a separate enabled/disabled option for the package itself in the package's /etc/config file.

tohojo, 2015-04-28:
So, I bumped the package revision in the cero repo... Always forget
that... Will await your testing before pushing it upstream (yeah, it's
trivial, but this time I'm erring on the side of caution instead of my
previous trigger-happiness ;)).

I think the indicator in the GUI can be a separate patch (and it's for a
different package as well). In any case we're currently just doing what
every other package is. I don't think we should be overriding the
default state of enabling/disabling init scripts in our little corner of
the world. So just having a warning with a link to the init script admin
page would be fine. Probably also easier to implement ;)

moeller0, 2015-04-28:
Hi Toke,

> So, I bumped the package revision in the cero repo... Always forget
> that... Will await your testing before pushing it upstream (yeah, it's
> trivial, but this time I'm erring on the side of caution instead of my
> previous trigger-happiness ;)).

	Thanks a lot. And thanks for your patience ;)

braveheartleo, 2015-04-28:
Hi,

I'll skip ahead of the lively discussion and share my experience on this. ;-)

I always put that additional check in sqm's hotplug script because the intention is that a service can either be enabled or disabled by manipulating (1) its configuration (setting option ``enabled`` to ``'0'`` in ``/etc/config/sqm``) or (2) the actual service itself (by disabling the sqm service via ``/etc/init.d/sqm disable``).

Given this flexibility, I sometimes don't bother changing the config, and instead keep the service file disabled. However, if any hotplug event matches the conditions for the sqm hotplug script, then sqm service is activated, despite having the service disabled. This behavior does not respect (2) that is a valid way of controlling a service.

I hope that clears it up, and since this behavior has gone unnoticed, I thought I would say something about it here.

moeller0, 2015-04-28:
Hi braveheartleo,

> Hi,
>
> I'll skip ahead of the lively discussion and share my experience on this. ;-)
>
> I always put that additional check in sqm's hotplug script because the intention is that a service can either be [...]

	I got an intense lesson in the init system of unices today; after what I learned I understand the actual issue (at least I think I do) and see that it actually solves a real problem. I had not looked into /etc/rc.common before and was under the delusion that unless /etc/init.d/sqm would handle an enabled argument this change would not do anything; since this is now cleared up I fully endorse your change. Many thanks a) for bringing it up and b) suffering through my public ;earning experience�

Best Regards
	Sebastian

hnyman, 2015-04-28:
I edited manually the change to the hotplug and it seemed to work ok, just as expected.

braveheartleo, 2015-04-28:
Hi Sebastian,

> I got an intense lesson in the init system of unices today; after what I learned I understand the actual issue (at least I think I do) and see that it actually solves a real problem. I had not [...]

Thank you for the endorsement. All's well that ends well right? And we learn one way or another. ;-)

Hi Hannu,
> I edited manually the change to the hotplug and it seemed to work ok, just as expected.

Thank you for the confirmation, and also for being able to hunt down and provide a fix to the *bad number* bug report I made sometime ago.

Hi Toke,
> So, I bumped the package revision in the cero repo... Always forget
that... Will await your testing before pushing it upstream (yeah, it's
trivial, but this time I'm erring on the side of caution instead of my
previous trigger-happiness ;)).

Thank you for the change. Now if only other package maintainers would do like you do and bump the package revision for any changes made to the files inside package...

Cheers,
ianp

moeller0, 2015-04-29:
Hi Toke,

so I tested this quickly with cerowrt, and it works as we intended. Hotplug does not start/restart sqm unless it is enabled in the init system. so the fix works and should be pushed into the openwrt repository as well, for all I know.

Thanks & Best Regards
	Sebastian

commit e9db7f40 (2015-04-30): sqm-scripts: Only run on hotplug if the init script is enabled.

Fixes #1202.
commit 7d10659c (2015-04-30): sqm-scripts: Only run on hotplug if the init script is enabled.

Fixes #1202.
closed 2015-04-30 by tohojo