openwrt/packages #1207: how I can add and compile external packages in my image openwrt

RIMEH1, 2015-04-28:
I want to add other package in my image and compile it from its package and there's netaddr json 
I used in my script the python json rpc packages but this package does not exist in my image I do not know how I can add.
is what I have downloaded a .zip folder and how I can do to add or there is another method to add some outher package in my image to complete the compilation

I need your help.

thess, 2015-04-28:
As said before -- head on over to http://forum.openwrt.org and post your request for help there.

closed 2015-04-28 by thess

RIMEH1, 2015-04-28:
If you have solution you can give me answer.thanks  https://forum.openwrt.org/viewtopic.php?pid=274286#p274286