openwrt/packages #1212: sqm-scripts: any reason why we can't optimize for dnscrypt packets as well?

braveheartleo, 2015-04-28:
I've been doing this on my setup:

    --- /usr/lib/sqm/simple.qos.orig	2015-04-28 23:48:06.000000000 +0800
    +++ /usr/lib/sqm/simple.qos	2015-04-28 00:15:42.000000000 +0800
    @@ -60,7 +60,7 @@
     # Emanating from router, do a little more optimization
     # but don't bother with it too much. 
     
    -ipt -t mangle -A OUTPUT -p udp -m multiport --ports 123,53 -j DSCP --set-dscp-class AF42
    +ipt -t mangle -A OUTPUT -p udp -m multiport --ports 123,53,443 -j DSCP --set-dscp-class AF42
    
     #Not clear if the second line is needed
     #ipt -t mangle -A OUTPUT -o $IFACE -g QOS_MARK_${IFACE}

    --- /usr/lib/sqm/stop.sh.orig	2015-04-28 23:48:17.000000000 +0800
    +++ /usr/lib/sqm/stop.sh	2015-04-28 00:15:55.000000000 +0800
    @@ -27,7 +27,7 @@
     	[ ! -z "$CUR_IFB" ] && ipt -t mangle -D POSTROUTING -o $CUR_IFB -m mark --mark 0x00 -g QOS_MARK_${IFACE} 
     	ipt -t mangle -D POSTROUTING -o $IFACE -m mark --mark 0x00 -g QOS_MARK_${IFACE} 
     	ipt -t mangle -D PREROUTING -i vtun+ -p tcp -j MARK --set-mark 0x2
    -	ipt -t mangle -D OUTPUT -p udp -m multiport --ports 123,53 -j DSCP --set-dscp-class AF42
    +	ipt -t mangle -D OUTPUT -p udp -m multiport --ports 123,53,443 -j DSCP --set-dscp-class AF42
     	ipt -t mangle -F QOS_MARK_${IFACE}
      	ipt -t mangle -X QOS_MARK_${IFACE}
     }

as I have dnscrypt-proxy running. Is there any reason why we can't optimize for dnscrypt packets as well in upstream sqm-scripts, given that those may be considered dns (albeit encrypted) packets as well?

hnyman, 2015-04-28:
Isn't port 443 the generic https port? Wouldn't that change affect quite a lot of traffic?

braveheartleo, 2015-04-28:
Not if you're matching for 443/udp. HTTPS is transported over 443/tcp. Because I'm only changing the iptables rule emanating from the router, this shouldn't match a lot of traffic and be a problem as there are only a few services that can be run on the router, much less that use UDP port 443.

Please see this for reference http://www.speedguide.net/port.php?port=443

moeller0, 2015-04-29:
I wonder whether it would not be easier, if dnscrypt-proxy would just mark its packets AF42, we try to honor user supplied DSCP-values in sqm-scripts? If there are no false positives for 443/udp I guess we could add them in simple.qos, but I would prefer if the packet source would mark them properly ;). That said, I can easily be convinced to change my opinion as you know. As I do not run dns-crypt I cannot test this myself...

Best Regards
        M.

moeller0, 2015-05-12:
So what is the verdict on this? Anybody beside @braveheartleo with an opinion? @hnyman can you actually test this? (I am sorry to report that I am still on cerowrt, sorry in that I can not really test how a change would work, since cerowrt does not have dnscryp-proxy, as far as I know, so I would like a second opinion). Also I would love to learn whether these packets need prioritization anyway (are they often dropped by SQM?) as I would assume that they should be sparse enough to be treated nicely by fq_codel...

hnyman, 2015-05-12:
I don't use dnscrypt, so no testing from me.

In principle I understand the prioritization of all dns traffic, but just wonder if there is other traffic to that port.

@braveheartleo has probably a point in the thought that changing the rule for 443/udp originating at the router itself should not catch more additional traffic.

braveheartleo, 2015-05-14:
> @moeller0 Also I would love to learn whether these packets need prioritization anyway (are they often dropped by SQM?) as I would assume that they should be sparse enough to be treated nicely by fq_codel...

I don't claim to be knowledgable in queue management, nor in sqm for that matter. I only did those changes based on the existing rules in sqm and the comment from the code `Emanating from router, do a little more optimization but don't bother with it too much.` Also, qos-scripts create an iptable rule for packet prioritization that matches 443/udp. It only makes sense that sqm could do that as well, unless there are some caveats to this that I'm not aware of.

> I wonder whether it would not be easier, if dnscrypt-proxy would just mark its packets AF42, we try to honor user supplied DSCP-values in sqm-scripts?

That's well and good if the service supports such flexible configuration, but with queue management mechanisms such as sqm-scripts and qos-scripts, it may choose to override those values depending upon its design (like in the case of sqm squashing inbound DSCP by default).

dtaht, 2015-05-16:
Quic is carried over port 443/udp. Not a good idea to generally optimize for that!

However that particular iptables rule is just for packets being sourced by the router, not for traffic passing through, and so far as I know nobody is going to write any quic proxies.

The AF42 marking for dns was quite arbitrary and I would enjoy a discussion as to what dns and ntp packets SHOULD be marked as. It kind of crept into the production code by accident! It started with a hack to mosh to make it use that bit pattern, to see what bits ISPs stripped away. Comcast strips away all but the CS1 bit, for example....

My own opinion is that we should find ways to deprioritize certain kinds of traffic, and be extremely cautious about prioritizing stuff. for example I deprio rsync... I don't match on ssh, depending on it to choose the right marking depending on it's mode...

We are working on the design of cake to try to resolve thorny issues such as this, please see the bufferbloat "cake" mailing list for more details.

dtaht, 2015-05-16:
and I am also kind of resistant to making it easy for people to play optimization bingo with these scripts. Goal was to have a simple, very few knobs, default that was nearly impossible to get wrong, and if it takes an expert to change the script to suit them, all the better.

as it turned out lots of people got it wrong, leaving gro and tso offloads ON by default in their latest routers, while qos was present. Sigh.

If you can show a measurable performance difference for dnscrypt during other traffic I would consider it for sqm, but fq_codel's "sparse" optimization should usually do it for you, and if that doesn't cake will.

braveheartleo, 2015-05-19:
@dtaht
Hi Dave,

> The AF42 marking for dns was quite arbitrary and I would enjoy a discussion as to what dns and ntp packets SHOULD be marked as. It kind of crept into the production code by accident! It started with [...]

> but fq_codel's "sparse" optimization should usually do it for you, and if that doesn't cake will. 

With these, I withdraw my initial inquiry and yield to your suggestions.

closed 2015-05-19 by braveheartleo