openwrt/packages #1221: choova-chilli - missing parameters

obsy, 2015-05-01:
Coova-chilli's config defines option dhcpif. This parameter does not exists, because init script  parses option network, not dhcpif.

teslamint, 2015-05-04:
option network just converts interface name(e.g. wan, lan) to interface name(ethx.x) and assigns to option dhcpif afaik.

obsy, 2015-05-04:
In coova-chilli config, but not in /etc/config/chilli

teslamint, 2015-05-18:
Well, if i set network option with bridge network interface(e.g: br-wifi) with multiple devices(wlanX.X), I got this:
```
Mon May 18 13:34:58 2015 local6.err coova-chilli[13843]: net.c: 80: 19 (No such device) ioctl(SIOCSIFFLAGS) failed on
Mon May 18 13:34:58 2015 local6.err coova-chilli[13843]: net.c: 57: 19 (No such device) ioctl(SIOCSIFFLAGS) failed
Mon May 18 13:34:58 2015 local6.err coova-chilli[13843]: net.c: 120: 19 (No such device) ioctl(SIOCSIFADDR) failed
Mon May 18 13:34:58 2015 local6.err coova-chilli[13843]: net.c: 1291: 19 (No such device) ioctl(d=5, request=35111) failed
Mon May 18 13:34:58 2015 local6.err coova-chilli[13843]: chilli.c: 7339: Failed to create dhcp listener on
```
so I think it's better use dhcpif option than network option in that case.

obsy, 2015-05-18:
Strange, the script gets the physical interface from  network option. In coova config exists only dhpif.

teslamint, 2015-05-18:
It would be OK if we use single physical interface(e.g. eth1.0, wlan0.1, etc.) on network option.

PR #1340 (2016-04-09): coova-chilli: add network option in chilli.config
fix #1221
commit eed8b911 (2016-04-09): coova-chilli: add network option

fix #1221
closed 2016-04-09 by champtar