openwrt/packages #1234: prosody fails to link against liblua

nortobor, 2015-05-05:
always fails with "liblua.so: error adding symbols: File in wrong format"... the files (liblua and the object files from prosody) look pretty normal to me, so not compiled with host compiler or anything

```
mips64-openwrt-linux-gnu-ld -o net.so net.o -L/data/embedded/openwrt_erlite/staging_dir/target-mips64_octeon_64_glibc-2.19/usr/lib -L/data/embedded/openwrt_erlite/staging_dir/target-mips64_octeon_64_glibc-2.19/lib -L/data/embedded/openwrt_erlite/staging_dir/toolchain-mips64_octeon_64_gcc-4.9-linaro_glibc-2.19/usr/lib -L/data/embedded/openwrt_erlite/staging_dir/toolchain-mips64_octeon_64_gcc-4.9-linaro_glibc-2.19/lib -L/data/embedded/openwrt_erlite/staging_dir/target-mips64_octeon_64_glibc-2.19/usr/lib -llua -lm -ldl -shared
/data/embedded/openwrt_erlite/staging_dir/target-mips64_octeon_64_glibc-2.19/usr/lib/liblua.so: error adding symbols: File in wrong format
Makefile:39: recipe for target 'net.so' failed
make[4]: *** [net.so] Error 1
```

nortobor, 2015-05-06:
i found out, it works, if i tell the linker to use the format -melf64btsmip

but how to do that (elegantly) in the Makefile, and only for mips64?

diizzyy, 2017-06-07:
@nortobor 
Does this still occur on LEDE?

diizzyy, 2018-06-11:
@hnyman @thess 
Close, reporter timeout.

diizzyy, 2018-07-22:
@yousong

closed 2018-07-22 by karlp