openwrt/packages #1243: unable to access some web sites with Privoxy

tukeylee, 2015-05-09:
When accessing the following websites with Privoxy, LUCI page shows, no matter how configure Privoxy or even toggle off Privoxy:
www.uc.cn
go.uc.cn
www.ucweb.cn

Everything goes fine if Privoxy is shutdown (/etc/init.d/privoxy stop).

I tried both openwrt trunk and 14.07.
12.09 seems OK.

chris5560, 2015-05-09:
LuCI Page shows ?
I tried all 3 links on my own environment without problems running trunk with Privoxy 3.0.23.
It look like problems with the action/filter files, please ask Privoxy support.
12.09 has Privoxy 3.0.19
14.07 has Privoxy 3.0.21
Trunk has Privoxy 3.0.23

tukeylee, 2015-05-09:
I leave all default settings (including default action and filter files), except my IP.
Below is the screen shot what I saw.
If you have any clue, please kindly let me know.

[image]
[image]
[image]

chris5560, 2015-05-09:
Hi,
that's looks like a dns problem, because if you address www.uc.cn you are routed to a uHTTPd/LuCI web page (hopefully yours). Please check from your "browser" box nslookup www.uc.cn and do the same on the box running privoxy. Retry the same with the ping command and check the IP's you get.
You don't need an answer from ping only verify IP's.

tukeylee, 2015-05-14:
Hi, chris5560,

Thank you for replying. I have some clues now!

When I do "nslookup go.uc.cn", the following result shows (the DNS result may be different in your country):
----------------------------------
# nslookup go.uc.cn
Server:    127.0.0.1
Address 1: 127.0.0.1 localhost

Name:      go.uc.cn
Address 1: ::
Address 2: 111.161.46.115 dns115.online.tj.cn
----------------------------------

Pay attention on "Address 1: ::" here, which means IPv6 address of this domain name is not set. Although an IPv4 address (111.161.46.115) is specified, it malfunctions with Privoxy (in 14.07 or trunk), which causes "go.uc.cn" resolved as OpenWrt box itself.

You can do a test with /etc/hosts. Just add ":: go.uc.cn" into /etc/hosts, then the same thing will happen, I believe.

If you know how to solve this problem, please let me know.

tukeylee, 2015-05-14:
I also tested on Ubuntu 12.04, compiling Privoxy 3.0.23. Everything went OK.
So maybe it's a bug of Openwrt 14.07 and later.

chris5560, 2015-05-14:
Hi,

AA12.09 has no IPv6 stack by default. That was changed in BB14.07.
Ubuntu is awful on IPv6 support. It's not using IPv6 as prefered communication.
In the current client (15.04) you have modify configs until you get IPv6 adress assigned by dnsmasq server.
Windows works fine on the same network. It takes me days until my Ubuntu client was able to ping6 ipv6.google.com.

It's not a problem of Privoxy if the nameserver is replying :: as the IP of go.uc.cn.
On my router here in Germany I got the same :: as IPv6 for go.uc.cn also if using 8.8.8.8 (Google public dns) as nameserver.

Please ask Privoxy support if there is any option to set Privoxy to prefer IPv4 address for connection.
Possibly you can configure your dns-server to send/prefer IPv4 as "Address1".
You can configure many browsers to prefer IPv4 or IPv6 when opening connection to web server.

Christian

PS: [::] is not localhost, IPv6 localhost is [::1]

tukeylee, 2015-05-15:
Hi,
So, is it possible to make OpenWrt 14.07 use IPv4 as preferred communication just as Ubuntu does?

chris5560, 2015-05-16:
Search / post a message to OpenWrt forum, I personally don't know.
Also inform admin of uc.cn domain that his DNS IPv6 entry provided to the public world is broken.
He or his provider is sending an invalid IPv6. [::] is replicated around the world.
Try i.e. Google's public DNS.
I think it make more sense that uc.cn has a valid or no IPv6 DNS entry in stead of manipulating your well working system. IPv6 is the future.

Christian

hnyman, 2016-08-23:
Old issue and mainly a DNS issue outside Openwrt.

closed 2016-08-23 by hnyman