openwrt/packages #1261: coova-chili: build fails without selected kmod

plntyk, 2015-05-16:
Its possible to build coova-chili without Kernel modules / kmod-ipt-coova

This leads to a build failure when trying to copy library files unconditionally.
error:
```cp: cannot stat '/home/ashkan/openwrt/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/coova-chilli-1.3.0+20141128/ipkg-install/usr/lib/iptables/lib*.so': No such file or directory```
```
$(INSTALL_DIR) $(1)/usr/lib/iptables
	$(CP) $(PKG_INSTALL_DIR)/usr/lib/iptables/lib*.so $(1)/usr/lib/iptables/
```

reported in IRC
log: http://paste.ubuntu.com/11163340/

@teslamint

AndrewMargarit, 2015-05-16:
I found the same issue.
Is not the kmod, believe it to be this line that is causing the issue.

define Package/coova-chilli/install
                (CP) $(PKG_INSTALL_DIR)/usr/lib/iptables/lib*.so 
$(1)/usr/lib/iptables/

Believe that PKG_INSTALL_DIR is ipkg-install witch doesn't contain the 
iptables directory.
Instead it is in here - ipkg-ar71xx/coova-chilli/usr/lib/iptables

Best regards,
Andrew Margarit
Technical Support Engineer

PolkaSpots Supafly Wi-Fi
www.polkaspots.com
         
Phone: 020 7183 1471
Email: Andrew@polkaspots.com
---------------------------------------------------------------------------------------
The information contained in this email and its attachments is confidential and may be the subject of legal, professional or other privilege. It is intended only for the named addressees and may not be disclosed to anyone else without consent from PolkaSpots Limited. If you are not the named addressee you must not use, disclose, distribute, copy, print or rely on the contents of this email and should destroy it immediately. Whilst PolkaSpots Limited takes care to protect its systems from electronic virus attack or other harmful event, the firm gives no warranty that this email message (including any attachments to it) is free of any virus or other harmful matter and accepts no responsibility for any loss or damage resulting from the recipient receiving, opening or using it. If you need any further information, please contact the originator of this message on +44 (0)20 7183 1471.
PolkaSpots is a limited company registered in the UK (registered number 05508105). Registered Office: 204 Buspace Studios, Conlan Street, London W10 5AP
Website: www.polkaspots.com
WiFi Hotspot Blog: www.polkaspots.com/blog
---------------------------------------------------------------------------------------

plntyk, 2015-05-16:
that copy line was introduced with the kmod feature - see https://github.com/openwrt/packages/commit/ca9d3f24cffa57caa101020923c33404e98bee43

teslamint, 2015-05-18:
Okay. How about this code:
```
	$(if $(CONFIG_PACKAGE_kmod-ipt-coova), \
		$(INSTALL_DIR) $(1)/usr/lib/iptables; \
		$(CP) $(PKG_INSTALL_DIR)/usr/lib/iptables/lib*.so $(1)/usr/lib/iptables/; \
	)
```
Is there any better solutions than this?

closed 2015-05-21 by kaloz

irongomme, 2017-02-02:
For target platform 'WNR1200AC', I had to replace : 

```
$(CP) $(PKG_INSTALL_DIR)/usr/lib/iptables/lib*.so $(1)/usr/lib/iptables/; \
```

By 

```
$(CP) $(PKG_BUILD_DIR)/src/linux/lib*.so $(1)/usr/lib/iptables/; \
```

to make it works ! If it can helps someone ...

PR #1269 (2015-05-21): coova-chilli: fix stat failed error when compile without kmod
This patch fixes #1261.
commit c14e55e4 (2015-05-21): coova-chilli: fix stat failed error when compile without kmod

This patch fixes #1261.
commit 3c129346 (2016-05-03): coova-chilli: fix stat failed error when compile without kmod

This patch fixes #1261.