openwrt/packages #1262: libusb-1.0: can't install on trunk

JamesT42, 2015-05-16:
I cant install libusb-1.0 on trunk:

root@OpenWrt:~# opkg install libusb-1.0
Installing libusb-1.0 (1.0.19-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ramips/mt7620/packages/base/libusb-1.0_1.0.19-1_ramips_24kec.ipk.
Configuring libusb-1.0.
grep: /usr/lib/opkg/info/libusb-1.control: No such file or directory
cat: can't open '/usr/lib/opkg/info/libusb-1.list': No such file or directory

thess, 2015-05-17:
Neither this package nor the helper functions in /lib/functions.sh are maintained in this repo.

Submitted bug report in: https://dev.openwrt.org/ticket/19668

JamesT42, 2015-05-17:
Hi! Ok thanks, I thougt this was something wrong with the package itself.

KanjiMonster, 2015-05-19:
This has been fixed in OpenWrt with r45702, thanks for reporting.

closed 2015-05-19 by KanjiMonster