openwrt/packages #1282: PostgreSQL fails to build on arm64/aarch64 & affects other pkgs

plntyk, 2015-05-22:
9.0.x does not really support arm64 it seems
BUT 9.0.x still receives security updates
9.4.x does work on arm64 (compile only test)

see https://github.com/openwrt/telephony/issues/26
atm several other packages (>8) fail hard if pgsql is missing

I created a patch:
https://gist.github.com/plntyk/7259bee75576e67f8d2c

the added stuff (PKG_MACRO_PATH & first patch) is due to autoreconf & missing macro errors

dangowrt, 2015-06-24:
@jow- updating PostgreSQL seems like a good idea. The patch looks fine, changes to configure arguments should be reviewed. Unless you are working on this already, I'll pick it up.

dangowrt, 2015-06-30:
I picked up the patch posted by @plntyk and started to work on modernizing the init script to use procd.
https://github.com/dangowrt/packages/tree/update-postgresql

Eventually this could elimitate the need for shadow-su (@blogic: apart from procd being able to launch an instance using another user, there is no defined way how to run processes as another user, neither in init scripts nor anywhere else. Some folks use sudo, others use shadow-su, some enable busybox-su and yet others use SUID/GUID binaries - none of these methods are both tight and clean enough imho).
@jow- PostgreSQL 9.0 is going to be EOL this September....

jow-, 2015-08-13:
9cfcbeeeac85e67a64527ee3fac890212b5fccbb

closed 2015-08-13 by jow-