openwrt/packages #1285: DDNS requiring package update while ddns-scripts already at most recent version

Borromini, 2015-05-22:
I ran a build on a trunk checkout from yesterday (May 21st), and the LuCI DDNS package is complaining about the ddns-scripts package not being the most recent version. However, it's at 2.4.2, which is the latest.

[image]

db260179, 2015-05-23:
The change of min version in the luci-app-ddns seems to be the cause of this

+DDNS_MIN = "2.4.2-1"	-- minimum version of service required

Previously
-DDNS_MIN = "2.2.0-1"	-- minimum version of service required
Doesnt show this error

chris5560, 2015-05-24:
Not an issue of ddns-scripts.
Will be fixed in luci-app-ddns 2.2.4-1.
Christian

closed 2015-07-27 by champtar