openwrt/packages #1297: qrencode  compile error

taeasy, 2015-05-24:
checking if mips-openwrt-linux-uclibc-gcc supports -fno-rtti -fno-exceptions... no
checking for mips-openwrt-linux-uclibc-gcc option to produce PIC... -fPIC -DPIC
checking if mips-openwrt-linux-uclibc-gcc PIC flag -fPIC -DPIC works... yes
checking if mips-openwrt-linux-uclibc-gcc static flag -static works... yes
checking if mips-openwrt-linux-uclibc-gcc supports -c -o file.o... yes
checking if mips-openwrt-linux-uclibc-gcc supports -c -o file.o... (cached) yes
checking whether the mips-openwrt-linux-uclibc-gcc linker (mips-openwrt-linux-uclibc-ld) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... yes
checking for mips-openwrt-linux-ranlib... (cached) mips-openwrt-linux-uclibc-gcc-ranlib
checking pkg-config is at least version 0.9.0... yes
checking for strdup... yes
checking for pthread_mutex_init in -lpthread... yes
checking for png... no
configure: error: Package requirements ("libpng") were not met:

No package 'libpng' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables png_CFLAGS
and png_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
make[3]: *** [/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/.configured_yyyyyyyy] Error 1
make[3]: Leaving directory `/home/main/ar71xx/feeds/packages/libs/qrencode'
make[2]: *** [package/feeds/packages/qrencode/compile] Error 2
make[2]: Leaving directory `/home/main/ar71xx'
make[1]: *** [/home/main/ar71xx/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/ar71xx'
make: *** [world] Error 2

thess, 2015-05-25:
Fixed in 1e2cd3318b67584d12ea9ee27e1b23927510f630

taeasy, 2015-05-25:
rm -f /home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/.configured_*
touch /home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/.configured_yyyyyyyy
make -C /home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4 AR="mips-openwrt-linux-uclibc-gcc-ar" AS="mips-openwrt-linux-uclibc-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float" LD=mips-openwrt-linux-uclibc-ld NM="mips-openwrt-linux-uclibc-gcc-nm" CC="mips-openwrt-linux-uclibc-gcc" GCC="mips-openwrt-linux-uclibc-gcc" CXX="mips-openwrt-linux-uclibc-g++" RANLIB="mips-openwrt-linux-uclibc-gcc-ranlib" STRIP=mips-openwrt-linux-uclibc-strip OBJCOPY=mips-openwrt-linux-uclibc-objcopy OBJDUMP=mips-openwrt-linux-uclibc-objdump SIZE=mips-openwrt-linux-uclibc-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16" LDFLAGS="-L/home/main/ar71xx/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/home/main/ar71xx/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/home/main/ar71xx/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/main/ar71xx/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib -s" DESTDIR="/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/ipkg-install" all install
make[4]: Entering directory `/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4'
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/use/missing aclocal-1.14 -I m4
/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/use/missing: line 81: aclocal-1.14: command not found
WARNING: 'aclocal-1.14' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make[4]: *** [aclocal.m4] Error 127
make[4]: Leaving directory `/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4'
make[3]: *** [/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/.built] Error 2
make[3]: Leaving directory `/home/main/ar71xx/feeds/packages/libs/qrencode'
make[2]: *** [package/feeds/packages/qrencode/compile] Error 2
make[2]: Leaving directory `/home/main/ar71xx'
make[1]: *** [/home/main/ar71xx/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/ar71xx'
make: *** [world] Error 2
main@bogon:~/ar71xx$

thess, 2015-05-25:
Not sure what your problem is. I could not reproduce this error in a clean build after update. I did however, push a new 'fix' to the repo which completely removes the libpng dependency. I suggest
> ./scripts/feeds update
> make menuconfig
> make clean
> make package/qrencode/compile

... and see if you get a good result.

taeasy, 2015-05-25:
make[3]: Entering directory `/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4'
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/use/missing aclocal-1.14 -I m4
/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/use/missing: line 81: aclocal-1.14: command not found
WARNING: 'aclocal-1.14' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make[3]: *** [aclocal.m4] Error 127
make[3]: Leaving directory `/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4'
make[2]: *** [/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/.built] Error 2
make[2]: Leaving directory `/home/main/ar71xx/feeds/packages/libs/qrencode'
make[1]: *** [package/feeds/packages/qrencode/compile] Error 2
make[1]: Leaving directory `/home/main/ar71xx'
make: *** [package/qrencode/compile] Error 2
main@bogon:~/ar71xx$ 

like the problem on here:
/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/use/missing: line 81: aclocal-1.14: command not found

thess, 2015-05-26:
I think I found the original problem -- PKG_FIXUP is wrong (syntax error) and therefore autoreconf is not happening. I think there is another problem with configure after this one is resolved.

@oneru - please help here.

oneru, 2015-05-26:
@thess There is a patch in /patches that removes the libpng requirement. Please do not make changes to packages that you do not maintain. I try to keep an eye out for issues for packages that I maintain, but I'm on vacation for the long weekend (here in the U. S.). The correct response would be to tag me in your first response and give me at least a day to respond.

 I'll try to figure out what is going on here, but you need to revert your changes to my package. As per http://wiki.openwrt.org/doc/devel/packages it is not considered the right solution to just hack changes into a ./configure " Instead of patching ./configure one should fix the file from which ./configure is generated in autotools: configure.ac (or configure.in, for very old packages). Another important file is Makefile.am from which Makefiles (with configure output) are generated."

Yes, there appears to be a typo with PKG_FIXUP. However, the real problem here seems to be this: "line 81: aclocal-1.14: command not found WARNING: 'aclocal-1.14' is missing on your system." from taeasy's last build log.

oneru, 2015-05-26:
@thess I've fixed the typo you discovered. Pull request 1307. This also contains commits reverting your changes.

@taeasy Once that update gets pulled, I'll ping you here and you can try compiling again. Thanks for reporting the problem; We'll get it working for you.

taeasy, 2015-05-26:
The problem persists.

make[3]: Entering directory `/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4'
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/use/missing aclocal-1.14 -I m4
/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/use/missing: line 81: aclocal-1.14: command not found
WARNING: 'aclocal-1.14' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make[3]: *** [aclocal.m4] Error 127
make[3]: Leaving directory `/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4'
make[2]: *** [/home/main/ar71xx/build_dir/target-mips_34kc_uClibc-0.9.33.2/qrencode-3.4.4/.built] Error 2
make[2]: Leaving directory `/home/main/ar71xx/feeds/packages/libs/qrencode'
make[1]: *** [package/feeds/packages/qrencode/compile] Error 2
make[1]: Leaving directory `/home/main/ar71xx'
make: *** [package/qrencode/compile] Error 2
main@bogon:~/ar71xx$

closed 2015-05-26 by taeasy

thess, 2015-05-26:
@oneru - I am sorry for the trouble I have caused here. What started out as something that looked like a simple dependency problem turned into a bad move on my part. For that, I sincerely apologize.

You are totally correct, there should never be a need to directly edit a _configure_ file if you can re-generate it. My misunderstanding came from the fact it wasn't being re-created due to the missed typo. A simple problem just got worse by my attempt to fix it quickly.

Your fixes have been merged by @sbyx this morning. He beat me to it while I was having coffee. Thanks for bailing me out.

oneru, 2015-05-26:
@thess no actual harm done.

@taeasy Did that fix it for you?

taeasy, 2015-05-26:
i'm trying now, thanks

taeasy, 2015-05-27:
is work now, thanks