openwrt/packages #1312: python: Segmentation fault in bsddb module

FlorianFieber, 2015-05-27:
Python2 segfaults on opening a bsddb (which is the default for anydbm and thus shelve if available) somewhere in _bsddb.db.DBEnv.open.

```
Python 2.7.9 (default, May  1 2015, 23:21:05) 
[GCC 4.8.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import bsddb
>>> bsddb.hashopen("db")
Segmentation fault
```

```
spam@egg:~$ opkg list-installed | grep "python[^3]"
python - 2.7.9-5
python-base - 2.7.9-5
python-codecs - 2.7.9-5
python-compiler - 2.7.9-5
python-ctypes - 2.7.9-5
python-db - 2.7.9-5
python-decimal - 2.7.9-5
python-distutils - 2.7.9-5
python-email - 2.7.9-5
python-gdbm - 2.7.9-5
python-light - 2.7.9-5
python-logging - 2.7.9-5
python-multiprocessing - 2.7.9-5
python-ncurses - 2.7.9-5
python-openssl - 2.7.9-5
python-pip - 1.5.6-1
python-pydoc - 2.7.9-5
python-setuptools - 7.0-1
python-sqlite3 - 2.7.9-5
python-unittest - 2.7.9-5
python-xml - 2.7.9-5
```

commodo, 2015-05-27:
I wonder if this would be fixed by upgrading libdb ?
Will look into this one as well.

commodo, 2015-05-28:
Actually this should be left open a bit longer.
This is still an issue.

sbyx, 2015-05-28:
Hmm seems this was auto-closed by github when i merged your PR.

commodo, 2015-05-28:
Ah, github being too smart (like Python's build scripts sometimes).

commodo, 2015-05-28:
Right.
So after a bit of digging.

Segfault seems to occur in 
```
(gdb) bt
#0  __db_fcntl_mutex_lock (env=0x103d1860, mutex=<optimized out>) at ../dist/../mutex/mut_fcntl.c:87
#1  0x48387860 in __mutex_alloc_int (env=env@entry=0x103d1860, locksys=locksys@entry=1, alloc_id=alloc_id@entry=20, flags=8, indxp=indxp@entry=0xbfe82c88)
    at ../dist/../mutex/mut_alloc.c:115
#2  0x48387a64 in __mutex_alloc (env=env@entry=0x103d1860, alloc_id=alloc_id@entry=20, flags=<optimized out>, flags@entry=0, indxp=indxp@entry=0xbfe82c88) at ../dist/../mutex/mut_alloc.c:54
#3  0x4838867c in __mutex_open (env=env@entry=0x103d1860, create_ok=create_ok@entry=1) at ../dist/../mutex/mut_region.c:126
#4  0x483be85c in __env_open (dbenv=0x104e6a58, db_home=<optimized out>, flags=4421, mode=<optimized out>) at ../dist/../env/env_open.c:358
#5  0x482d8394 in DBEnv_open (self=0x10316318, args=<optimized out>) at /home/sandu/work/Wrt/openwrt/build_dir/target-powerpc_8540_uClibc-0.9.33.2_ewok/Python-2.7.10/Modules/_bsddb.c:4950
#6  0x480f234c in ast_for_testlist_comp (c=<optimized out>, n=<optimized out>) at Python/ast.c:2148
```

Line 87 looks like this (the `fctnl()` call)
```
       /* Acquire an exclusive kernel lock. */
        k_lock.l_type = F_WRLCK;
        if (fcntl(env->lockfhp->fd, F_SETLKW, &k_lock))
            goto err;
```

Removing line 65 `--with-mutex=UNIX/fcntl \` from `CONFIGURE_ARGS` `libs/db47/Makefile` fixes the issue.
( See this commit https://github.com/commodo/packages/commit/a9271d80f472b8b4acd89efc34ae4be3ec2f9826 )
fcntl mutexes seem to have been removed after Berkley DB 4.8 anyway.

Adding @Naoir to the discussion, since he maintains the `libs/db47` package.
Would it be fine to remove that mutex support from db47 ?
If not, we can add db53 (slightly newer version) and I can maintain that for a while.
I wouldn't go with db61 just yet, since db53 is a bit bloat-ier than 4.7 anyway.

Unless there's a reasonable demand for it.
I'd go with the minimum stuff required to fix this segfault, and in this case remove fcntl support would be it.

Naoir, 2015-05-28:
Removing fcntl mutexes is fine with me. Note that db47 will default to pthread mutexes though. I would like to hardcode that to avoid stupid autodetection shenanigans, like Naoir@e42aa3526fcdfabc7f1fef816117c5f7cefe1190.

Can you confirm that this change indeed works as intended? All that I did was to check that python doesn't crash like in the case above and perl's DB_File tests still succeed.

commodo, 2015-05-29:
@FlorianFieber I think he would need to also confirm, as he is the reporter.
I also tested with @Naoir's patch and it works. Thanks :)

I don't use Berkley DB much, so I don't have test cases for it.
But, I should invest some time in packaging the Python tests and probably running them in a test env.
At the moment the Python tests are stripped from the python packages to reduce size.

FlorianFieber, 2015-05-30:
@commodo: I've compiled and tested it. Removing `--with-mutex=UNIX/fcntl` does resolve the problem, bsddb seems to work as expected.

Naoir, 2015-05-31:
Fixed in master. I'll prepare a pull request for CC. Thanks for reporting this and digging through the code ;).

commit 93bd063f (2015-05-31): db47: Force pthread mutexes, bump PKG_RELEASE

Fixes #1312.
PR #1328 (2015-05-31): db47: Force pthread mutexes, bump PKG_RELEASE [for-15.05]
The same thing that just went into master. Fixes #1312 for CC.
Thanks to Florian Fieber and commodo.
commit 541193dd (2015-05-31): db47: Force pthread mutexes, bump PKG_RELEASE

Fixes #1312.
closed 2015-05-31 by Naoir