openwrt/packages #1316: libtorrent error

taeasy, 2015-05-28:
OpenWrt-libtool: compile:  mipsel-openwrt-linux-uclibc-g++ -DHAVE_CONFIG_H -I. -I../.. -I. -I./.. -I../.. -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/include -I/home/main/brcm/staging_dir/toolchain-mipsel_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/toolchain-mipsel_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/include -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include -pthread -Os -pipe -mno-branch-likely -mips32 -mtune=mips32 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -DNDEBUG -Wall -std=c++11 -fvisibility=hidden -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include -MT socket_base.lo -MD -MP -MF .deps/socket_base.Tpo -c socket_base.cc -o socket_base.o >/dev/null 2>&1
mv -f .deps/socket_base.Tpo .deps/socket_base.Plo
/bin/bash ../../libtool  --tag=CXX   --mode=compile mipsel-openwrt-linux-uclibc-g++ -DHAVE_CONFIG_H -I. -I../..  -I. -I./.. -I../.. -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/include -I/home/main/brcm/staging_dir/toolchain-mipsel_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/toolchain-mipsel_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/include  -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include  -pthread -Os -pipe -mno-branch-likely -mips32 -mtune=mips32 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float  -DNDEBUG -Wall -std=c++11 -fvisibility=hidden -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include  -MT socket_datagram.lo -MD -MP -MF .deps/socket_datagram.Tpo -c -o socket_datagram.lo socket_datagram.cc
OpenWrt-libtool: compile:  mipsel-openwrt-linux-uclibc-g++ -DHAVE_CONFIG_H -I. -I../.. -I. -I./.. -I../.. -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/include -I/home/main/brcm/staging_dir/toolchain-mipsel_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/toolchain-mipsel_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/include -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include -pthread -Os -pipe -mno-branch-likely -mips32 -mtune=mips32 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -DNDEBUG -Wall -std=c++11 -fvisibility=hidden -I/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include -MT socket_datagram.lo -MD -MP -MF .deps/socket_datagram.Tpo -c socket_datagram.cc  -fPIC -DPIC -o .libs/socket_datagram.o
socket_datagram.cc: In member function 'int torrent::SocketDatagram::write_datagram(const void*, unsigned int, rak::socket_address*)':
socket_datagram.cc:77:9: error: 'm_ipv6_socket' was not declared in this scope
     if (m_ipv6_socket && sa->family() == rak::socket_address::pf_inet) {
         ^
socket_datagram.cc:90:65: error: invalid use of incomplete type 'class rak::socket_address_inet6'
       r = ::sendto(m_fileDesc, buffer, length, 0, sa->sa_inet6()->c_sockaddr(), sizeof(rak::socket_address_inet6));
                                                                 ^
In file included from socket_datagram.cc:44:0:
../../rak/socket_address.h:61:7: error: forward declaration of 'class rak::socket_address_inet6'
 class socket_address_inet6;
       ^
socket_datagram.cc:90:113: error: invalid application of 'sizeof' to incomplete type 'rak::socket_address_inet6'
       r = ::sendto(m_fileDesc, buffer, length, 0, sa->sa_inet6()->c_sockaddr(), sizeof(rak::socket_address_inet6));
                                                                                                                 ^
make[7]: *** [socket_datagram.lo] Error 1
make[7]: Leaving directory `/home/main/brcm/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/libtorrent-0.13.4-git-1/src/net'
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory `/home/main/brcm/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/libtorrent-0.13.4-git-1/src'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/main/brcm/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/libtorrent-0.13.4-git-1'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/main/brcm/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/libtorrent-0.13.4-git-1'
make[3]: *** [/home/main/brcm/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/libtorrent-0.13.4-git-1/.built] Error 2
make[3]: Leaving directory `/home/main/brcm/feeds/packages/libs/libtorrent'
make[2]: *** [package/feeds/packages/libtorrent/compile] Error 2
make[2]: Leaving directory `/home/main/brcm'
make[1]: *** [/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/brcm'
make: *** [world] Error 2
main@bogon:~/brcm$

ryzhovau, 2015-05-31:
Please, replace `$(call autoconf_bool,CONFIG_IPV6,ipv6)` to `--disable-ipv6` in `feeds/packages/net/rtorrent/Makefile` and `feeds/packages/libs/libtorrent/Makefile` and provide a feedback.
Looks like [this PR](https://github.com/openwrt/packages/pull/1181) brings this trouble.

What toolchain are you using? What target? I'm [using](https://github.com/Entware/entware/blob/master/packages/rtorrent/fix-path.patch#L33) external toolchain to compile rtorrent\libtorrent with IPv6 support.

closed 2015-06-10 by taeasy

ryzhovau, 2015-06-10:
So, how did you solve it?

FireCulex, 2015-09-07:
Well, apparently this issue still occurs 3-4 months later. I just deleted the 120-fix-ipv6_socket_datagram.patch and got it to compile