openwrt/packages #1321: snort not build can use CONFIG_PKG_CHECK_FORMAT_SECURITY=y

build000, 2015-05-29:
slawek@slawek-ubuntu:/projekty/openwrt/trunk$ make package/snort/compile V=s
make[1]: Entering directory '/projekty/openwrt/trunk'
make[2]: Entering directory '/projekty/openwrt/trunk/package/libs/toolchain'
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libssp" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libpthread" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "librt" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/toolchain.default.install
make[2]: Leaving directory '/projekty/openwrt/trunk/package/libs/toolchain'
make[2]: Entering directory '/projekty/openwrt/trunk/feeds/packages/libs/libdnet'
make[2]: Leaving directory '/projekty/openwrt/trunk/feeds/packages/libs/libdnet'
make[2]: Entering directory '/projekty/openwrt/trunk/package/libs/libpcap'
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/libpcap.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/libpcap.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/libpcap.default.install.clean; fi; echo "libpcap" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/libpcap.default.install
make[2]: Leaving directory '/projekty/openwrt/trunk/package/libs/libpcap'
make[2]: Entering directory '/projekty/openwrt/trunk/feeds/packages/libs/libdaq'
make[2]: Leaving directory '/projekty/openwrt/trunk/feeds/packages/libs/libdaq'
make[2]: Entering directory '/projekty/openwrt/trunk/feeds/packages/libs/pcre'
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/pcre.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/pcre.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/pcre.default.install.clean; fi; echo "libpcre" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/pcre.default.install
make[2]: Leaving directory '/projekty/openwrt/trunk/feeds/packages/libs/pcre'
make[2]: Entering directory '/projekty/openwrt/trunk/package/libs/librpc'
make[2]: Leaving directory '/projekty/openwrt/trunk/package/libs/librpc'
make[2]: Entering directory '/projekty/openwrt/trunk/package/libs/ocf-crypto-headers'
make[2]: Leaving directory '/projekty/openwrt/trunk/package/libs/ocf-crypto-headers'
make[2]: Entering directory '/projekty/openwrt/trunk/package/libs/zlib'
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/zlib.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/zlib.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/zlib.default.install.clean; fi; echo "zlib" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/zlib.default.install
make[2]: Leaving directory '/projekty/openwrt/trunk/package/libs/zlib'
make[2]: Entering directory '/projekty/openwrt/trunk/package/libs/openssl'
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/openssl.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/openssl.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/openssl.default.install.clean; fi; echo "libopenssl" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/openssl.default.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/openssl.default.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/openssl.default.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/openssl.default.install.clean; fi; echo "openssl-util" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/openssl.default.install
make[2]: Leaving directory '/projekty/openwrt/trunk/package/libs/openssl'
make[2]: Entering directory '/projekty/openwrt/trunk/package/libs/ncurses'
make[2]: Leaving directory '/projekty/openwrt/trunk/package/libs/ncurses'
make[2]: Entering directory '/projekty/openwrt/trunk/package/libs/ncurses'
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean; fi; echo "terminfo" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean; fi; echo "libncurses" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install
make[2]: Leaving directory '/projekty/openwrt/trunk/package/libs/ncurses'
make[2]: Entering directory '/projekty/openwrt/trunk/package/libs/ncurses'
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean; fi; echo "terminfo" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install
if [ -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean ]; then rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install.clean; fi; echo "libncurses" >> /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/pkginfo/ncurses.libncurses.install
make[2]: Leaving directory '/projekty/openwrt/trunk/package/libs/ncurses'
[... cut ...]
rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.snort_installed
rm -f /projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/packages/snort.list /projekty/openwrt/trunk/staging_dir/host/packages/snort.list
rm -rf /projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/snort/snort-2.9.7.2
make[2]: Leaving directory '/projekty/openwrt/trunk/feeds/packages/net/snort'
make[1]: Leaving directory '/projekty/openwrt/trunk'
slawek@slawek-ubuntu:/projekty/openwrt/trunk$

build000, 2015-05-29:
my fragment .config :

#
# Hardening build options
#
CONFIG_PKG_CHECK_FORMAT_SECURITY=y
# CONFIG_PKG_CC_STACKPROTECTOR_NONE is not set
CONFIG_PKG_CC_STACKPROTECTOR_REGULAR=y
# CONFIG_KERNEL_CC_STACKPROTECTOR_NONE is not set
CONFIG_KERNEL_CC_STACKPROTECTOR_REGULAR=y
# CONFIG_PKG_FORTIFY_SOURCE_NONE is not set
# CONFIG_PKG_FORTIFY_SOURCE_1 is not set
CONFIG_PKG_FORTIFY_SOURCE_2=y
# CONFIG_PKG_RELRO_NONE is not set
# CONFIG_PKG_RELRO_PARTIAL is not set
CONFIG_PKG_RELRO_FULL=y

hnyman, 2016-08-26:
snort compiles currently ok in trunk buildbot with CONFIG_PKG_CHECK_FORMAT_SECURITY=y

closed 2016-08-26 by hnyman