openwrt/packages #1322: php5-pecl-libevent compile error

taeasy, 2015-05-29:
In file included from /home/main/x86/staging_dir/host/usr/include/php/main/php.h:406:0,
                 from /home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/libevent-0.1.0/libevent.c:26:
/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/libevent-0.1.0/libevent.c: In function 'zm_startup_libevent':
/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/libevent-0.1.0/libevent.c:1403:42: error: 'EVBUFFER_READ' undeclared (first use in this function)
  REGISTER_LONG_CONSTANT("EVBUFFER_READ", EVBUFFER_READ, CONST_CS | CONST_PERSISTENT);
                                          ^
/home/main/x86/staging_dir/host/usr/include/php/Zend/zend_constants.h:43:103: note: in definition of macro 'REGISTER_LONG_CONSTANT'
 #define REGISTER_LONG_CONSTANT(name, lval, flags)  zend_register_long_constant((name), sizeof(name), (lval), (flags), module_number TSRMLS_CC)
                                                                                                       ^
/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/libevent-0.1.0/libevent.c:1403:42: note: each undeclared identifier is reported only once for each function it appears in
  REGISTER_LONG_CONSTANT("EVBUFFER_READ", EVBUFFER_READ, CONST_CS | CONST_PERSISTENT);
                                          ^
/home/main/x86/staging_dir/host/usr/include/php/Zend/zend_constants.h:43:103: note: in definition of macro 'REGISTER_LONG_CONSTANT'
 #define REGISTER_LONG_CONSTANT(name, lval, flags)  zend_register_long_constant((name), sizeof(name), (lval), (flags), module_number TSRMLS_CC)
                                                                                                       ^
/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/libevent-0.1.0/libevent.c:1404:43: error: 'EVBUFFER_WRITE' undeclared (first use in this function)
  REGISTER_LONG_CONSTANT("EVBUFFER_WRITE", EVBUFFER_WRITE, CONST_CS | CONST_PERSISTENT);
                                           ^
/home/main/x86/staging_dir/host/usr/include/php/Zend/zend_constants.h:43:103: note: in definition of macro 'REGISTER_LONG_CONSTANT'
 #define REGISTER_LONG_CONSTANT(name, lval, flags)  zend_register_long_constant((name), sizeof(name), (lval), (flags), module_number TSRMLS_CC)
                                                                                                       ^
/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/libevent-0.1.0/libevent.c:1405:41: error: 'EVBUFFER_EOF' undeclared (first use in this function)
  REGISTER_LONG_CONSTANT("EVBUFFER_EOF", EVBUFFER_EOF, CONST_CS | CONST_PERSISTENT);
                                         ^
/home/main/x86/staging_dir/host/usr/include/php/Zend/zend_constants.h:43:103: note: in definition of macro 'REGISTER_LONG_CONSTANT'
 #define REGISTER_LONG_CONSTANT(name, lval, flags)  zend_register_long_constant((name), sizeof(name), (lval), (flags), module_number TSRMLS_CC)
                                                                                                       ^
/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/libevent-0.1.0/libevent.c:1406:43: error: 'EVBUFFER_ERROR' undeclared (first use in this function)
  REGISTER_LONG_CONSTANT("EVBUFFER_ERROR", EVBUFFER_ERROR, CONST_CS | CONST_PERSISTENT);
                                           ^
/home/main/x86/staging_dir/host/usr/include/php/Zend/zend_constants.h:43:103: note: in definition of macro 'REGISTER_LONG_CONSTANT'
 #define REGISTER_LONG_CONSTANT(name, lval, flags)  zend_register_long_constant((name), sizeof(name), (lval), (flags), module_number TSRMLS_CC)
                                                                                                       ^
/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/libevent-0.1.0/libevent.c:1407:45: error: 'EVBUFFER_TIMEOUT' undeclared (first use in this function)
  REGISTER_LONG_CONSTANT("EVBUFFER_TIMEOUT", EVBUFFER_TIMEOUT, CONST_CS | CONST_PERSISTENT);
                                             ^
/home/main/x86/staging_dir/host/usr/include/php/Zend/zend_constants.h:43:103: note: in definition of macro 'REGISTER_LONG_CONSTANT'
 #define REGISTER_LONG_CONSTANT(name, lval, flags)  zend_register_long_constant((name), sizeof(name), (lval), (flags), module_number TSRMLS_CC)
                                                                                                       ^
make[4]: *** [libevent.lo] Error 1
make[4]: Leaving directory `/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/libevent-0.1.0'
make[3]: *** [/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/libevent-0.1.0/.built] Error 2
make[3]: Leaving directory `/home/main/x86/feeds/packages/lang/php5-pecl-libevent'
make[2]: *** [package/feeds/packages/php5-pecl-libevent/compile] Error 2
make[2]: Leaving directory `/home/main/x86'
make[1]: *** [/home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/x86'
make: *** [world] Error 2
main@bogon:~/x86$

mhei, 2015-05-29:
Hm, I see the buildbots failing with the same error, too. But I cannot reproduce it here at home on my machine. Could you please send me your .config via mail?

taeasy, 2015-05-30:
my machine is debian x86_64 in vmware, i have selected all packages by default.

mhei, 2015-05-30:
Does it work for you now? Would be nice if you could post the output of `scripts/diffconfig.sh` I would like to fix the buildbots, but I do not have any clue. I'm compiling under and older Kubuntu, but I could setupa a Debian VM. But the host distribution shouldn't have any influence...

taeasy, 2015-05-31:
hi, the problem still, i have output build log.

http://104.224.133.76/output.txt

taeasy, 2015-05-31:
scripts/diffconfig.sh:

#!/bin/sh
grep \^CONFIG_TARGET_ .config | head -n3 > tmp/.diffconfig.head
grep '^CONFIG_ALL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_DEVEL=y' .config >> tmp/.diffconfig.head
grep '^CONFIG_TOOLCHAINOPTS=y' .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage1 Config.in >/dev/null
./scripts/kconfig.pl '>+' tmp/.diffconfig.stage1 .config >> tmp/.diffconfig.head
./scripts/config/conf --defconfig=tmp/.diffconfig.head -w tmp/.diffconfig.stage2 Config.in >/dev/null
./scripts/kconfig.pl '>' tmp/.diffconfig.stage2 .config >> tmp/.diffconfig.head
cat tmp/.diffconfig.head
rm -f tmp/.diffconfig tmp/.diffconfig.head

.config link: http://104.224.133.76/config.txt

mhei, 2015-05-31:
Thanks for the config, but your config builds fine on my system here :cry: I uploaded a patch to force the use of libevent2 headers, please give it a try... Just a guess...

taeasy, 2015-06-01:
working now, thanks

closed 2015-06-01 by taeasy