openwrt/packages #1324: seafile-ccnet compile error

taeasy, 2015-05-30:
-MT packet-io.o -MD -MP -MF .deps/packet-io.Tpo -c -o packet-io.o `test -f '../common/packet-io.c' || echo './'`../common/packet-io.c
../common/packet-io.c: In function 'canReadWrapper':
../common/packet-io.c:62:5: warning: implicit declaration of function 'EVBUFFER_LENGTH' [-Wimplicit-function-declaration]
     g_return_if_fail (EVBUFFER_LENGTH (e->input) >= CCNET_PACKET_LENGTH_HEADER);
     ^
In file included from /home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include/glib-2.0/glibconfig.h:9:0,
                 from /home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include/glib-2.0/glib/gtypes.h:32,
                 from /home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include/glib-2.0/glib/galloca.h:32,
                 from /home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include/glib-2.0/glib.h:30,
                 from ../../include/ccnet/option.h:7,
                 from ../common/common.h:18,
                 from ../common/packet-io.c:3:
../common/packet-io.c:62:41: error: dereferencing pointer to incomplete type
     g_return_if_fail (EVBUFFER_LENGTH (e->input) >= CCNET_PACKET_LENGTH_HEADER);
                                         ^
/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include/glib-2.0/glib/gmacros.h:319:8: note: in definition of macro '_G_BOOLEAN_EXPR'
    if (expr)                                    \
        ^
/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/usr/include/glib-2.0/glib/gmessages.h:368:9: note: in expansion of macro 'G_LIKELY'
      if G_LIKELY(expr) { } else            \
         ^
../common/packet-io.c:70:9: warning: implicit declaration of function 'EVBUFFER_DATA' [-Wimplicit-function-declaration]
         packet = (ccnet_packet *) EVBUFFER_DATA (e->input);
         ^
../common/packet-io.c:70:51: error: dereferencing pointer to incomplete type
         packet = (ccnet_packet *) EVBUFFER_DATA (e->input);
                                                   ^
../common/packet-io.c:77:31: error: dereferencing pointer to incomplete type
         if (EVBUFFER_LENGTH (e->input) - CCNET_PACKET_LENGTH_HEADER < len)
                               ^
../common/packet-io.c:93:9: warning: implicit declaration of function 'evbuffer_drain' [-Wimplicit-function-declaration]
         evbuffer_drain (e->input, len + CCNET_PACKET_LENGTH_HEADER);
         ^
../common/packet-io.c:93:26: error: dereferencing pointer to incomplete type
         evbuffer_drain (e->input, len + CCNET_PACKET_LENGTH_HEADER);
                          ^
../common/packet-io.c:95:29: error: dereferencing pointer to incomplete type
         if(EVBUFFER_LENGTH(e->input) >= CCNET_PACKET_LENGTH_HEADER)
                             ^
../common/packet-io.c: In function 'ccnet_packet_io_new':
../common/packet-io.c:133:5: warning: implicit declaration of function 'bufferevent_socket_new' [-Wimplicit-function-declaration]
     io->bufev = bufferevent_socket_new (NULL, io->socket, BEV_OPT_CLOSE_ON_FREE);
     ^
../common/packet-io.c:133:59: error: 'BEV_OPT_CLOSE_ON_FREE' undeclared (first use in this function)
     io->bufev = bufferevent_socket_new (NULL, io->socket, BEV_OPT_CLOSE_ON_FREE);
                                                           ^
../common/packet-io.c:133:59: note: each undeclared identifier is reported only once for each function it appears in
../common/packet-io.c:134:5: warning: implicit declaration of function 'bufferevent_setcb' [-Wimplicit-function-declaration]
     bufferevent_setcb (io->bufev, canReadWrapper,
     ^
../common/packet-io.c:136:5: warning: implicit declaration of function 'bufferevent_enable' [-Wimplicit-function-declaration]
     bufferevent_enable (io->bufev, EV_READ | EV_WRITE);
     ^
../common/packet-io.c: In function 'ccnet_packet_io_free':
../common/packet-io.c:205:9: warning: implicit declaration of function 'bufferevent_free' [-Wimplicit-function-declaration]
         bufferevent_free (io->bufev);
         ^
../common/packet-io.c: In function 'ccnet_packet_io_try_read':
../common/packet-io.c:222:33: error: dereferencing pointer to incomplete type
     if(EVBUFFER_LENGTH(io->bufev->input))
                                 ^
../common/packet-io.c: In function 'ccnet_packet_io_set_timeout_secs':
../common/packet-io.c:250:5: warning: implicit declaration of function 'bufferevent_settimeout' [-Wimplicit-function-declaration]
     bufferevent_settimeout (io->bufev, io->timeout, io->timeout);
     ^
../common/packet-io.c:252:9: warning: implicit declaration of function 'bufferevent_disable' [-Wimplicit-function-declaration]
         bufferevent_disable (io->bufev, EV_READ | EV_WRITE);
         ^
../common/packet-io.c: In function 'ccnet_packet_io_write_packet':
../common/packet-io.c:273:5: warning: implicit declaration of function 'bufferevent_write' [-Wimplicit-function-declaration]
     bufferevent_write (io->bufev, packet, len);
     ^
make[7]: *** [packet-io.o] Error 1
make[7]: Leaving directory `/home/main/brcm/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/seafile-ccnet-4.1.2/net/server'
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory `/home/main/brcm/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/seafile-ccnet-4.1.2/net'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/main/brcm/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/seafile-ccnet-4.1.2'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/main/brcm/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/seafile-ccnet-4.1.2'
make[3]: *** [/home/main/brcm/build_dir/target-mipsel_mips32_uClibc-0.9.33.2/seafile-ccnet-4.1.2/.built] Error 2
make[3]: Leaving directory `/home/main/brcm/feeds/packages/net/seafile-ccnet'
make[2]: *** [package/feeds/packages/seafile-ccnet/compile] Error 2
make[2]: Leaving directory `/home/main/brcm'
make[1]: *** [/home/main/brcm/staging_dir/target-mipsel_mips32_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/main/brcm'
make: *** [world] Error 2

kissg1988, 2015-06-14:
ccnet builds fine for me using the latest sources from master. Is this still an issue?

taeasy, 2015-08-29:
yes the same problem still.

kissg1988, 2015-08-29:
@taeasy which branch of openwrt are you using and what's the exact architecture (I can see it's brcm but not sure which variant)? I'd like to reproduce the issue but couldn't so far.

I've just built seafile for brcm63xx using the latest revision of the master branch and could not see any errors.

Also, please note that the packages are not backwards compatible (you can't build them against the 14.07 branch).

plntyk, 2015-08-31:
please provide the output of ./scripts/diffconfig.sh
Stating the system/variant you are using as a host might help too in some cases.

kissg1988, 2015-09-03:
I'm still not able to reproduce this on my own buildenv but I can see the very same issue with another package, as well:

http://buildbot.openwrt.org:8010/broken_packages/ar71xx.nand/tmux/compile.txt

Both packages build fine in my environment (CentOS 7 x86_64).

The missing compiler directives are declared in event2/buffer_compat.h so including this file manually might help but I still don't understand why I can't reproduce it...

Any ideas?

jow-, 2015-09-03:
OpenWrt provides two libevent packages, one libevent2 in the base and one libevent 1.4.x from the packages feed. I bet the problem only occurs if both libevents are enabled and built in the same environment (as it is the case for buildbots).

kissg1988, 2015-09-03:
I've just built all libevent packages and still could not reproduce the issue. Tried building one after the other and vice-versa, cleaned the source tree multiple times, enabled full language support and recompiled but still no signs of the issue.

Taking a look at the source file, I can see that event2/buffer_compat.h is always included, either directly (when building for *BSD) or through the event.h file. In case it cannot be included for some reason (eg, it's not found in the include search paths), gcc should fail with a different error message, right?

I'm going to create a patch soon which might fix this error. The plan is to include header files from "event2/" directory rather than through the old, deprecated "event.h" file.Hope this will help...

closed 2015-09-07 by sbyx

kissg1988, 2015-09-13:
Unfortunately, the build still fails. A recent patch merged 3 days ago (#1765) should fix this but the build still fails:

http://buildbot.openwrt.org:8010/broken_packages/ar71xx.nand/seafile-ccnet/compile.txt

I've double checked and there are no more includes for evdns.h.

As far as I can tell, the build system does not checkout the latest version from Git until "./scripts/feeds update -a" is run and uses the pre-generated gzipped package instead. Is this normal behaviour? Is there any way to force buildbot slaves to checkout the package from git?

PR #1754 (2015-09-07): seafile-ccnet: use correct libevent2 includes
Hopefully this would fix #1324.

Including `<event.h>` and `<evutil.h>` directly might not work as expected as those are deprecated header files in libevent2:

`The <event.h> header is deprecated in Libevent 2.0 and later; please use <event2/event.h> instead.`