openwrt/packages #1327: davfs2 can use full language suport - error build

build000, 2015-05-31:
make[3]: Entering directory '/projekty/openwrt/trunk/feeds/packages/net/davfs2'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include  -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include  -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include " LDFLAGS="-L/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib " make -j1 -C /projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/. AR="mips-openwrt-linux-uclibc-gcc-ar" AS="mips-openwrt-linux-uclibc-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include" LD=mips-openwrt-linux-uclibc-ld NM="mips-openwrt-linux-uclibc-gcc-nm" CC="mips-openwrt-linux-uclibc-gcc" GCC="mips-openwrt-linux-uclibc-gcc" CXX="mips-openwrt-linux-uclibc-g++" RANLIB="mips-openwrt-linux-uclibc-gcc-ranlib" STRIP=mips-openwrt-linux-uclibc-strip OBJCOPY=mips-openwrt-linux-uclibc-objcopy OBJDUMP=mips-openwrt-linux-uclibc-objdump SIZE=mips-openwrt-linux-uclibc-size CROSS="mips-openwrt-linux-uclibc-" ARCH="mips" ;
make[4]: Entering directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2'
make  all-recursive
make[5]: Entering directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2'
Making all in po
make[6]: Entering directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/po'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/po'
Making all in etc
make[6]: Entering directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/etc'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/etc'
Making all in man
make[6]: Entering directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/man'
Making all in de
make[7]: Entering directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/man/de'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/man/de'
Making all in es
make[7]: Entering directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/man/es'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/man/es'
make[7]: Entering directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/man'
make[7]: Nothing to be done for 'all-am'.
make[7]: Leaving directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/man'
make[6]: Leaving directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/man'
Making all in src
make[6]: Entering directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/src'
mips-openwrt-linux-uclibc-gcc -DPROGRAM_NAME=\"mount.davfs\" -DDAV_SYS_CONF_DIR=\"/etc/davfs2\" -DDAV_LOCALSTATE_DIR=\"/var/run\" -DDAV_SYS_RUN=\"/var/run/mount.davfs\" -DDAV_SYS_CACHE=\"/var/cache/davfs2\" -DDAV_SECRETS=\"secrets\" -DDAV_CONFIG=\"davfs2.conf\" -DDAV_CERTS_DIR=\"certs\" -DDAV_CLICERTS_DIR=\"private\" -DDAV_DATA_DIR=\"/usr/share/davfs2\" -DLOCALEDIR=\"/usr/share/locale\" -DDAV_USER=\"davfs2\" -DDAV_GROUP=\"davfs2\" -D_FORTIFY_SOURCE=2 -DHAVE_CONFIG_H -I. -I..   -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include  -Wall -Werror=format-security  -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include  -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include/neon -MT webdav.o -MD -MP -MF .deps/webdav.Tpo -c -o webdav.o webdav.c
webdav.c: In function 'dav_conv_from_utf_8':
webdav.c:583:9: error: 'from_utf_8' undeclared (first use in this function)
     if (from_utf_8)
         ^
webdav.c:583:9: note: each undeclared identifier is reported only once for each function it appears in
webdav.c:584:9: warning: implicit declaration of function 'convert' [-Wimplicit-function-declaration]
         convert(&new, from_utf_8);
         ^
webdav.c: In function 'dav_conv_to_utf_8':
webdav.c:595:9: error: 'to_utf_8' undeclared (first use in this function)
     if (to_utf_8)
         ^
webdav.c: In function 'dav_conv_from_server_enc':
webdav.c:607:9: error: 'from_server_enc' undeclared (first use in this function)
     if (from_server_enc)
         ^
webdav.c: In function 'dav_conv_to_server_enc':
webdav.c:619:9: error: 'to_server_enc' undeclared (first use in this function)
     if (to_server_enc)
         ^
webdav.c: At top level:
webdav.c:1269:19: error: unknown type name 'iconv_t'
 convert(char **s, iconv_t conv)
                   ^
webdav.c: In function 'prop_result':
webdav.c:1892:13: error: 'from_server_enc' undeclared (first use in this function)
         if (from_server_enc)
             ^
Makefile:423: recipe for target 'webdav.o' failed
make[6]: *** [webdav.o] Error 1
make[6]: Leaving directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/src'
Makefile:461: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2'
Makefile:381: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2'
Makefile:69: recipe for target '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/.built' failed
make[3]: *** [/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/.built] Error 2
make[3]: Leaving directory '/projekty/openwrt/trunk/feeds/packages/net/davfs2'
package/Makefile:187: recipe for target 'package/feeds/packages/davfs2/compile' failed
make[2]: *** [package/feeds/packages/davfs2/compile] Error 2
make[2]: Leaving directory '/projekty/openwrt/trunk'
package/Makefile:184: recipe for target '/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/projekty/openwrt/trunk'
/projekty/openwrt/trunk/include/toplevel.mk:181: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
slawek@slawek-ubuntu:/projekty/openwrt/trunk$

ryzhovau, 2015-05-31:
[See how](https://github.com/Entware/entware/blob/master/packages/davfs2/fix-path.patch#L16) to add NLS support.
```
DEPENDS=...  $(INTL_DEPENDS) $(ICONV_DEPENDS)
...
CONFIGURE_ARGS += \
	--with-libintl-prefix=$(INTL_PREFIX) \
	--with-libiconv-prefix=$(ICONV_PREFIX) \
```

build000, 2015-05-31:
Sorry, but the tips do not work (the same mistakes as in the first post) - in this case the question - where, what and how to change the Makefile to build a package that davfs2 ? Maybe not enough to change the Makefile itself and requires deeper changes in the sources (the davfs2 it for example) ?

ryzhovau, 2015-06-01:
I'm successfully building davfs2 with iconv_full, see link on my previous comment.

build000, 2015-06-01:
It remains latter issue / question: whether to build or not that davfs2 may affect the use of protection (...- Wformat -Werror = format-security -fstack-protector -D_FORTIFY_SOURCE = 2 -Wl, -z, now -Wl, - with, relro ...) and "full language support" at the same time (eg. when activated, the protection does not build me libdb47) ???

taeasy, 2015-09-01:
I have same problem:

x/staging_dir/target-mips_34kc_musl-1.1.10/usr/include/neon -MT webdav.o -MD -MP -MF .deps/webdav.Tpo -c -o webdav.o webdav.c
<command-line>:0:0: warning: "_FORTIFY_SOURCE" redefined [enabled by default]
<command-line>:0:0: note: this is the location of the previous definition
webdav.c: In function 'dav_conv_from_utf_8':
webdav.c:587:9: error: 'from_utf_8' undeclared (first use in this function)
     if (from_utf_8)
         ^
webdav.c:587:9: note: each undeclared identifier is reported only once for each function it appears in
webdav.c:588:9: warning: implicit declaration of function 'convert' [-Wimplicit-function-declaration]
         convert(&new, from_utf_8);
         ^
webdav.c: In function 'dav_conv_to_utf_8':
webdav.c:599:9: error: 'to_utf_8' undeclared (first use in this function)
     if (to_utf_8)
         ^
webdav.c: In function 'dav_conv_from_server_enc':
webdav.c:611:9: error: 'from_server_enc' undeclared (first use in this function)
     if (from_server_enc)
         ^
webdav.c: In function 'dav_conv_to_server_enc':
webdav.c:623:9: error: 'to_server_enc' undeclared (first use in this function)
     if (to_server_enc)
         ^
webdav.c: At top level:
webdav.c:1273:19: error: unknown type name 'iconv_t'
 convert(char **s, iconv_t conv)
                   ^
webdav.c: In function 'prop_result':
webdav.c:1896:13: error: 'from_server_enc' undeclared (first use in this function)
         if (from_server_enc)
             ^
In file included from webdav.c:22:0:
webdav.c: At top level:
compat.h:51:15: warning: 'canonicalize_file_name' defined but not used [-Wunused-function]
 static char * canonicalize_file_name(const char *path)
               ^
webdav.c:1544:1: warning: 'log_writer' defined but not used [-Wunused-function]
 log_writer(void *cookie, const char *buffer, size_t size)
 ^
make[5]: *** [webdav.o] Error 1
make[5]: Leaving directory `/home/main/ar71xx/build_dir/target-mips_34kc_musl-1.1.10/davfs2-1.5.2/src'
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory `/home/main/ar71xx/build_dir/target-mips_34kc_musl-1.1.10/davfs2-1.5.2'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/main/ar71xx/build_dir/target-mips_34kc_musl-1.1.10/davfs2-1.5.2'
make[2]: *** [/home/main/ar71xx/build_dir/target-mips_34kc_musl-1.1.10/davfs2-1.5.2/.built] Error 2
make[2]: Leaving directory `/home/main/ar71xx/feeds/packages/net/davfs2'
make[1]: *** [package/feeds/packages/davfs2/compile] Error 2
make[1]: Leaving directory `/home/main/ar71xx'
make: *** [package/davfs2/compile] Error 2

diizzyy, 2017-06-07:
@thess @hnyman 
This seems to be fixed, close?

fededim, 2017-06-07:
Wait the PR #2149 has not been merged

fededim, 2017-06-07:
New PR #4458 created, please merge it and close the issue

diizzyy, 2017-06-13:
@fededim 
We can close this one then?

build000, 2017-06-13:
confirm - eg. using last LEDE build (```make -j8 BUILD_LOG=1```) :
```
dom@machina:~/Pulpit/WNDR4700-LEDE-REBOOT$ scripts/getver.sh
r4407-a3332f2 
```
using eq. .config: https://www.wklej.to/A6TKZ (```CONFIG_BUILD_NLS=y```)
All firmware build properly an interest log view like this (logs/package/feeds/packages/davfs2/compile.txt):
```
+ curl -f --connect-timeout 20 --retry 5 --location --insecure http://download.savannah.gnu.org/releases/davfs2/davfs2-1.5.4.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0

  2  467k    2 14098    0     0  23288      0  0:00:20 --:--:--  0:00:20 23288
100  467k  100  467k    0     0   562k      0 --:--:-- --:--:-- --:--:-- 2024k

Applying ./patches/010-main_code_fix.patch using plaintext: 
patching file src/cache.c
patching file src/webdav.c
Hunk #1 succeeded at 2035 (offset 2 lines).

Applying ./patches/100-musl-compat.patch using plaintext: 
patching file configure.ac
patching file src/cache.c
patching file src/dav_fuse.c
patching file src/kernel_interface.c
patching file src/mount_davfs.c
patching file src/umount_davfs.c
patching file src/webdav.c
patching file src/compat.h

Applying ./patches/200-davfs2-1.5.4-no-forced-stack-protector.patch using plaintext: 
patching file src/Makefile.am

Applying ./patches/300-have_iconv_h.patch using plaintext: 
patching file src/webdav.c
Copying file ABOUT-NLS
Copying file config/config.rpath
Copying file config/codeset.m4
Copying file config/extern-inline.m4
Copying file config/fcntl-o.m4
Copying file config/gettext.m4
Copying file config/glibc2.m4
Copying file config/glibc21.m4
Copying file config/iconv.m4
Copying file config/intdiv0.m4
Copying file config/intl.m4
Copying file config/intldir.m4
Copying file config/intlmacosx.m4
Copying file config/intmax.m4
Copying file config/inttypes-pri.m4
Copying file config/inttypes_h.m4
Copying file config/lcmessage.m4
Copying file config/lib-ld.m4
Copying file config/lib-link.m4
Copying file config/lock.m4
Copying file config/longlong.m4
Copying file config/po.m4
Copying file config/printf-posix.m4
Copying file config/progtest.m4
Copying file config/size_max.m4
Copying file config/stdint_h.m4
Copying file config/threadlib.m4
Copying file config/uintmax_t.m4
Copying file config/visibility.m4
Copying file config/wchar_t.m4
Copying file config/wint_t.m4
Copying file config/xsize.m4
[... 360 lines cut ...]
               ^
webdav.c:1546:1: warning: 'log_writer' defined but not used [-Wunused-function]
 log_writer(void *cookie, const char *buffer, size_t size)
 ^
mv -f .deps/webdav.Tpo .deps/webdav.Po
powerpc-openwrt-linux-musl-gcc -Wall -Werror=format-security -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/build_dir/target-powerpc_464fp_musl/davfs2-1.5.4:davfs2-1.5.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-full/include -I/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-full/include -I/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/include   -I/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/include/neon  -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/lib -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/lib -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/usr/lib -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/lib -znow -zrelro -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-full/lib -Wl,-rpath-link=/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-full/lib -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-full/lib -Wl,-rpath-link=/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-full/lib -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/usr/lib -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/lib -lintl -liconv -o mount.davfs cache.o dav_coda.o dav_fuse.o kernel_interface.o mount_davfs.o webdav.o  -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/lib -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/lib -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/usr/lib -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/lib -znow -zrelro -lcrypto -lssl -lneon -lz -L/home/dom/Pulpit/WNDR4700-LEDE-REBOOT/staging_dir/target-powerpc_464fp_musl/usr/lib -lssl -lcrypto -lexpat 
powerpc-openwrt-linux-musl-gcc -DPROGRAM_NAME=\"mount.davfs\" -DDAV_SYS_CONF_DIR=\"/etc/davfs2\" -DDAV_LOCALSTATE_DIR=\"/var/run\" -DDAV_SYS_RUN=\"/var/run/mount.davfs\" -DDAV_SYS_CACHE=\"/var/cache/davfs2\" -DDAV_SECRETS=\"secrets\" -DDAV_CONFIG=\"da [... cut ...]
rstrip.sh: /home/dom/Pulpit/WNDR4700-LEDE-REBOOT/build_dir/target-powerpc_464fp_musl/davfs2-1.5.4/ipkg-powerpc_464fp/davfs2/usr/sbin/umount.davfs: executable
rstrip.sh: /home/dom/Pulpit/WNDR4700-LEDE-REBOOT/build_dir/target-powerpc_464fp_musl/davfs2-1.5.4/ipkg-powerpc_464fp/davfs2/usr/sbin/mount.davfs: executable
Packaged contents of /home/dom/Pulpit/WNDR4700-LEDE-REBOOT/build_dir/target-powerpc_464fp_musl/davfs2-1.5.4/ipkg-powerpc_464fp/davfs2 into /home/dom/Pulpit/WNDR4700-LEDE-REBOOT/bin/packages/powerpc_464fp/packages/davfs2_1.5.4-3_powerpc_464fp.ipk
```
Probably problem fixed.

diizzyy, 2017-06-13:
@build000 
Close it then...

closed 2017-06-13 by build000