openwrt/packages #1331: db47 error build (for using last patches ?)

build000, 2015-05-31:
make[3]: Entering directory '/projekty/openwrt/trunk/feeds/packages/libs/db47'
make   -C /projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/db-4.7.25.NC/build_unix DESTDIR="/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/db-4.7.25.NC/ipkg-install" all
make[4]: Entering directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/db-4.7.25.NC/build_unix'
/bin/sh ./libtool --mode=compile mips-openwrt-linux-uclibc-gcc -c -I. -I../dist/.. -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include  -D_GNU_SOURCE -D_REENTRANT -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -fpic   ../dist/../repmgr/repmgr_net.c
 mips-openwrt-linux-uclibc-gcc -c -I. -I../dist/.. -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/projekty/openwrt/trunk/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include -D_GNU_SOURCE -D_REENTRANT -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -fpic ../dist/../repmgr/repmgr_net.c  -fPIC -DPIC -o .libs/repmgr_net.o
../dist/../repmgr/repmgr_net.c: In function '__repmgr_listen':
../dist/../repmgr/repmgr_net.c:1139:2: error: format not a string literal and no format arguments [-Werror=format-security]
  __db_err(env, ret, why);
  ^
cc1: some warnings being treated as errors
Makefile:1775: recipe for target 'repmgr_net.lo' failed
make[4]: *** [repmgr_net.lo] Error 1
make[4]: Leaving directory '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/db-4.7.25.NC/build_unix'
Makefile:99: recipe for target '/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/db-4.7.25.NC/.built' failed
make[3]: *** [/projekty/openwrt/trunk/build_dir/target-mips_34kc_uClibc-0.9.33.2/db-4.7.25.NC/.built] Error 2
make[3]: Leaving directory '/projekty/openwrt/trunk/feeds/packages/libs/db47'
package/Makefile:187: recipe for target 'package/feeds/packages/db47/compile' failed
make[2]: *** [package/feeds/packages/db47/compile] Error 2
make[2]: Leaving directory '/projekty/openwrt/trunk'
package/Makefile:184: recipe for target '/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/projekty/openwrt/trunk/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/projekty/openwrt/trunk'
/projekty/openwrt/trunk/include/toplevel.mk:181: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
slawek@slawek-ubuntu:/projekty/openwrt/trunk$

Naoir, 2015-06-01:
Well, the last commit enabled Replication Manager, which triggers this warning. The fix for master will be simple. For CC, please see #1337.

build000, 2015-06-01:
Txh. :dancers: 
In fact - now a built and ...  please do so now, by those who can, adding the patch to the main tree because no matter libdb47 multimedia, etc. in openwrt just looks... very bad ... ;)

build000, 2015-06-02:
Thx - innstalled from oficial sources openwrt - patch is this.

build000, 2015-06-02:
Pleas close this issue - problem fixed - https://github.com/openwrt/packages/commit/58d762d6f36aef4a2e5e644693720f3329f7a972

Naoir, 2015-06-05:
Fixed in CC and master. Closing.

Thanks.

commit 4c4c2dbd (2015-06-05): db47: Disable Replication Manager

Replication Manager accidentally got enabled by 541193d.
Fixes #1331.
closed 2015-06-05 by Naoir