openwrt/packages #1332: strongswan: 5.3.0 regression - ipsec won't start when build with CONFIG_USE_STRIP=y

damian-kaczkowski, 2015-05-31:
Since 5.3.0 I got error like this when issuing `ipsec start`:

`/usr/lib/ipsec/starter: can't load library 'libstrongswan.so.0'`

Seems that strongswan doesn't search `/usr/lib/ipsec/` for libraries and this is the cause of an error.

I have been always using `CONFIG_USE_STRIP=y` with 5.0, 5.1, 5.2 and never came across this problem.

Could you please look into this?

damian-kaczkowski, 2015-05-31:
Maybe related to this commit(?):
https://dev.openwrt.org/changeset/44377

damian-kaczkowski, 2015-05-31:
I have opened a ticket on openwrt trac too:
https://dev.openwrt.org/ticket/19771

diizzyy, 2017-06-07:
@stintel 
Ping

stintel, 2017-06-29:
After reading the trac ticket, I conclude this isn't a problem in strongswan. Additionally, it works for me on LEDE master.

closed 2017-06-29 by stintel