openwrt/packages #1334: xl2tp client fails to reconnect

duvi, 2015-06-01:
Using the latest version of the package.

I'm running an l2tp over ipsec client on an Openwrt box.

By default, it's connecting fine after a reboot. However, every now and then, when the l2tp client wants to reconnect, it can't.

logread shows forever:
```
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
Mon Jun  1 13:57:07 2015 daemon.warn xl2tpd[16255]: Y: Host name lookup failed for wqĐwqĐ. Trying to look again in 0 seconds.
```
```
root@OpenWrt:~# cat /etc/config/network
config 'interface' 'vma'
        option 'proto'    'l2tp'
        option 'server'   'myserver'
        option 'username' 'myusername'
        option 'password' 'mypassword'
        option 'mtu'      '1410'
        option 'keepalive' '12'
        option 'defaultroute' '0'
```
```
root@OpenWrt:~# ifstatus vma
{
        "up": false,
        "pending": true,
        "available": true,
        "autostart": true,
        "proto": "l2tp",
        "data": {

        }
}
```
```
root@OpenWrt:~# ps
16255 root       896 S    xl2tpd -D -l -p /var/run/xl2tpd.pid
29463 root      1448 S    {l2tp.sh} /bin/sh ./l2tp.sh l2tp setup vma {"proto":"l2tp","defaultroute":false,"server":"myserver","username":"myusername","password":"mypassword","mtu":1410,"keepalive":"12","mtu":1410}
29472 root       784 S    xl2tpd-control add l2tp-vma pppoptfile=/tmp/l2tp/options.vma lns=myserver redial=yes redial timeout=20
```
"ifdown vma" or "ifup vma" has no effect either.

There's clearly something wrong with xl2tp, as I've no idea what "Y" and "wqĐwqĐ" mean in the log.
Maybe some bad pointer or memory corruption?

dangowrt, 2015-06-04:
Yeah, looks like a bad pointer... @yousong recently made some changes, maybe you two can work together to figure out what's going on here?

yousong, 2015-06-04:
Well, I also encountered the reported issue for one or two times when working with xl2tpd, but cannot reproduce it since then.  But the 'Y' and 'Trying to look agiain...' bit is not part of xelerance xl2tpd.  Nor are they applied in OpenWrt packages.  Have you locally applied the patch in [1]?

Anyway, if this is easily reproducible, fix it should also be easy...

 [1] https://github.com/xelerance/xl2tpd/issues/52

duvi, 2015-06-04:
I haven't modified anything or added any patch, I just build the openwrt package as it is.
But I looked at the source, and the referenced patch is already applied in the openwrt package; at least that is what xl2tpd.c looks like in build_dir/target-mips_34kc_uClibc-0.9.33.2/xl2tpd-1.3.7pre20150524

yousong, 2015-06-04:
Okay, never mind, that bit can be investigated later...  Have you found a more or less reliable way to trigger this issue?

duvi, 2015-06-04:
I checked, and I have 200-auto-reconnect.patch in the sources that includes this patch. But that is not here on github.

It seems "./scripts/feeds update" does not remove deleted files automatically locally?

duvi, 2015-06-05:
Without the patch I haven't seen this error yet.

However, there's another strange event. Sometimes the l2tp client loses echo "sync", and tries to reconnect. The first reconnect attempt is successful, the interface is up again.
Still, after 20 seconds, a reconnect is attempted again, which fails due to it can't rename the interface (which is correct, as an interface is already up and running with that name).
Then this goes on for ages, until suddenly after a couple of hours (I guess it's random) stops.
It seems like a confusion between xl2tpd and netifd trying to ahndle the reconnection.

```
Jun  5 05:29:20 pppd[6485]: No response to 12 echo-requests
Jun  5 05:29:20 pppd[6485]: Serial link appears to be disconnected.
Jun  5 05:29:20 pppd[6485]: Connect time 513.9 minutes.
Jun  5 05:29:20 pppd[6485]: Sent 252 bytes, received 252 bytes.
Jun  5 05:29:20 netifd: Network device 'l2tp-vma' link is down
Jun  5 05:29:20 syslog: 13[KNL] interface l2tp-vma deactivated
Jun  5 05:29:20 syslog: 12[KNL] 192.168.200.14 disappeared from l2tp-vma
Jun  5 05:29:20 pppd[6485]: Overriding mtu 1500 to 1410
Jun  5 05:29:20 pppd[6485]: Overriding mru 1500 to mtu value 1410
Jun  5 05:29:20 pppd[6485]: Connection terminated.
Jun  5 05:29:20 pppd[6485]: Modem hangup
Jun  5 05:29:20 syslog: 11[KNL] interface l2tp-vma deleted
Jun  5 05:29:20 pppd[6485]: Exit.
Jun  5 05:29:20 xl2tpd[1023]: child_handler : pppd exited for call 44518 with code 15
Jun  5 05:29:20 xl2tpd[1023]: call_close: Call 40692 to myserver disconnected
Jun  5 05:29:20 xl2tpd[1023]: Terminating pppd: sending TERM signal to pid 6485
Jun  5 05:29:21 xl2tpd[1023]: Disconnecting from myserver, Local: 11975, Remote: 26507
Jun  5 05:29:21 xl2tpd[1023]: Connection 26507 closed to myserver, port 1701 (Goodbye!)
Jun  5 05:29:21 netifd: Interface 'vma' is now down
Jun  5 05:29:21 xl2tpd[1023]: Can not find tunnel 11975 (refhim=0)
Jun  5 05:29:21 xl2tpd[1023]: network_thread: unable to find call or tunnel to handle packet.  call = 40692, tunnel = 11975 Dumping.
Jun  5 05:29:21 netifd: Interface 'vma' is setting up now
Jun  5 05:29:21 xl2tpd[1023]: Connecting to host myserver, port 1701
Jun  5 05:29:23 xl2tpd[1023]: Connection established to myserver, 1701.  Local: 63006, Remote: 18957 (ref=0/0).
Jun  5 05:29:23 xl2tpd[1023]: Calling on tunnel 63006
Jun  5 05:29:23 xl2tpd[1023]: check_control: Received out of order control packet on tunnel 18957 (got 0, expected 1)
Jun  5 05:29:23 xl2tpd[1023]: handle_packet: bad control packet!
Jun  5 05:29:23 xl2tpd[1023]: check_control: Received out of order control packet on tunnel 18957 (got 0, expected 1)
Jun  5 05:29:23 xl2tpd[1023]: handle_packet: bad control packet!
Jun  5 05:29:23 xl2tpd[1023]: Call established with myserver, Local: 64337, Remote: 20110, Serial: 82 (ref=0/0)
Jun  5 05:29:23 xl2tpd[1023]: start_pppd: I'm running:
Jun  5 05:29:23 xl2tpd[1023]: "/usr/sbin/pppd"
Jun  5 05:29:23 xl2tpd[1023]: "plugin"
Jun  5 05:29:23 xl2tpd[1023]: "pppol2tp.so"
Jun  5 05:29:23 xl2tpd[1023]: "pppol2tp"
Jun  5 05:29:23 xl2tpd[1023]: "9"
Jun  5 05:29:23 xl2tpd[1023]: "passive"
Jun  5 05:29:23 xl2tpd[1023]: "nodetach"
Jun  5 05:29:23 xl2tpd[1023]: ":"
Jun  5 05:29:23 xl2tpd[1023]: "file"
Jun  5 05:29:23 pppd[8164]: Plugin pppol2tp.so loaded.
Jun  5 05:29:23 pppd[8164]: pppd 2.4.7 started by root, uid 0
Jun  5 05:29:23 pppd[8164]: Using interface l2tp-vma
Jun  5 05:29:23 pppd[8164]: Connect: l2tp-vma <-->
Jun  5 05:29:23 pppd[8164]: Overriding mtu 1500 to 1410
Jun  5 05:29:23 pppd[8164]: Overriding mru 1500 to mtu value 1410
Jun  5 05:29:23 kernel: [50382.630000] l2tp-vma: renamed from ppp1
Jun  5 05:29:24 xl2tpd[1023]: Can not find tunnel 11975 (refhim=0)
Jun  5 05:29:24 xl2tpd[1023]: network_thread: unable to find call or tunnel to handle packet.  call = 40692, tunnel = 11975 Dumping.
Jun  5 05:29:26 pppd[8164]: EAP: Identity prompt "Name"
Jun  5 05:29:26 pppd[8164]: EAP authentication succeeded
Jun  5 05:29:26 pppd[8164]: local  IP address 192.168.200.14
Jun  5 05:29:26 pppd[8164]: remote IP address 192.168.200.1
Jun  5 05:29:26 syslog: 11[KNL] 192.168.200.14 appeared on l2tp-vma
Jun  5 05:29:26 syslog: 04[KNL] 192.168.200.14 disappeared from l2tp-vma
Jun  5 05:29:26 syslog: 05[KNL] 192.168.200.14 appeared on l2tp-vma
Jun  5 05:29:26 syslog: 03[KNL] interface l2tp-vma activated
Jun  5 05:29:26 netifd: Network device 'l2tp-vma' link is up
Jun  5 05:29:26 netifd: Interface 'vma' is now up
Jun  5 05:29:41 xl2tpd[1023]: Calling on tunnel 63006
Jun  5 05:29:41 xl2tpd[1023]: Call established with myserver, Local: 54805, Remote: 46959, Serial: 83 (ref=0/0)
Jun  5 05:29:41 xl2tpd[1023]: start_pppd: I'm running:
[... 120 lines cut ...]
Jun  5 09:35:18 xl2tpd[1023]: Call established with myserver, Local: 24735, Remote: 41008, Serial: 531 (ref=0/0)
Jun  5 09:35:18 xl2tpd[1023]: start_pppd: I'm running:
Jun  5 09:35:18 xl2tpd[1023]: "/usr/sbin/pppd"
Jun  5 09:35:18 xl2tpd[1023]: "plugin"
Jun  5 09:35:18 xl2tpd[1023]: "pppol2tp.so"
Jun  5 09:35:18 xl2tpd[1023]: "pppol2tp"
Jun  5 09:35:18 xl2tpd[1023]: "9"
Jun  5 09:35:18 xl2tpd[1023]: "passive"
Jun  5 09:35:18 xl2tpd[1023]: "nodetach"
Jun  5 09:35:18 xl2tpd[1023]: ":"
Jun  5 09:35:18 xl2tpd[1023]: "file"
Jun  5 09:35:18 xl2tpd[1023]: "/tmp/l2tp/options.vma"
Jun  5 09:35:18 pppd[21906]: Plugin pppol2tp.so loaded.
Jun  5 09:35:18 pppd[21906]: pppd 2.4.7 started by root, uid 0
Jun  5 09:35:18 pppd[21906]: Couldn't rename ppp2 to l2tp-vma
Jun  5 09:35:18 syslog: 10[KNL] interface ppp2 deleted
[... cut ...]
Jun  5 09:35:18 xl2tpd[1023]: child_handler : pppd exited for call 41008 with code 1
Jun  5 09:35:18 xl2tpd[1023]: call_close: Call 24735 to myserver disconnected
Jun  5 09:35:18 xl2tpd[1023]: Terminating pppd: sending TERM signal to pid 21906
Jun  5 09:35:18 xl2tpd[1023]: get_call: can't find call 24735 in tunnel 8714
 (ref=0/0)
Jun  5 09:35:38 xl2tpd[1023]: Calling on tunnel 8714
```
And nothing after this in the log, no reconnection, and the first original reconnected interface is still running and working.

duvi, 2015-06-05:
Maybe the redial triggered in l2tp.sh should be removed?
```
xl2tpd-control add l2tp-${interface} pppoptfile=${optfile} lns=${server} redial=yes redial timeout=20 || {
```
should be changed to
```
xl2tpd-control add l2tp-${interface} pppoptfile=${optfile} lns=${server} || {
```
?

dangowrt, 2015-06-05:
@duvi : I agree, changed in https://github.com/openwrt/packages/commit/0f562aa7c3af47e5a32b648e82dc648fd69d585a . Please try and report back if that fixes the issue.

duvi, 2015-06-05:
Not so good, I guess "timeout=20" should've been removed as well, it's probably "redial timeout = 20" fully.
Also:
```
Fri Jun  5 17:02:19 2015 daemon.notice netifd: Interface 'vma' is setting up now
Fri Jun  5 17:02:20 2015 daemon.crit xl2tpd[1025]: parse_one_line: token 3: Unknown field 'timeout'
Fri Jun  5 17:02:20 2015 daemon.notice netifd: vma (1344): xl2tpd-control: Add l2tp-vma failed
Fri Jun  5 17:02:20 2015 daemon.debug xl2tpd[1025]: No such tunnel 'l2tp-vma'
Fri Jun  5 17:02:20 2015 daemon.notice netifd: vma (1357): xl2tpd-control: Remove l2tp-vma failed
Fri Jun  5 17:02:20 2015 daemon.notice netifd: Interface 'vma' is now down
```

yousong, 2015-06-05:
> Not so good, I guess "timeout=20" should've been removed as well, it's probably "redial timeout = 20" fully.

Indeed, `redial timeout=20` is a single configuration unit.

yousong, 2015-06-05:
I still have not tried ipsec with xl2tpd yet.  But the newly reported phenomenon seems like a issue on xl2tpd side.  xl2tpd should not try redialing  the same LAC if it is already running and up.

The `redial=yes` option was intentionally left there in case the dial request from netifd fails on the first try.  For details please see link [1]

 [1] https://lists.openwrt.org/pipermail/openwrt-devel/2015-May/033114.html

yousong, 2015-06-05:
@dangowrt I got another trivial change [1] in my local repository for a while.  Could you pick it up for the next commit cycle?  Feel free to adapt it as it does not apply now :)

 [1] xl2tpd: set procd respawn param. http://sprunge.us/aCCC

duvi, 2015-06-05:
All right, but you removed "redial=yes redial" and left "timeout=20" in the commit, that's what I meant.

duvi, 2015-06-05:
Forgot to say, I'm trying l2tp over ispec. And sometimes after the hourly ipsec rekey come the lost l2tp echoes, that's what triggers the l2tp reconnection.

dangowrt, 2015-06-05:
Hopefully fixed in https://github.com/openwrt/packages/commit/e0b2a226a37845fdb8b789432174c788d62c6a64

Generally, I agree that xl2tpd generally should not attempt to redail. 
Probably we have to handle the case of a connection failing to connect initially separately...

duvi, 2015-06-05:
Thanks, I'll test this version for a while.

duvi, 2015-06-05:
Unfortunately not so good results:
```
Jun  5 19:59:50 xl2tpd[1021]: check_control: Received out of order control packet on tunnel 25233 (got 33, expected 34)
Jun  5 19:59:50 xl2tpd[1021]: handle_packet: bad control packet!
Jun  5 19:59:51 xl2tpd[1021]: check_control: Received out of order control packet on tunnel 25233 (got 33, expected 34)
Jun  5 19:59:51 xl2tpd[1021]: handle_packet: bad control packet!
Jun  5 19:59:52 xl2tpd[1021]: check_control: Received out of order control packet on tunnel 25233 (got 33, expected 34)
Jun  5 19:59:52 xl2tpd[1021]: handle_packet: bad control packet!
Jun  5 19:59:53 xl2tpd[1021]: check_control: Received out of order control packet on tunnel 25233 (got 33, expected 34)
Jun  5 19:59:53 xl2tpd[1021]: handle_packet: bad control packet!
Jun  5 19:59:54 xl2tpd[1021]: control_finish: Connection closed to myserver, port 1701 (Timeout), Local: 58409, Remote: 25233
Jun  5 19:59:54 xl2tpd[1021]: Terminating pppd: sending TERM signal to pid 2640
Jun  5 19:59:54 pppd[2640]: Terminating on signal 15
Jun  5 19:59:54 pppd[2640]: Connect time 32.1 minutes.
Jun  5 19:59:54 pppd[2640]: Sent 1904 bytes, received 1904 bytes.
Jun  5 19:59:54 netifd: Network device 'l2tp-vma' link is down
Jun  5 19:59:54 xl2tpd[1021]: Can not find tunnel 58409 (refhim=0)
Jun  5 19:59:54 xl2tpd[1021]: network_thread: unable to find call or tunnel to handle packet.  call = 14351, tunnel = 58409 Dumping.
Jun  5 19:59:54 syslog: 16[KNL] interface l2tp-vma deactivated
Jun  5 19:59:54 syslog: 05[KNL] 192.168.200.14 disappeared from l2tp-vma
Jun  5 19:59:54 pppd[2640]: Overriding mtu 1500 to 1410
Jun  5 19:59:54 pppd[2640]: Overriding mru 1500 to mtu value 1410
Jun  5 19:59:54 pppd[2640]: Connection terminated.
Jun  5 19:59:54 syslog: 03[KNL] interface l2tp-vma deleted
Jun  5 19:59:54 pppd[2640]: Exit.
Jun  5 19:59:54 netifd: Interface 'vma' is now down
Jun  5 19:59:54 netifd: Interface 'vma' is setting up now
Jun  5 19:59:54 xl2tpd[1021]: Connecting to host myserver, port 1701
Jun  5 19:59:55 xl2tpd[1021]: Can not find tunnel 58409 (refhim=0)
Jun  5 19:59:55 xl2tpd[1021]: network_thread: unable to find call or tunnel to handle packet.  call = 0, tunnel = 58409 Dumping.
Jun  5 19:59:56 xl2tpd[1021]: Can not find tunnel 58409 (refhim=0)
Jun  5 19:59:56 xl2tpd[1021]: network_thread: unable to find call or tunnel to handle packet.  call = 0, tunnel = 58409 Dumping.
Jun  5 19:59:57 xl2tpd[1021]: Can not find tunnel 58409 (refhim=0)
Jun  5 19:59:57 xl2tpd[1021]: network_thread: unable to find call or tunnel to handle packet.  call = 0, tunnel = 58409 Dumping.
Jun  5 19:59:57 xl2tpd[1021]: Can not find tunnel 58409 (refhim=0)
Jun  5 19:59:57 xl2tpd[1021]: network_thread: unable to find call or tunnel to handle packet.  call = 14351, tunnel = 58409 Dumping.
Jun  5 19:59:58 xl2tpd[1021]: Can not find tunnel 58409 (refhim=0)
Jun  5 19:59:58 xl2tpd[1021]: network_thread: unable to find call or tunnel to handle packet.  call = 0, tunnel = 58409 Dumping.
Jun  5 19:59:59 xl2tpd[1021]: Maximum retries exceeded for tunnel 46783.  Closing.
Jun  5 19:59:59 xl2tpd[1021]: Connection 0 closed to myserver, port 1701 (Timeout)
Jun  5 20:00:04 xl2tpd[1021]: Unable to deliver closing message for tunnel 46783. Destroying anyway.
```
And that's the end, no more reconnect attempts from this point, nothing in the logs about xl2tpd or ppp or netifd.

yousong, 2015-06-05:
looks like xl2tpd has really given up redialing.  it's expected.  as a side note, i usually enable pppd_options "debug" for more verbose output,  "nomppe" to prevent corruption in l2tp data channel.

dangowrt, 2015-06-28:
I updated l2tp.sh handler script, it'd be great if anyone can test that before I merge it:
https://github.com/dangowrt/packages/tree/xl2tpd-netifd-sh-scripts-sync
https://github.com/dangowrt/packages/commit/aed5a4fc716ccaadcb1f1d64d0590001d1a79dbf

duvi, 2015-06-28:
Testing right away, thanks!
I'll report with results, but it can take some time, as the old script did also reconnect most of the time, and only sometimes it didn't.

puleglot, 2015-06-28:
I'm also testing it. And it fails to pass trivial test: disconnect router's uplink for 10 secs and plug it again:
```
... snip ...
Sun Jun 28 22:20:22 2015 daemon.notice netifd: Interface 'vpn' is now down
Sun Jun 28 22:20:22 2015 daemon.notice netifd: Interface 'vpn' is setting up now
Sun Jun 28 22:20:22 2015 daemon.notice xl2tpd[1727]: Connecting to host XXXXXXX, port 1701
Sun Jun 28 22:20:22 2015 daemon.info xl2tpd[1727]: control_finish: Connection closed to 85.21.0.242, port 1701 (Reached configured max-tunnels limit), Local: 6675, Remote: 22003
Sun Jun 28 22:20:22 2015 daemon.info xl2tpd[1727]: Can not find tunnel 5447 (refhim=0)
Sun Jun 28 22:20:22 2015 daemon.debug xl2tpd[1727]: network_thread: unable to find call or tunnel to handle packet.  call = 64335, tunnel = 5447 Dumping.
Sun Jun 28 22:20:22 2015 daemon.info xl2tpd[1727]: Can not find tunnel 5447 (refhim=0)
Sun Jun 28 22:20:22 2015 daemon.debug xl2tpd[1727]: network_thread: unable to find call or tunnel to handle packet.  call = 64335, tunnel = 5447 Dumping.
Sun Jun 28 22:20:22 2015 daemon.info xl2tpd[1727]: Can not find tunnel 5447 (refhim=0)
Sun Jun 28 22:20:22 2015 daemon.debug xl2tpd[1727]: network_thread: unable to find call or tunnel to handle packet.  call = 64335, tunnel = 5447 Dumping.
Sun Jun 28 22:20:23 2015 daemon.info xl2tpd[1727]: Can not find tunnel 5447 (refhim=0)
Sun Jun 28 22:20:23 2015 daemon.debug xl2tpd[1727]: network_thread: unable to find call or tunnel to handle packet.  call = 64335, tunnel = 5447 Dumping.
Sun Jun 28 22:20:23 2015 daemon.info xl2tpd[1727]: Can not find tunnel 5447 (refhim=0)
Sun Jun 28 22:20:23 2015 daemon.debug xl2tpd[1727]: network_thread: unable to find call or tunnel to handle packet.  call = 64335, tunnel = 5447 Dumping.
Sun Jun 28 22:20:23 2015 daemon.info xl2tpd[1727]: Can not find tunnel 5447 (refhim=0)
Sun Jun 28 22:20:23 2015 daemon.debug xl2tpd[1727]: network_thread: unable to find call or tunnel to handle packet.  call = 64335, tunnel = 5447 Dumping.
Sun Jun 28 22:20:23 2015 daemon.info xl2tpd[1727]: Can not find tunnel 5447 (refhim=0)
Sun Jun 28 22:20:23 2015 daemon.debug xl2tpd[1727]: network_thread: unable to find call or tunnel to handle packet.  call = 64335, tunnel = 5447 Dumping.
Sun Jun 28 22:20:23 2015 daemon.info xl2tpd[1727]: Can not find tunnel 5447 (refhim=0)
Sun Jun 28 22:20:23 2015 daemon.debug xl2tpd[1727]: network_thread: unable to find call or tunnel to handle packet.  call = 64335, tunnel = 5447 Dumping.
Sun Jun 28 22:20:23 2015 daemon.info xl2tpd[1727]: Can not find tunnel 5447 (refhim=0)
Sun Jun 28 22:20:23 2015 daemon.debug xl2tpd[1727]: network_thread: unable to find call or tunnel to handle packet.  call = 64335, tunnel = 5447 Dumping.
Sun Jun 28 22:20:23 2015 daemon.info xl2tpd[1727]: Can not find tunnel 5447 (refhim=0)
... snip ...
```
... and so on. After ~20 seconds xl2tpd is giving up and stops retrying. Error "Reached configured max-tunnels limit" is returned by LNS probably because it think that client is still connected. Manual "ifup vpn" brings tunnel up without any problem.

dangowrt, 2015-06-28:
This looks like the L2TP control session survives the interruption but the PPP session does not.
Thus, when netifd tries to reconnect the interface, it fails due to the control session not having been closed or re-used...
The easiest way I can imagine to work around this would be to also pass pre-PPP errors to netifd, so it can decide whether it's worth retrying (like in the above case) or to halt (intended in case of e.g. credentials being wrong). In both cases, we may need to take more care not to leak tunnels and sessions and destroy them once they are no longer used.
@yousong what do you think?

yousong, 2015-06-29:
Hi, Daniel

I am currently using OpenWrt BB with backported netifd and xl2tpd (`redial=yes` bit included, and no IPSec).  I tried both `ifconfig eth1 down` and `ifdown wan` and later brought up the wan network, l2tp interface came up just fine.

`ppp_generic_teardown` is a nop in the case of xl2tpd as there is now `no_proto_task` for it in netifd.

With xl2tpd, netifd will only retry when l3_device of the interface goes down.  After netifd has done its part (teardown and setup), it depends on xl2tpd to redial on and on until the l3_device goes up again.  So my guess is that the failed test case from @puleglot can be resolved by adding back `redial=yes redial timeout=20` to the xl2tpd-control add command.

Cheers.

dangowrt, 2015-06-29:
@yousong `ppp_generic_teardown` should handle the exit-code of the ppp process, see
https://dev.openwrt.org/browser/trunk/package/network/services/ppp/files/ppp.sh#L143
It can be used to replace most of
https://github.com/openwrt/packages/blob/master/net/xl2tpd/files/l2tp.sh#L101
However, as pppd is not started by netifd, I wonder how and if `$ERROR` is actually defined.

Somehow I get the feeling that the nicest thing would be if xtl2pd didn't start pppd by itself but rather requests netifd to do so. Maybe a drop-in /usr/bin/pppd executable can hand-over the L2TP endpoint parameters among with all other options to netifd via ubus which can then start pppd. @nbd168 @blogic ?

Anyway, I agree that for now the easiest would be to add back `redial=yes redial timeout=20` and then back-port the whole thing to https://github.com/openwrt/packages/tree/for-15.05

yousong, 2015-06-29:
Hi, Daniel

That part handling `$ERROR` should indeed be removed.  `ppp_generic_teardown` only tells netifd to kill the `proto_task` which is never present anyway.  netifd currently has no way of knowing the exit status of the `pppd` process started by `xl2tpd`.

Anyway I think there should be many other potential bugs or inconsistencies in xl2tpd itself.  I got the feeling that it's not that robust and maybe even more fragile when working with IPSec and that's why it needs so much care such as redial, teardown/setup to make sure the call procedure start over and work again.

dangowrt, 2015-06-29:
@yousong @puleglot @duvi Please have a look at and try with https://github.com/dangowrt/packages/commit/38cf1424182b6ab219458a92c35eec668e71ab9

dangowrt, 2015-06-29:
For ppp unnumbered support to work we will need to include ppp.sh. Updated patch:
https://github.com/dangowrt/packages/commit/e9e24008a76689f8d7b6de7c70c7d114b7f41d0b

puleglot, 2015-06-29:
Thanks! This version works better. However once I've got a strange situation in which from netifd's point of view the tunnel is up but xl2tpd is constantly trying to start second pppd process:
```
Mon Jun 29 16:16:42 2015 daemon.notice netifd: Interface 'vpn' is now down
Mon Jun 29 16:16:42 2015 daemon.notice netifd: Interface 'vpn' is setting up now
Mon Jun 29 16:16:43 2015 daemon.notice xl2tpd[1709]: Connecting to host <HOST>, port 1701
Mon Jun 29 16:16:43 2015 daemon.warn xl2tpd[1709]: handle_avps:  don't know how to handle atribute 46.
Mon Jun 29 16:16:43 2015 daemon.notice xl2tpd[1709]: Connection established to XX.XX.XX.XX, 1701.  Local: 45836, Remote: 60636 (ref=0/0).
Mon Jun 29 16:16:43 2015 daemon.notice xl2tpd[1709]: Calling on tunnel 45836
Mon Jun 29 16:16:43 2015 daemon.notice xl2tpd[1709]: Call established with XX.XX.XX.XX, Local: 31301, Remote: 33636, Serial: 6 (ref=0/0)
Mon Jun 29 16:16:43 2015 daemon.debug xl2tpd[1709]: start_pppd: I'm running: 
Mon Jun 29 16:16:43 2015 daemon.debug xl2tpd[1709]: "/usr/sbin/pppd" 
Mon Jun 29 16:16:43 2015 daemon.debug xl2tpd[1709]: "plugin" 
Mon Jun 29 16:16:43 2015 daemon.debug xl2tpd[1709]: "pppol2tp.so" 
Mon Jun 29 16:16:43 2015 daemon.debug xl2tpd[1709]: "pppol2tp" 
Mon Jun 29 16:16:43 2015 daemon.debug xl2tpd[1709]: "11" 
Mon Jun 29 16:16:43 2015 daemon.debug xl2tpd[1709]: "passive" 
Mon Jun 29 16:16:43 2015 daemon.debug xl2tpd[1709]: "nodetach" 
Mon Jun 29 16:16:43 2015 daemon.debug xl2tpd[1709]: ":" 
Mon Jun 29 16:16:43 2015 daemon.debug xl2tpd[1709]: "file" 
Mon Jun 29 16:16:43 2015 daemon.info pppd[3223]: Plugin pppol2tp.so loaded.
Mon Jun 29 16:16:43 2015 daemon.notice pppd[3223]: pppd 2.4.7 started by root, uid 0
Mon Jun 29 16:16:43 2015 daemon.info pppd[3223]: Using interface l2tp-vpn
Mon Jun 29 16:16:43 2015 daemon.notice pppd[3223]: Connect: l2tp-vpn <--> 
Mon Jun 29 16:16:43 2015 kern.info kernel: [  355.930000] l2tp-vpn: renamed from ppp0
Mon Jun 29 16:16:43 2015 daemon.info xl2tpd[1709]: Will redial in 20 seconds
Mon Jun 29 16:16:46 2015 daemon.info pppd[3223]: CHAP authentication succeeded: CHAP authentication success, unit 3650
Mon Jun 29 16:16:46 2015 daemon.notice pppd[3223]: CHAP authentication succeeded
Mon Jun 29 16:16:46 2015 daemon.notice pppd[3223]: local  IP address XX.XX.XX.XX
Mon Jun 29 16:16:46 2015 daemon.notice pppd[3223]: remote IP address XX.XX.XX.XX
Mon Jun 29 16:16:46 2015 daemon.notice pppd[3223]: primary   DNS address XX.XX.XX.XX
Mon Jun 29 16:16:46 2015 daemon.notice pppd[3223]: secondary DNS address XX.XX.XX.XX
Mon Jun 29 16:16:46 2015 daemon.notice netifd: Network device 'l2tp-vpn' link is up
Mon Jun 29 16:16:46 2015 daemon.notice netifd: Interface 'vpn' is now up
Mon Jun 29 16:16:47 2015 user.notice firewall: Reloading firewall due to ifup of vpn (l2tp-vpn)
Mon Jun 29 16:17:03 2015 daemon.notice xl2tpd[1709]: Connecting to host <HOST>, port 1701
Mon Jun 29 16:17:03 2015 daemon.notice xl2tpd[1709]: Connection established to XX.XX.XX.XX, 1701.  Local: 32113, Remote: 36925 (ref=0/0).
Mon Jun 29 16:17:03 2015 daemon.notice xl2tpd[1709]: Calling on tunnel 32113
Mon Jun 29 16:17:03 2015 daemon.info xl2tpd[1709]: control_finish: Connection closed to XX.XX.XX.XX, serial 7 (Resource Shortage)
Mon Jun 29 16:17:13 2015 daemon.info xl2tpd[1709]: control_finish: Connection closed to XX.XX.XX.XX, port 1701 (No application/session timer expired), Local: 32113, Remote: 36925
Mon Jun 29 16:17:23 2015 daemon.notice xl2tpd[1709]: Connecting to host <HOST>, port 1701
Mon Jun 29 16:17:23 2015 daemon.notice xl2tpd[1709]: Connection established to XX.XX.XX.XX, 1701.  Local: 19532, Remote: 2241 (ref=0/0).
Mon Jun 29 16:17:23 2015 daemon.notice xl2tpd[1709]: Calling on tunnel 19532
Mon Jun 29 16:17:23 2015 daemon.notice xl2tpd[1709]: Call established with XX.XX.XX.XX, Local: 27440, Remote: 32674, Serial: 8 (ref=0/0)
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: start_pppd: I'm running: 
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: "/usr/sbin/pppd" 
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: "plugin" 
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: "pppol2tp.so" 
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: "pppol2tp" 
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: "11" 
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: "passive" 
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: "nodetach" 
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: ":" 
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: "file" 
Mon Jun 29 16:17:23 2015 daemon.info pppd[3550]: Plugin pppol2tp.so loaded.
Mon Jun 29 16:17:23 2015 daemon.notice pppd[3550]: pppd 2.4.7 started by root, uid 0
Mon Jun 29 16:17:23 2015 daemon.err pppd[3550]: Couldn't rename ppp1 to l2tp-vpn
Mon Jun 29 16:17:23 2015 daemon.info pppd[3550]: Exit.
Mon Jun 29 16:17:23 2015 daemon.debug xl2tpd[1709]: child_handler : pppd exited for call 32674 with code 1
Mon Jun 29 16:17:23 2015 daemon.info xl2tpd[1709]: call_close: Call 27440 to XX.XX.XX.XX disconnected
Mon Jun 29 16:17:33 2015 daemon.info xl2tpd[1709]: control_finish: Connection closed to XX.XX.XX.XX, port 1701 (No application/session timer expired), Local: 19532, Remote: 2241
Mon Jun 29 16:17:33 2015 daemon.debug xl2tpd[1709]: Terminating pppd: sending TERM signal to pid 3550
Mon Jun 29 16:17:53 2015 daemon.notice xl2tpd[1709]: Connecting to host <HOST>, port 1701
[... cut ...]
Mon Jun 29 16:19:04 2015 daemon.debug xl2tpd[1709]: child_handler : pppd exited for call 7660 with code 1
Mon Jun 29 16:19:04 2015 daemon.info xl2tpd[1709]: call_close: Call 16626 to XX.XX.XX.XX disconnected
Mon Jun 29 16:19:14 2015 daemon.info xl2tpd[1709]: control_finish: Connection closed to XX.XX.XX.XX, port 1701 (No application/session timer expired), Local: 7389, Remote: 18170
Mon Jun 29 16:19:14 2015 daemon.debug xl2tpd[1709]: Terminating pppd: sending TERM signal to pid 3605
... and so on
```
The old pppd process is still running and the tunnel works fine.

dangowrt, 2015-06-29:
@puleglot Looks similar to https://github.com/openwrt/packages/issues/1334#issuecomment-109211342

duvi, 2015-07-31:
For the last 4 weekes I tried all versions here.
The main pppoe connection reconnects every 7 days. None of the l2tp.sh files here (aed5a4f, 38cf142, e9e2400) managed to reconnect the l2tp connection after pppoe reconnection. Although, sometimes l2tp did reconnect automatically, if it was dropped for unknown reason. But never after pppoe reconnection.
Ps: I always overwrote l2tp.sh with the next version in the running system, then did "/etc/init.d/xl2tpd stop", then started the vpn interface with ifup. I never restarted the router to test a different l2tp.sh.

yousong, 2015-08-05:
I will try adding a magic timeout for reconnection to netifd and then `timeout=yes` can be dropped from `l2tp.sh`.  But xl2tpd with IPSec is still another story I guess.

dangowrt, 2015-08-10:
@yousong Given the high quality of your work in the past and in case you are willing to do that job, I'd happily transfer maintainership/package ownership to you. I got only limited resources and no real-life testing environment for xl2tpd by now, thus would be happy to have someone more actively working on it to be the OpenWrt package maintainer.

duvi, 2015-08-10:
@yousong I've been testing with IPSec, so that might also give an extra delay to reconnection after the wan interface comes back.
I could also test without ipsec if that could help.

yousong, 2015-08-12:
Hi, I just posted patches for both netifd [1] and xl2tpd [2] for you to review and test.

[1] http://patchwork.ozlabs.org/patch/506592/
[2] https://github.com/openwrt/packages/pull/1653

yousong, 2015-08-18:
Hi, in case it may help, 

 1. The patch series for xl2tpd can be downloaded from URL [1].
 2. Patch for netifd can be applied locally by putting the it into directory `package/network/config/netifd/patches` (create one it is not there).
 3. Rebuild.

[1] https://github.com/openwrt/packages/pull/1653.patch

uk3, 2015-08-18:
My quick (crude) workaround was a cron-based watchdog for the tunnel:

* * * * * /usr/bin/jshn -r "$( /sbin/ifstatus Dialer)" | /bin/grep "json_add_boolean 'up' 0;" > /dev/null && /sbin/ifup Dialer

Replace 'Dialer' with the name of the interface (twice). This brings the tunnel back up after connection loss.

yousong, 2015-08-21:
Hello, the patch for netifd was just updated [1]

 [1] http://patchwork.ozlabs.org/patch/509240/

yousong, 2015-09-01:
Now that patches for both netifd and xl2tpd were merged, I will close this issue now.  Thanks.

closed 2015-09-01 by yousong